ct-Open-Source/tuya-convert #841: Failure, new wifi chip, Smart Plug Amysen

duckythescientist, 2020-12-02:
I'm making this issue so that other people can google this and avoid the problem. I bought a four-pack of plugs last week from the same listing as I had bought back in May of this year. The old ones worked with tuya-convert, but the new ones don't. They use a chip different than the ESP.

Brand: "Smart Plug Amysen"
Model: YX-WS02B
FCCID: 2AOT8-YXWS06
Microcontroller SoC: ASR Microelectronics ASR5502 (ASR5502A)

Alibaba's AliOS is pretty much the only thing I've found so far that references this chip. I may try poking at it, but I can't read Chinese.
https://code.aliyun.com/vivid8710/AliOS-Things/tree/master/platform/mcu/asr5501

amarjandu, 2020-12-02:
This might be related to #483 do you get a error like `could not establish sslpsk socket: [SSL: DECRYPTION_FAILED_OR_BAD_RECORD_MAC] decryption failed or bad record mac (_ssl.c:852)` within the logs?

kueblc, 2020-12-02:
Thanks for the heads up @duckythescientist 

Unfortunately a new wifi chip makes it impossible to install ESP firmware, regardless of whether is has a new PSK format, so it is not the same as #483. Thanks for chiming in anyway @amarjandu.

closed 2020-12-02 by kueblc