ct-Open-Source/tuya-convert #843: Gosund Smart Plug flash issue

compata, 2020-12-02:
I have eight of these devices which I successfully flashed about a year ago. Now I have acquired a few more and am having trouble flashing them. I presume there has been some change to their internals. Reading other posts online, I see others with issues similar to mine, though on different devices. What I didn't see was a clear statement of any solution so I'm asking here.

I first tried flashing with the same tuya-convert that I used last year. That didn't work at all. So I downloaded the latest and tried again. All logs, the console output, and a tcpdump capture file are attached (if I'm successful in the attachment). It appears to me that in packet 3318 of the capture the sfile:///home/dave/try3/gosund.zipmart plug is assigned address 10.42.42.27. Of course, any conversion code that expects to find it at 10.42.42.42 will fail. Or maybe I don't understand how this should work. (The device assigned 10.42.42.15 appears to be my phone using a randomized MAC address.)

I see that start_flash.sh now includes a strong warning not to change the IP address so I haven't tried that. But it certainly looks as though something is wrong with the address assignments.

compata, 2020-12-02:
[gosund.zip](https://github.com/ct-Open-Source/tuya-convert/files/5627483/gosund.zip)

It appears that Github doesn't play well with Waterfox. To get the attachment here I had to use Firefox. Why do so many sites fail to provide compatibility with the wide variety of browsers available? (Side comment. Let's not divert from the Tasmota issue.)

kueblc, 2020-12-02:
The device will only get the address `10.42.42.42` once tuya-convert has successfully taken command of the device. If the device does not have that address, it is still running Tuya firmware. You are correct not to change the script.

Unfortunately the issue with this device appears to be the new firmware from Tuya which patches against tuya-convert. It uses a new PSK format that we have not been able to exploit. See #483 and the wiki.

> It appears that Github doesn't play well with Waterfox. To get the attachment here I had to use Firefox. Why do so many sites fail to provide compatibility with the wide variety of browsers available? (Side comment. Let's not divert from the Tasmota issue.)

Here here! I have been frustrated by this too. We are no longer in the dark ages of MSIE. How hard is it to support a few minor browser variations?

closed 2020-12-02 by kueblc

compata, 2020-12-02:
Thanks for the explanation, Colin. Does this mean that these devices are useless with Tasmota, at least for now? They are returnable for refund through January.

If they are already running incompatible Tuya firmware, then completing the Tuya installation would not change their status, right? So that,  even if I do that, if and when a solution to the PSK issue is found they could still be converted to Tasmota?

kueblc, 2020-12-02:
Tuya may release more firmware updates in the future which patch yet undiscovered vulnerabilities, so there may be a case for keeping these offline.

However it is impossible to say if or when we will overcome their latest patch, so if you are comfortable using the stock firmware I would say go for it. Subscribe to #483 so that if we discover a workaround, you will know right away to take your devices off the cloud before it gets patched again.