ct-Open-Source/tuya-convert #845: Problems flashing avatar AWP08L

rodi-nrw, 2020-12-04:
I flashed some gosund SP111 plugs already with tuya-convert on a Rpi3.

Trying to flash the avatar AWP08L i got issues.
First i updated the raspian, then i got the newest tuya-convert and started the install_prereq.sh.
Then i started the start_flash.sh, conntected a smartphone to the network and hold the button on the avatar plug until the light started blinking fast. Last i pressed ENTER to start the flashingprocess.

The blinking light stopped immidiatly. An on the Rpi it showed SmartConfig complete and retry in 3s.

Is it possible to flash this device?

kueblc, 2020-12-04:
Hi @rodi-nrw 

You will need to post your logs for assistance, as the directions indicate when opening a new issue.

rodi-nrw, 2020-12-05:
smathack-psk.log
`new client on port 443 from 10.42.42.28:36140
ID: 0242416f68626d64366147393149465231fcb1ff964e929bc240b21f3fcd521c3ce715b9a72c975a6f6d2a83239e2bf77d                
PSK: 1eee9359d85e9506ef1a7355cb5fc591da1ce39e3dee7863dae327cf682fba53                                                 
could not establish sslpsk socket: [SSL: DECRYPTION_FAILED_OR_BAD_RECORD_MAC] decryption failed or bad record mac (_ssl.c:1056)
new client on port 443 from 10.42.42.28:12239
ID: 0242416f68626d6436613f3f3f3f3f3f3ffcb1ff964e929bc240b21f3fcd521c3ce715b9a72c975a6f6d2a83239e2bf77d                
Prefix: b'BAohbmd6a???????'
PSK: 88859b1121db139edca301ac79cb9eaade93da69dfde94bbd558516a71699b9d                                                 
could not establish sslpsk socket: [SSL: DECRYPTION_FAILED_OR_BAD_RECORD_MAC] decryption failed or bad record mac (_ssl.c:1056)`

These lines for every reconnection.

smarthack-wifi.log
`wlan0: AP-STA-DISCONNECTED 88:11:96:b3:1b:28`

This line for the first contact.

kueblc, 2020-12-05:
Duplicate of #483, this firmware has been patched against tuya-convert. It is not currently possible to flash this OTA.

closed 2020-12-05 by kueblc