ct-Open-Source/tuya-convert #846: Device did not appear with the intermediate firmware

Cheddaz, 2020-12-05:
Hi, having trouble flashing some tuya smart switches I purchased from ebay. Very new to all this so excuse my very stupid question of, where do I get my log files? Running from a Raspbian Lite install on an RPI4, and accessing via ssh on a Macbook. The following is a link to the switch. 
(1/2/3gang) https://www.ebay.com.au/itm/Smart-Home-WiFi-Touch-Light-Wall-Switch-Panel-For-Alexa-Google-APP-1-2-3-Gang/174338811133?ssPageName=STRK%3AMEBIDX%3AIT&var=473534553974&_trksid=p2057872.m2749.l2649
(4 gang) https://www.ebay.com.au/itm/Smart-WiFi-Touch-Switch-RF-Wall-Light-Panel-1-2-3-4-Gang-With-Alexa-Google-Home/363092177851?ssPageName=STRK%3AMEBIDX%3AIT&var=632304674396&_trksid=p2057872.m2749.l2649
I have all gang iterations (1,2,3,4) in the house that need to be flashed, but currently I'm flashing the 2-gang version.

ortegafernando, 2020-12-13:
https://github.com/ct-Open-Source/tuya-convert/issues/483

kueblc, 2020-12-13:
Indeed @ortegafernando is correct, this is the new patched firmware from Tuya, #483.

closed 2020-12-13 by kueblc