ct-Open-Source/tuya-convert #848: Feit PLUGWIFIG2 can't flash

rpavlik, 2020-12-05:
I bought a 3 pack of these at Costco (us) because I thought they were the same as the PLUGWIFI I had previously bought and converted - couldn't see fccid until I got them home. However the internal photos appear to be similar, with the 2S 8285 based Tuya module.

I'm running t-c on a raspberry pi 3b that I've used for this purpose previously. When following the instructions, it just gets stuck repeatedly re-sending the smart config packets. After the first time though, the device is no longer in the right fast-blinking mode, despite not seeing any connections to the ap.

Afraid I might have a different module internally, though I'd be surprised if a big company like this changed modules without appropriate fcc paperwork. Unfortunately it's really hard to open, so I don't know for sure.  It doesn't match the other issue I found searching like this, since I don't see any connections to the WiFi.

I'll upload the logs later today

leeuwa, 2020-12-07:
I've got the same problem with [these]([url]https://www.aliexpress.com/item/4000478798085.html?spm=a2g0s.9042311.0.0.27424c4dVkSNy7(url)) Smartplugs, just like @rpavlik described. These Smartplugs connect to the Tuya app. When using Tuya Convert the LED stops blinking, so its seems its making some connection with the plug. But after that nothing happens anymore.

I see also the message 'wlan0: Could not connect to kernel driver' in the wifi log-file.

[smarthack-mqtt.log](https://github.com/ct-Open-Source/tuya-convert/files/5654292/smarthack-mqtt.log)
[smarthack-psk.log](https://github.com/ct-Open-Source/tuya-convert/files/5654294/smarthack-psk.log)
[smarthack-udp.log](https://github.com/ct-Open-Source/tuya-convert/files/5654295/smarthack-udp.log)
[smarthack-web.log](https://github.com/ct-Open-Source/tuya-convert/files/5654296/smarthack-web.log)
[smarthack-wifi.log](https://github.com/ct-Open-Source/tuya-convert/files/5654297/smarthack-wifi.log)

I've managed to flash Blitzwolf Smartplugs before with Tuya Convert. The are a little bit big, so I thought I buy some smaller ones. No luck so far.

rpavlik, 2020-12-07:
I think these are actually "blocked on https://github.com/ct-Open-Source/tuya-convert/wiki/Collaboration-document-for-PSK-Identity-02#known-affected-devices " - since my device appears there.

Troubleshooting page says 'wlan0: Could not connect to kernel driver' is harmless.

leeuwa, 2020-12-07:
> I think these are actually "blocked on https://github.com/ct-Open-Source/tuya-convert/wiki/Collaboration-document-for-PSK-Identity-02#known-affected-devices " - since my device appears there.
> 
> Troubleshooting page says 'wlan0: Could not connect to kernel driver' is harmless.

I don't know the brand, so I don't know if they are blocked.

llluis, 2020-12-07:
We don't get to the point to see the PSK in the log, so not sure if it's the "PSK 02" issue.

geoffory, 2020-12-09:
I've got the exact same issue as rpavlik with these Feit SYW-PLUGWIFIG2's.

Tried all 3, and all just get stuck looping with sending the smart plug config packets. Plug goes from fast flashing 'pairing' mode to no LED once I start tuya-convert, so seems it finds the device fine.

This is my first time trying to use tuya-convert, so was afraid it was user-error, but sounds like it may be an issue with this specific device.

geoffory, 2020-12-09:
I opened up one of my Feit SYW-PLUGWIFIG2 to have a look, in hopes I could solder and flash it directly.

However the wifi chip I found inside does not match the internal photos from [https://fccid.io/SYW-PLUGWIFIG2](https://fccid.io/SYW-PLUGWIFIG2).

FCC photos show a TYWE2S module, containing an ESP8285. I found Tuya documentation for it [here](https://developer.tuya.com/en/docs/iot/device-development/module/wifi-module/we-series-module/wifie2smodule?id=K9605u79tgxug).

Instead mine contained a WB2S module, which instead is based on a BK7231T RF Chip. There's Tuya documentation [here](https://developer.tuya.com/en/docs/iot/device-development/module/wifibt-dual-mode-module/wb-series-module/wb2s-module-datasheet?id=K9ghecl7kc479).

I suspect this may mean flashing with Tasmota is impossible.

Teardown photos are [posted here](https://imgur.com/a/ZMu7eWV).

rpavlik, 2020-12-09:
oh thank you for doing the teardown of this one. Quite a bummer, but at least it's not a tuya-convert problem.

gytisgreitai, 2021-01-03:
> I now can move on to the alternate way of getting these to play locally for me.

What's the alternate way?

z3braman, 2021-01-03:
I have the Costco plugs but they have the TYWE2S module. I had the same experience rpavlik had and could not flash. My internals match the internal photos in the link in geoffory's post. I am considering different ways to access the TX/RX pins; the reset I believe is already accessible with the push button. The FFC photo is more explicit in that it labels the RX trace so accessing the TX trace remains. I may explore drilling first.

rpavlik, 2021-01-04:
oh interesting! Please do update us if you can do a low-impact mod to get it working. I tried a different approach with the dimmer switches I also got from Costco: they're definitely TYWE2S, and I can even get to a serial header, but GPIO0 is hidden by the PCB, so couldn't reflash manually without desoldering the module. I did that to one, but dumped the original firmware first (used tasmotizer) and have poked at it in Ghidra to see if I could find a way to get into bootloader/OTA using serial but without IO0. Unfortunately not having a ton of luck.

aargvark, 2021-12-10:
Any updates on this?  I snagged a 3 pack for ~$12 and would love to get them off the cloud.  I was going to tear one apart in hopes of doing the hardware flash, but they are glued really well!  The wifi based tuya-convert is not working (in either the fast blink or slow-blink AP mode).