ct-Open-Source/tuya-convert #850: Unable to Flash KMC 4 Outlet Wifi

kdvlr, 2020-12-07:
According to the Wiki and Issue #91 mentioned here, the KMC 4 outlet should work with tuya-convert. In my case I have two and I see that it is connecting to the Wifi and downloading upgrade.bin but nothing happens after that. 

Any pointers would be much appreciated. 

Log Files attached

[smarthack-psk.log](https://github.com/ct-Open-Source/tuya-convert/files/5654424/smarthack-psk.log)
[smarthack-udp.log](https://github.com/ct-Open-Source/tuya-convert/files/5654425/smarthack-udp.log)
[smarthack-web.log](https://github.com/ct-Open-Source/tuya-convert/files/5654426/smarthack-web.log)
[smarthack-wifi.log](https://github.com/ct-Open-Source/tuya-convert/files/5654427/smarthack-wifi.log)
[smarthack-mqtt.log](https://github.com/ct-Open-Source/tuya-convert/files/5654428/smarthack-mqtt.log)

n8henrie, 2021-01-23:
@kdvlr Did you figure this out?

I just tried and 2/2 of these KMC 4 outlet plugs seemed like they worked (no errors with the flash) but are now nonfunctional.

kdvlr, 2021-01-23:
It was an issue with my Python install.. can you paste your logs ?

n8henrie, 2021-01-23:
Thanks for quick response.

I've done several flashes from this device (a dedicated Pi) without issues.

[device-info.txt](https://github.com/ct-Open-Source/tuya-convert/files/5860503/device-info.txt)
[smarthack-mqtt.log](https://github.com/ct-Open-Source/tuya-convert/files/5860504/smarthack-mqtt.log)
[smarthack-psk.log](https://github.com/ct-Open-Source/tuya-convert/files/5860505/smarthack-psk.log)
[smarthack-udp.log](https://github.com/ct-Open-Source/tuya-convert/files/5860506/smarthack-udp.log)
[smarthack-web.log](https://github.com/ct-Open-Source/tuya-convert/files/5860507/smarthack-web.log)
[smarthack-wifi.log](https://github.com/ct-Open-Source/tuya-convert/files/5860508/smarthack-wifi.log)

n8henrie, 2021-01-23:
It has a [TYWE3S](https://tasmota.github.io/docs/devices/TYWE3S/) inside.

I've wired it up and flashed tasmota directly with esptool -- it seems to flash without error, but the Tasmota AP never shows up after reboot. ¯\_(ツ)_/¯

n8henrie, 2021-01-25:
After a little fiddling, I was able to get the device to flash without error and some serial log output that made it look like things were starting. Unfortunately, it then starts toggling and flashing at 1 second intervals which is apparently [a sign of an error during flashing](https://tasmota.github.io/docs/FAQ/#relay-clicks-and-led-flashes-at-1-second-intervals).

I've erased and reflashed multiple times with no errors and this same result. I've also tried tasmota-lite just to see if it happened to be a size issue (though it should be 1M flash size).

I don't see any other MCUs on here that have continuity with the Rx and Tx lines (<https://tasmota.github.io/docs/TuyaMCU-Devices/>).

I tried going back to the backed up firmware but it didn't seem to like that either; hopefully I haven't just goofed something up soldering on the leads.

Any suggestions would be greatly appreciated!

```console
$ esptool.py write_flash -fs 1MB -fm dout 0x0 tasmota.bin
esptool.py v3.0
Found 3 serial ports
Serial port /dev/cu.usbserial-A50285BI
Connecting....
Detecting chip type... ESP8266
Chip is ESP8266EX
Features: WiFi
Crystal is 26MHz
MAC: 98:f4:ab:c2:42:d5
Uploading stub...
Running stub...
Stub running...
Configuring flash size...
Compressed 600960 bytes to 428969...
Wrote 600960 bytes (428969 compressed) at 0x00000000 in 37.9 seconds (effective 126.7 kbit/s)...
Hash of data verified.

Leaving...
Hard resetting via RTS pin...
```

```
00:00:00 CFG: Loaded from flash at F9, Count 3
00:00:00 QPC: Count 3
00:00:00 Project tasmota Tasmota Version 9.2.0(tasmota)-2_7_4_9(2020-12-21T15:03:40)
00:00:00 WIF: WifiManager active for 3 minutes
00:00:00 HTP: Web server active on tasmota_C242D5-0725 with IP address 192.168.4.1
00:00:06 QPC: Reset
```

[image]

kdvlr, 2021-01-26:
I think you have tried more than I did. I don’t have any more suggestions. Maybe someone else here can help

kueblc, 2021-01-26:
@n8henrie as far as the backup goes:

When you first restore the backup made by tuya-convert, it will be running tuya-convert's intermediate firmware. This firmware is responsible for making the backup, reformatting the flash, and downloading the new firmware. To return to stock firmware you will need to issue the undo command. You can do this through tuya-convert's `start_flash` or alternatively by connecting to `vtrust-recovery` and `curl` or browse to http://10.42.42.42/undo.

n8henrie, 2021-02-10:
Had some time to tinker with this again today, unfortunately not much luck.

Using the wired setup, I can flash tasmota-lite or tasmota -- no luck with either of them. They all seem to flash okay, but either nothing happens afterward or I get the "clicking on and off every second." Doesn't seem to matter if I leave that little header (top of the picture) plugged in or unplugged, if I plug AC power in or leave it out. Making sure to `erase_flash` in between attempts.

With the wired setup I was able to re-flash the backup firmware. Tuya-Convert then detects the intermediate firmware and lets me retry the process, but the results are the same as when I flash wired -- either nothing or clicking.

Thanks to @kueblc I was able to successfully get the stock firmware running again, so at least I haven't let the magic smoke out of anything yet.

I'm outta ideas.
¯\\_(ツ)_/¯

chaseadam, 2021-02-26:
I was successful flashing the KMC units purchased via https://www.amazon.com/gp/product/B07SWZ2N4J/

There was one unit which didn't download properly, but I power cycled it and the conversion continued.

n8henrie, 2021-02-26:
@chaseadam i was just hoping someone else would chime in.

Did you have to do anything unusual or open the units?

How many did you flash?

I was 0 for 2. Have done 8 other devices without issue, so I thought maybe it was just this switch. But if both of you have flashed without issues I must be doing something wrong.

chaseadam, 2021-02-26:
flashed 6 total. 2 packs labeled with:
30401WB
X0027LELDX

Units have markings in plastic mold for:
model: 30608
manufacture: 04/20

Pics:
[image]
[image]

n8henrie, 2021-02-26:
Thanks.

Did you have to open them (other than for pictures)? Or were you able to do
it wirelessly?

chaseadam, 2021-02-26:
> Thanks. Did you have to open them (other than for pictures)? Or were you able to do it wirelessly?

No need to open, I was just curious. Happy with the build quality for the price. No UL rating on it, but at least it has >15A relays.

NickJLange, 2021-02-27:
Alas, I purchased from Amazon and no longer flashable- using the new firmware.

chaseadam, 2021-02-28:
> Alas, I purchased from Amazon yesterday and no longer flashable- using the new firmware.

Can you share the manufacture month in the plastic on the back. There may be come correlation with incompatible firmware which may provide others with an explanation if they run into difficulty.

Aside: the date is likely specific to the plastic injection process which may not give us useful information as to when the electronics where flashed (i.e. they may have done a big run in 04/2020 and be using up stock over time resulting it various firmware versions)

n8henrie, 2021-02-28:
Mine also say 30608, 4/20. Purchased from Amazon.

The conversion seems like it goes fine, just doesn't work afterwards.

chaseadam, 2021-03-02:
That was on the box containing the two units.

Are you selecting "tasmota.bin" to flash? Is it the one in this repo?

I see you mentioned the "clicking on and off every second". That actually means **the flashing worked** (assuming you selected tasmota.bin). The template hasn't yet been applied, so it doesn't know what is the LED or the relays. Turns out the default GPIO for the LED in tasmota.bin. You should be able to login to the device via its `tasmota` containing SSID. If you get the credentials wrong here (before applying a template) you will be stuck, so get it right.

n8henrie, 2021-03-04:
> If it is only the middle relay, the flashing worked 

Doh! I had seen [this](https://tasmota.github.io/docs/FAQ/#relay-clicks-and-led-flashes-at-1-second-intervals) and thought it was clearly not working!

> Relay clicks and LED flashes at 1 second intervals~
>
> This indicates that your device did not get flashed properly. In this case it will toggle all it's pins at 1 sec intervals. A flash erase and a new flash is required.

Sure enough, once I looked under hidden / unrecognized networks, there it was. 🤦‍♂️

A few upgrades later and now I'm happily running 9.3.1 hooked up via mqtt to HomeAssistant. Woot!

Unfortunately, in my excitement I clumsily let the smoke out of 1 of 2 devices (which I had taken apart for the serial connections). Oh well, at least one is up and running, and if these go on sale again maybe I'll grab a few more.

@kdvlr you can close the issue. Thanks to all for the help, what a great community!

hubmon, 2021-03-04:
> @n8henrie as far as the backup goes:
>
> When you first restore the backup made by tuya-convert, it will be running tuya-convert's intermediate firmware. This firmware is responsible for making the [...]

I can login to vtrust-recovery when the KMC4 is on(clicking mode) but cannot get to http://10.42.42.42/undo. Device was not opened
Any additional guidance will be appreciated.

hubmon, 2021-03-04:
No Tasmota WiFi AP.  I have 2 KMC4 devices with identical problem. 
Possible to recover without soldering?

Previously successfully flashed 3 Teckin devices using vmware.

hubmon, 2021-03-07:
Got it going. Did a Tasmota Reset.  When it started the 7th time the Tasmota AP was available. Login & configured.

NickJLange, 2021-07-04:
I'm going to necrobump this issue to point out that for whatever reason, my initial failure was mine own. The device flashes correctly with tuya-convert.

n3gwg, 2023-12-17:
Users, developers, et alia:

I'm considering to obtain some of these devices. I am a bit confused by some of the comments here so let me ask some questions such that someone might be able to disambiguate.

1) Can these devices be Tasmotized via Wi-Fi?

2) Is it the case that only specific models and manufacture dates work?

3) Is there any pre-flashing advice that I ought to be aware of in precedence to ordering and attempting to place Tasmota firmware on one of these devices?

Thank you in advance for your time and consideration regarding the issues before us in the instant case.

Stuart

n8henrie, 2023-12-17:
> Can these devices be Tasmotized via Wi-Fi?

I had to open mine as per above

> Is it the case that only specific models and manufacture dates work?

I couldn't tell you, I only own one

> Is there any pre-flashing advice that I ought to be aware of in precedence to ordering and attempting to place Tasmota firmware on one of these devices?

AFAIK you'll need to be comfortable soldering onto the pads as I did above. Once flashed, the device has worked beautifully for me for years.

manuel-arguelles, 2024-04-15:
Hi, sorry to revisit this, but I'm having the same issue as @kdvlr, or so it seems, while using tuya-convert I see the GET of upgrade.bin in the web logs, but after that nothing else. What was the python installation issue? I'm running it on a Pi 3B.

I also opened it and as @n8henrie described, it has a TYW3S, but mine has a big black cable over it with little room to move it to solder wires to it.

Could someone confirm if tuya-convert works over wifi for it? or is soldering the way to go?

Right now it just blinks fast 5 times, then pause for a bit and repeats... If I hold the power button down for like 10 seconds or so it starts to blink continuously, but there's no tasmota AP, I tried reset (power cycle 7 times) but did nothing...

This are my logs:

[smarthack-mqtt.log](https://github.com/ct-Open-Source/tuya-convert/files/14972973/smarthack-mqtt.log)
[smarthack-psk.log](https://github.com/ct-Open-Source/tuya-convert/files/14972974/smarthack-psk.log)
[smarthack-udp.log](https://github.com/ct-Open-Source/tuya-convert/files/14972975/smarthack-udp.log)
[smarthack-web.log](https://github.com/ct-Open-Source/tuya-convert/files/14972976/smarthack-web.log)
[smarthack-wifi.log](https://github.com/ct-Open-Source/tuya-convert/files/14972977/smarthack-wifi.log)

manuel-arguelles, 2024-04-15:
Managed to flash it using tuya-convert, but with the development branch and with this patch: https://github.com/ct-Open-Source/tuya-convert/pull/1128