ct-Open-Source/tuya-convert #851: Device will not flash WS01

alexyao2015, 2020-12-08:
### Before opening a new issue, please read the following

**Are you reporting a successful flash?**
_Please add it to the wiki instead!_

**Do you have a question about configuring your newly flashed device?**
_Please ask in your alternative firmware's support channel instead!_

**Did you have issues getting a device to flash?**
_Please be sure to include your device **name, model, and firmware version** as well as your **logs**!_
I'm trying to flash an Esicoo WS01 device which I was able to successfully flash 1 month ago on a raspberry pi. This time, I'm trying to use a laptop with new plugs that I just purchased. The FCC ID is the same as the last one so I can only assume that it still uses an ESP.

From the logs, it seems like the device will not connect at all to the ap. Not sure if it is because the firmware is patched or if its because my setup on my laptop isn't correct.
My phone connects just fine to the ap.

[smarthack-mqtt.log](https://github.com/ct-Open-Source/tuya-convert/files/5662624/smarthack-mqtt.log)
[smarthack-wifi.log](https://github.com/ct-Open-Source/tuya-convert/files/5662625/smarthack-wifi.log)
[smarthack-udp.log](https://github.com/ct-Open-Source/tuya-convert/files/5662626/smarthack-udp.log)
[smarthack-psk.log](https://github.com/ct-Open-Source/tuya-convert/files/5662627/smarthack-psk.log)
[smarthack-web.log](https://github.com/ct-Open-Source/tuya-convert/files/5662628/smarthack-web.log)

**Please remember to close your issue when the problem has been addressed!**

alexyao2015, 2020-12-09:
It appears that the device has been changed to no longer be Tuya, but instead uses the cloud intelligence app.

kueblc, 2020-12-09:
> The FCC ID is the same as the last one so I can only assume that it still uses an ESP.

Unfortunately this is not a safe assumption. There have been many reported cases of willful FCC violations by these companies.

> It appears that the device has been changed to no longer be Tuya, but instead uses the cloud intelligence app.

Too bad, hopefully you can still find ESP based devices in your area. Best of luck.

closed 2020-12-09 by kueblc