ct-Open-Source/tuya-convert #853: Amnesia:33?

untergeek, 2020-12-09:
This is hypothetical, of course, and I don't know whether Tuya uses any of the [vulnerable TCP stacks](https://www.forescout.com/research-labs/amnesia33/). However, if they are, and it's one of the remote code execution exploitable stacks, is there a chance to use that to get Tuya-convert up and running again?

untergeek, 2020-12-09:
The white paper outlining all of this can be found [here](https://www.forescout.com/company/resources/amnesia33-how-tcp-ip-stacks-breed-critical-vulnerabilities-in-iot-ot-and-it-devices/).

fzimmermann89, 2020-12-10:
most likely: no.

The espressif sdk uses lwip.

 a quick look in one of the psk-id 02 firmware shows the following strings
`tcp_in.c
tcpip.c
tcp.c
tcp_out.c`, matching lwip as well. 

they did not find any vulnerability in lwip.

kueblc, 2020-12-10:
Indeed @fzimmermann89 is correct, the Tuya SDK (built on top of the Espressif SDK) uses LWIP.

Interesting read nonetheless. Thanks for sharing @untergeek!

closed 2020-12-10 by kueblc