ct-Open-Source/tuya-convert #855: Issues to flash a smartbulb

titiviking, 2020-12-11:
[smarthack-mqtt.log](https://github.com/ct-Open-Source/tuya-convert/files/5679146/smarthack-mqtt.log)
[smarthack-psk.log](https://github.com/ct-Open-Source/tuya-convert/files/5679147/smarthack-psk.log)
[smarthack-udp.log](https://github.com/ct-Open-Source/tuya-convert/files/5679148/smarthack-udp.log)
[smarthack-web.log](https://github.com/ct-Open-Source/tuya-convert/files/5679149/smarthack-web.log)
[smarthack-wifi.log](https://github.com/ct-Open-Source/tuya-convert/files/5679150/smarthack-wifi.log)

Hi there,

I'm attempting to flash my lightbulb, but it seems to be failing...

`pi@raspberrypi:~/tuya-convert $ ./start_flash.sh
tuya-convert v2.4.4
Checking for network interface wlan0... Found.
Checking UDP port 53... Available.
Checking UDP port 67... Available.
Checking TCP port 80... Available.
Checking TCP port 443... Available.
Checking UDP port 6666... Available.
Checking UDP port 6667... Available.
Checking TCP port 1883... Available.
Checking TCP port 8886... Available.
======================================================
  Starting AP in a screen
  Starting web server in a screen
  Starting Mosquitto in a screen
  Starting PSK frontend in a screen
  Starting Tuya Discovery in a screen

======================================================

IMPORTANT
1. Connect any other device (a smartphone or something) to the WIFI vtrust-flash
   This step is IMPORTANT otherwise the smartconfig may not work!
2. Put your IoT device in autoconfig/smartconfig/pairing mode (LED will blink fast). This is usually done by pressing and holding the primary button of the device
   Make sure nothing else is plugged into your IoT device while attempting to flash.
3. Press ENTER to continue

======================================================
Starting smart config pairing procedure
Waiting for the device to install the intermediate firmware
Put device in EZ config mode (blinking fast)
Sending SSID                  vtrust-flash
Sending wifiPassword
Sending token                 00000000
Sending secret                0101
................
SmartConfig complete.
Resending SmartConfig Packets
..............................................................................................................
Device did not appear with the intermediate firmware
Check the *.log files in the scripts folder
Do you want to try flashing another device? [y/N] n
======================================================
Cleaning up...
Closing AP
Exiting...
`

Brand:  LSC
Model number:  970720
[image]

I've tried multiple time in a row as I read in other issues, but didn't change anything.
Could you help me out on this one?
Many thanks.

malosaa, 2020-12-16:
i did just try exaclty the same and if i put the bulb in rapid flash and run tuya-convert it goes off and nothing happens

malosaa, 2020-12-16:
the rgb is not esp based.. 
when i check my log it says that (the udp one )

malosaa, 2020-12-16:
this is what i get with that same bulb 

WARNING: it appears this device does not use an ESP82xx and therefore cannot install ESP based firmware

tarontop, 2021-02-23:
There are pre-installed Tasmota bulbs here
https://templates.blakadder.com/preflashed.html