ct-Open-Source/tuya-convert #860: Unable to flash lonsonho x801a-l

benlad, 2020-12-12:
Unable to flash light switch: lonsonho x801a-l 
See attached logs ann device image
[smarthack-psk.log](https://github.com/ct-Open-Source/tuya-convert/files/5683442/smarthack-psk.log)
[smarthack-udp.log](https://github.com/ct-Open-Source/tuya-convert/files/5683443/smarthack-udp.log)
[smarthack-web.log](https://github.com/ct-Open-Source/tuya-convert/files/5683444/smarthack-web.log)
[smarthack-wifi.log](https://github.com/ct-Open-Source/tuya-convert/files/5683445/smarthack-wifi.log)
[smarthack-mqtt.log](https://github.com/ct-Open-Source/tuya-convert/files/5683446/smarthack-mqtt.log)
[image]

kueblc, 2020-12-13:
Unsupported firmware, see pinned issue #483 and the wiki for more information.

closed 2020-12-13 by kueblc

benlad, 2020-12-13:
Thanks, I added this device to the _Known Affected Devices_ on the [wiki](https://github.com/ct-Open-Source/tuya-convert/wiki/Collaboration-document-for-PSK-Identity-02).