ct-Open-Source/tuya-convert #862: TYWE2S without GPIO0

fragsalat, 2020-12-13:
Hey there,

not sure though this is a proper place for this but I bought a DELOCK 11827 and the tasmota update corrupted the whole socket. So I wanted to flash it via serial and opened the socket and found out the chip used there is a TYWE2S. The datasheet mentiones I have to pull down GPIO0 to get it into boot mode but my chip actually doesn't have these GPIO0 pin. Does anyone have an idea what I can do to flash it?

Screenshot from the datasheet clearly shows IO0 pin.
[image]

Mine doesn't have it o.O
[image]
[image]

Thanks upfront for help :)

fzimmermann89, 2020-12-14:
Interesting..

You could look for the datasheet for the esp chip on the board, find which pin is gpio0 (might be pin 15, on the bottom 2nd pin from the right) and look around with a multimeter if you find gpio0 somewhere easy to access on the pcb

If not, you could try to connect there directly to the ic pad. Either solder on a thin non enameled copper wire or use a thin multimeter probe to temporarily pull it to GND during boot...

alexyao2015, 2020-12-14:
If you have very steady hands, you could solder directly onto the esp itself. Check out the pin definitions [here](https://www.espressif.com/sites/default/files/documentation/0a-esp8266ex_datasheet_en.pdf) on page 9.

I had a device where the backside of the board was completely covered making it impossible to flash normally.

fhloston, 2020-12-14:
I converted 4 bulbs that way. Had help from my daughter, who managed to shortly touch GPIO0 to ground upon boot. No need to solder...

alexyao2015, 2020-12-14:
Did you solder the other pins to the pads? Or how did you connect to the serial pins?

fragsalat, 2020-12-14:
Thanks for the help. There seem to be really no IO0 pin but pulling the 2nd pin from right on the bottom down worked and I were able to flash tasmota with tasmotize 👍

closed 2020-12-14 by fragsalat

DunkinDonat, 2021-01-07:
Hi:) I am also trying to flash this chip. https://tasmota.github.io/docs/devices/CE-Smart-Home---LA-WF3-Wifi-Plug/ Here you can find a picture of how to wire the cables.