ct-Open-Source/tuya-convert #863: Costco/Feit Electric Bulbs with new hardware?

mew1033, 2020-12-14:
I just got a few of the Feit Electric bulbs from Costco. Here's what they look like on the inside:
[image]
[image]
[image]

Any ideas what that board is and if/how to flash it? I tried tuya-convert and got the 

    smartconfig complete.
    resending smartconfig packets

message over and over before it failed.

mew1033, 2020-12-14:
Ah, just found it. Looks like it's one of Tuya's proprietary chips. https://github.com/ct-Open-Source/tuya-convert/issues/484

mew1033, 2020-12-14:
Looks almost exactly like this: https://github.com/ct-Open-Source/tuya-convert/issues/777

closed 2020-12-14 by mew1033

mikesbytes, 2020-12-26:
> Ah, just found it. Looks like it's one of Tuya's proprietary chips. #484

Are you sure about this? I just got some of these bulbs and am having the same issue, but it looks like it should be reported if it's not actually using ESP. Could you share the exact model number and FCC ID off the side of the bulb? This could be #483 related maybe.

deznamenosnove, 2023-07-19:
did you manage to flash that bulb with any firmware? the i2go brand in Brazil is selling one almost identical to yours and i'm trying to figure out what to do