ct-Open-Source/tuya-convert #866: BHT-002GBLW Series 3/16A- Device did not appear with the intermediate firmware

gamgamstylep, 2020-12-16:
BHT-002 GB - Device did not appear with the intermediate firmware. Pressed Temp down 8 Seconds. The device went in the config mode. But it it not working.

[smarthack-wifi.log](https://github.com/ct-Open-Source/tuya-convert/files/5704086/smarthack-wifi.log)
[smarthack-psk.log](https://github.com/ct-Open-Source/tuya-convert/files/5704087/smarthack-psk.log)

Already flash successfully Gosund SP1 with this raspberry. 

tuya-convert v2.4.4
raspberrypi 2

root@raspberrypi:/home/pi/tuya-convert# ./start_flash.sh
tuya-convert v2.4.4
Checking for network interface wlan0... Found.
Checking UDP port 53... Available.
Checking UDP port 67... Available.
Checking TCP port 80... Available.
Checking TCP port 443... Available.
Checking UDP port 6666... Available.
Checking UDP port 6667... Available.
Checking TCP port 1883... Available.
Checking TCP port 8886... Available.
======================================================
  Starting AP in a screen
  Starting web server in a screen
  Starting Mosquitto in a screen
  Starting PSK frontend in a screen
  Starting Tuya Discovery in a screen

======================================================

IMPORTANT
1. Connect any other device (a smartphone or something) to the WIFI vtrust-flash
   This step is IMPORTANT otherwise the smartconfig may not work!
2. Put your IoT device in autoconfig/smartconfig/pairing mode (LED will blink fast). This is usually done by pressing and holding the primary button of the device
   Make sure nothing else is plugged into your IoT device while attempting to flash.
3. Press ENTER to continue

======================================================
Starting smart config pairing procedure
Waiting for the device to install the intermediate firmware
.Put device in EZ config mode (blinking fast)
Sending SSID                  vtrust-flash
Sending wifiPassword
Sending token                 00000000
Sending secret                0101
................
SmartConfig complete.
Resending SmartConfig Packets
.................
SmartConfig complete.
Resending SmartConfig Packets
................
SmartConfig complete.
Resending SmartConfig Packets
................
SmartConfig complete.
Resending SmartConfig Packets
................
SmartConfig complete.
Resending SmartConfig Packets
.................
SmartConfig complete.
Resending SmartConfig Packets
.................
SmartConfig complete.
Resending SmartConfig Packets
.................
SmartConfig complete.
Resending SmartConfig Packets
.................
SmartConfig complete.
Resending SmartConfig Packets
................
SmartConfig complete.
Resending SmartConfig Packets
...............
Device did not appear with the intermediate firmware
Check the *.log files in the scripts folder
Do you want to try flashing another device? [y/N] N

devzwf, 2020-12-16:
I'm not the maintainer or related to this repo but you're probably affected by https://github.com/ct-Open-Source/tuya-convert/wiki/Collaboration-document-for-PSK-Identity-02 - of which there is no solution yet except for hardware level flashing.

kueblc, 2020-12-18:
@devzwf is correct, in this case the firmware is unfortunately unsupported due to a patch by Tuya. Follow #483 to be notified of any news on this front.

closed 2020-12-18 by kueblc