ct-Open-Source/tuya-convert #871: Can't Flash Aoycocr X5P Plug

spikex, 2020-12-19:
Purchased from Amazon 12/2020. Previously known to work per the Wiki.

The plug requests the token but never requests the firmware. This looks like the 02 issue, the LED stops blinking quickly, but the ID starts with 01 and there's no PSK error. I've tried the master and development branches.

The hardware could have changed, there's no nondestructive way to open the plugs, but the MAC address is right for an ESP8266 and the user agent reports SP8266SDK.

Any thoughts on what to try?

[smarthack-wifi.log](https://github.com/ct-Open-Source/tuya-convert/files/5719978/smarthack-wifi.log)
[smarthack-web.log](https://github.com/ct-Open-Source/tuya-convert/files/5719979/smarthack-web.log)
[smarthack-udp.log](https://github.com/ct-Open-Source/tuya-convert/files/5719980/smarthack-udp.log)
[smarthack-psk.log](https://github.com/ct-Open-Source/tuya-convert/files/5719981/smarthack-psk.log)
[smarthack-mqtt.log](https://github.com/ct-Open-Source/tuya-convert/files/5719982/smarthack-mqtt.log)

edwardhughes, 2020-12-21:
I can confirm this issue exactly as described by OP.  I received a four pack of these switches from Amazon today (12/20).  I am using the same hardware that was used in December 2019 to flash these same plugs, also from Amazon.

colinodell, 2020-12-25:
I was unable to flash the same device purchased from Amazon this month.  Log files from a fresh attempt via docker-compose: https://gist.github.com/colinodell/1578307a6fb67390bd230f7e3384cc5b

colinodell, 2020-12-26:
Although I was not able to get it working via docker-compose on my laptop, I was able to flash it successfully using a Raspberry Pi 4.  It's entirely possible the issue was something with my laptop's configuration.

kueblc, 2020-12-27:
@colinodell this was your issue:
https://gist.github.com/colinodell/1578307a6fb67390bd230f7e3384cc5b#file-smarthack-web-log
```
Could not start server on port 80
Close the process on this port and try again
```