ct-Open-Source/tuya-convert #875: Tuya water leak sensor

gabeale, 2020-12-20:
### Before opening a new issue, please read the following

**Are you reporting a successful flash?**
No

**Do you have a question about configuring your newly flashed device?**
No

**Did you have issues getting a device to flash?**
Yes

**Please remember to close your issue when the problem has been addressed!**

kueblc, 2020-12-20:
I can not help if you do not provide information.

gabeale, 2020-12-20:
Hello everybody!
I'm having issues in flashing Tuya water leak sensor based on TYWE3S chip through Tuya-converter (OTA).
Log files are the following:

**smarthack-psk.log**

new client on port 443 from 10.42.42.31:63396
could not establish sslpsk socket: [SSL: HTTPS_PROXY_REQUEST] https proxy request (_ssl.c:1056)
new client on port 443 from 10.42.42.31:63401
could not establish sslpsk socket: [SSL: HTTPS_PROXY_REQUEST] https proxy request (_ssl.c:1056)
new client on port 443 from 10.42.42.10:29845
ID: 0242416f68626d6436614739314946523189171e11c10b33ac675ac84173c862e8cf2101eb31025a140ed4b3190120fea2
PSK: a78407407c485620c80ee44da304ba68ee026b43eb2598cbb38de60c995513b0
could not establish sslpsk socket: [SSL: DECRYPTION_FAILED_OR_BAD_RECORD_MAC] decryption failed or bad record mac (_ssl.c:1056)
new client on port 443 from 10.42.42.10:121
ID: 0242416f68626d6436614739314946523189171e11c10b33ac675ac84173c862e8cf2101eb31025a140ed4b3190120fea2
PSK: a78407407c485620c80ee44da304ba68ee026b43eb2598cbb38de60c995513b0
could not establish sslpsk socket: [SSL: DECRYPTION_FAILED_OR_BAD_RECORD_MAC] decryption failed or bad record mac (_ssl.c:1056)
new client on port 443 from 10.42.42.10:7455
ID: 0242416f68626d6436614739314946523189171e11c10b33ac675ac84173c862e8cf2101eb31025a140ed4b3190120fea2
PSK: a78407407c485620c80ee44da304ba68ee026b43eb2598cbb38de60c995513b0
could not establish sslpsk socket: [SSL: DECRYPTION_FAILED_OR_BAD_RECORD_MAC] decryption failed or bad record mac (_ssl.c:1056)
new client on port 443 from 10.42.42.10:43427
ID: 0242416f68626d6436614739314946523189171e11c10b33ac675ac84173c862e8cf2101eb31025a140ed4b3190120fea2
PSK: a78407407c485620c80ee44da304ba68ee026b43eb2598cbb38de60c995513b0
could not establish sslpsk socket: [SSL: DECRYPTION_FAILED_OR_BAD_RECORD_MAC] decryption failed or bad record mac (_ssl.c:1056)
new client on port 443 from 10.42.42.10:42231
ID: 0242416f68626d6436614739314946523189171e11c10b33ac675ac84173c862e8cf2101eb31025a140ed4b3190120fea2
PSK: a78407407c485620c80ee44da304ba68ee026b43eb2598cbb38de60c995513b0
could not establish sslpsk socket: [SSL: DECRYPTION_FAILED_OR_BAD_RECORD_MAC] decryption failed or bad record mac (_ssl.c:1056)
^CTraceback (most recent call last):
  File "./psk-frontend.py", line 113, in <module>
    main()
  File "./psk-frontend.py", line 106, in main
    r,_,_ =  select.select(readables, [], [])
KeyboardInterrupt

**smarthack-wifi.log**

===========
Configuring AP interface...
RTNETLINK answers: File exists
RTNETLINK answers: File exists
Starting DNSMASQ server...
Starting AP on wlan0...
Configuration file: /dev/stdin
wlan0: Could not connect to kernel driver
Using interface wlan0 with hwaddr dc:a6:32:cd:3e:00 and ssid "vtrust-flash"
wlan0: interface state UNINITIALIZED->ENABLED
wlan0: AP-ENABLED
wlan0: AP-STA-CONNECTED 5e:02:91:09:89:d2
wlan0: AP-STA-CONNECTED 10:52:1c:f8:21:f4
wlan0: AP-STA-DISCONNECTED 10:52:1c:f8:21:f4
wlan0: AP-STA-CONNECTED 10:52:1c:f8:21:f4
wlan0: AP-STA-DISCONNECTED 10:52:1c:f8:21:f4
wlan0: AP-STA-CONNECTED 10:52:1c:f8:21:f4
wlan0: interface state ENABLED->DISABLED
wlan0: AP-STA-DISCONNECTED 10:52:1c:f8:21:f4
wlan0: AP-STA-DISCONNECTED 5e:02:91:09:89:d2
wlan0: AP-DISABLED
wlan0: CTRL-EVENT-TERMINATING
nl80211: deinit ifname=wlan0 disabled_11b_rates=0
AP closed
Stopping DNSMASQ server...

**smarthack-mqtt.log**

1608479289: mosquitto version 1.5.7 starting
1608479289: Using default config.
1608479289: Opening ipv4 listen socket on port 1883.
1608479289: Opening ipv6 listen socket on port 1883.
^C1608479465: mosquitto version 1.5.7 terminating

**smarthack-udp.log**

Listening for encrypted Tuya broadcast on UDP 6667
^CListening for Tuya broadcast on UDP 6666
Listening for encrypted Tuya broadcast on UDP 6667
^CListening for Tuya broadcast on UDP 6666
Listening for encrypted Tuya broadcast on UDP 6667
^CListening for Tuya broadcast on UDP 6666
Listening for encrypted Tuya broadcast on UDP 6667
^CListening for Tuya broadcast on UDP 6666
Listening for encrypted Tuya broadcast on UDP 6667
^C

**smarthack-web.log**

Listening on 10.42.42.1:80
^[[32m[I 201220 16:49:25 web:2243]^[[m^O 302 GET /RZBkMRD6Mwsg/roadw.pac (10.42.42.31) 5.53ms
^[[32m[I 201220 16:49:25 web:2243]^[[m^O 200 GET / (10.42.42.31) 5.18ms
^[[33m[W 201220 16:49:25 web:2243]^[[m^O 405 CONNECT www.zscaler.com:80 (10.42.42.31) 6.89ms
^[[32m[I 201220 16:49:26 web:2243]^[[m^O 200 GET / (10.42.42.31) 1.05ms
^[[33m[W 201220 16:49:44 web:2243]^[[m^O 405 CONNECT www.zscaler.com:80 (10.42.42.31) 1.35ms
^[[32m[I 201220 16:50:41 web:2243]^[[m^O 302 GET /RZBkMRD6Mwsg/roadw.pac (10.42.42.31) 1.07ms
^[[32m[I 201220 16:50:41 web:2243]^[[m^O 200 GET / (10.42.42.31) 1.13ms
^[[33m[W 201220 16:50:41 web:2243]^[[m^O 405 CONNECT www.zscaler.com:80 (10.42.42.31) 1.17ms
^[[33m[W 201220 16:50:44 web:2243]^[[m^O 405 CONNECT www.zscaler.com:80 (10.42.42.31) 1.28ms
^[[32m[I 201220 16:50:45 web:2243]^[[m^O 302 GET /RZBkMRD6Mwsg/roadw.pac (10.42.42.31) 1.39ms
^[[32m[I 201220 16:50:45 web:2243]^[[m^O 200 GET / (10.42.42.31) 1.41ms
^[[33m[W 201220 16:50:45 web:2243]^[[m^O 405 CONNECT www.zscaler.com:80 (10.42.42.31) 1.27ms
^CReceived SIGINT, exiting...

Thank you in advance!

kueblc, 2020-12-20:
Unsupported, please see pinned issue #483 and the wiki for more information.

closed 2020-12-20 by kueblc