ct-Open-Source/tuya-convert #877: Light globe flashing issue

lilfellabob, 2020-12-21:
Hi,

I bought a 'Brilliant smart' globe (model 20699) from Bunnings (in Australia) and am trying to flash tasmota. I have sucessfully flashed a mirabella genio bulb previously using the same raspberry pi 3 but this one does not work. The bulb flashes rapidly and when the script starts, the bulb goes to solid on very shortly after the script outputs 'Sending secret'. The script then outputs:
`'SmartConfig complete.
Resending SmartConfig Packets'`
10 times before saying 'Device did not appear with the intermediate firmware'.

I've noticed in the smarthack-psk log the top line is 'Cannot exec './psk-frontend.py': Permission denied' and I have tried to fix the permissions but I don't know if this is the problem or if my attempt did anything.

Thanks,
[smarthack-psk.log](https://github.com/ct-Open-Source/tuya-convert/files/5722101/smarthack-psk.log)
[smarthack-udp.log](https://github.com/ct-Open-Source/tuya-convert/files/5722102/smarthack-udp.log)
[smarthack-web.log](https://github.com/ct-Open-Source/tuya-convert/files/5722103/smarthack-web.log)
[smarthack-wifi.log](https://github.com/ct-Open-Source/tuya-convert/files/5722104/smarthack-wifi.log)
[smarthack-mqtt.log](https://github.com/ct-Open-Source/tuya-convert/files/5722105/smarthack-mqtt.log)

jmiahman, 2021-01-02:
I am seeing a similar traceback (on another device) but I do not get the exec perm error. My perms on psk-frontend are:
 0775 or -rwxrwxr-x 
 a lot of times if a file is not executable you will see the 'Cannot exec './psk-frontend.py': Permission denied' error. Hope that helps.

tusharraval102, 2021-01-24:
I am also getting the same problem on a Globe bulb that I bought from Costco in Canada.

gullygossner, 2021-03-04:
> I am also getting the same problem on a Globe bulb that I bought from Costco in Canada.

Did you ever find a solution or is it simply a no go at the moment due to the current firmware on the bulb? I would like to get these over to Tasmota if possible. I have tried to open one up to flash on the pins but it is a destructive process.