ct-Open-Source/tuya-convert #878: "any other device" can not connect vtrust-flash SSID.

radiumfu, 2020-12-21:
I keep getting disconnected when I use my iphone as well as another windows 10 laptop connect  "vtrust-flash" after ./start_flash.sh

the start_flash script seems no error, but log shows "Failed to set beacon parameters" very quick right after my iphone connect it.(sometimes my iphone can see redirect to a web page "current connect 
[image]
[image]

I tried on two  pc with same kali Linux 2020.4
pc1. NUC 6i5SYH
pc2.  DELL OptiPlex 7070 micro

below dmesg log got from pc2
[   11.767426] Intel(R) Wireless WiFi driver for Linux
[   11.999587] Bluetooth: hci0: Found device firmware: intel/ibt-17-16-1.sfi
[   12.039671] iwlwifi 0000:00:14.3: firmware: direct-loading firmware iwlwifi-9000-pu-b0-jf-b0-46.ucode
[   12.039683] iwlwifi 0000:00:14.3: WRT: Overriding region id 0
[   12.039684] iwlwifi 0000:00:14.3: WRT: Overriding region id 1
[   12.039685] iwlwifi 0000:00:14.3: WRT: Overriding region id 2
[   12.039685] iwlwifi 0000:00:14.3: WRT: Overriding region id 3
[   12.039686] iwlwifi 0000:00:14.3: WRT: Overriding region id 4
[   12.039686] iwlwifi 0000:00:14.3: WRT: Overriding region id 6
[   12.039686] iwlwifi 0000:00:14.3: WRT: Overriding region id 8
[   12.039687] iwlwifi 0000:00:14.3: WRT: Overriding region id 9
[   12.039687] iwlwifi 0000:00:14.3: WRT: Overriding region id 10
[   12.039688] iwlwifi 0000:00:14.3: WRT: Overriding region id 11
[   12.039688] iwlwifi 0000:00:14.3: WRT: Overriding region id 15
[   12.039689] iwlwifi 0000:00:14.3: WRT: Overriding region id 16
[   12.039689] iwlwifi 0000:00:14.3: WRT: Overriding region id 18
[   12.039690] iwlwifi 0000:00:14.3: WRT: Overriding region id 19
[   12.039690] iwlwifi 0000:00:14.3: WRT: Overriding region id 20
[   12.039691] iwlwifi 0000:00:14.3: WRT: Overriding region id 21
[   12.039692] iwlwifi 0000:00:14.3: Found debug destination: EXTERNAL_DRAM
[   12.039693] iwlwifi 0000:00:14.3: Found debug configuration: 0
[   12.039897] iwlwifi 0000:00:14.3: loaded firmware version 46.8902351f.0 9000-pu-b0-jf-b0-46.ucode op_mode iwlmvm
[   12.042288] iwlwifi 0000:00:14.3: firmware: failed to load iwl-debug-yoyo.bin (-2)
[   12.042289] firmware_class: See https://wiki.debian.org/Firmware for information about missing firmware
[   12.162275] intel_rapl_common: Found RAPL domain package
[   12.162276] intel_rapl_common: Found RAPL domain core
[   12.162277] intel_rapl_common: Found RAPL domain uncore
[   12.162278] intel_rapl_common: Found RAPL domain dram
[   12.177817] iwlwifi 0000:00:14.3: Detected Intel(R) Wireless-AC 9560, REV=0x318
[   12.233719] iwlwifi 0000:00:14.3: base HW address: f8:e4:e3:2d:65:a9
[  358.624648] IPv6: ADDRCONF(NETDEV_CHANGE): wlan0: link becomes ready
[  609.186294] IPv6: ADDRCONF(NETDEV_CHANGE): wlan0: link becomes ready
[  638.310109] IPv6: ADDRCONF(NETDEV_CHANGE): wlan0: link becomes ready
[  697.581454] IPv6: ADDRCONF(NETDEV_CHANGE): wlan0: link becomes ready
[  736.800463] IPv6: ADDRCONF(NETDEV_CHANGE): wlan0: link becomes ready
[  770.134518] IPv6: ADDRCONF(NETDEV_CHANGE): wlan0: link becomes ready
[  837.640501] IPv6: ADDRCONF(NETDEV_CHANGE): wlan0: link becomes ready
[  891.364972] IPv6: ADDRCONF(NETDEV_CHANGE): wlan0: link becomes ready
[  927.456906] IPv6: ADDRCONF(NETDEV_CHANGE): wlan0: link becomes ready
[  971.983671] IPv6: ADDRCONF(NETDEV_CHANGE): wlan0: link becomes ready
[ 1082.921664] IPv6: ADDRCONF(NETDEV_CHANGE): wlan0: link becomes ready
[ 1114.484714] IPv6: ADDRCONF(NETDEV_CHANGE): wlan0: link becomes ready
[ 1254.953442] IPv6: ADDRCONF(NETDEV_CHANGE): wlan0: link becomes ready
[ 1302.586458] IPv6: ADDRCONF(NETDEV_CHANGE): wlan0: link becomes ready
[ 1336.298857] IPv6: ADDRCONF(NETDEV_CHANGE): wlan0: link becomes ready

jmiahman, 2021-01-02:
Maybe firmware issues? https://forums.kali.org/showthread.php?45023-iwlwifi-broken-Kernel-5-2

sominaik, 2021-01-06:
I am having similar issue. When connecting to the vtrust-flash via a phone or laptop, it doesn't connect with a message "couldn't authenticate connection" pls help

MIIB1, 2021-03-12:
> I am having similar issue. When connecting to the vtrust-flash via a phone or laptop, it doesn't connect with a message "couldn't authenticate connection" pls help

same problem

davidbelius, 2023-12-12:
I had also had the issue of not being able to connect to vtrust-flash and seeing "Failed to set beacon parameters". I found a bizarre solution: adding "sleep 10" before hostapd is run in start_ap.sh fixed the problem.

In the end it didn't help me because it turned out that the current version of the device I wanted to flash ([URC-150.app IR remote](https://www.pearl.de/a-NX4519-1254.shtml) from pearl.ch) no longer use ESP, but rather [WB3S](https://zorruno.com/2022/wb3s-replacement-for-home-automation-devices/).