ct-Open-Source/tuya-convert #879: New devices - can not flash with 'tuya-convert'... Any solution...

darekhome, 2020-12-21:
I recently purchased few LED GU10 and E27 bulbs (they are on compatibility list), but they ca not not be flashed with 'tuya-convert'.
In hardware market (where I purchased them) they said that this is 'fresh' delivery from china producer...
So I assume - they have a new firmware, not allowing use the 'tuya-convert'.

Any solution to put a Tasmota firmware on the new devices?

wormiedk, 2020-12-22:
I have the same problem with a WOFEA garage door opener:

`ID: 0242416f68626d64366147393149465231bd52161255c1574a40ae03b9d1ab340e2eb6ae2e1db8bcc948d86ea50e3615d6
PSK: ee37f5e51086213c7049d30f2f811514f7a5c8ad8c255f6de72598f5a36f8f12
could not establish sslpsk socket: [SSL: DECRYPTION_FAILED_OR_BAD_RECORD_MAC] decryption failed or bad record mac (_ssl.c:1056)
new client on port 443 from 10.42.42.30:61293`

kueblc, 2020-12-22:
Duplicate of #483

closed 2020-12-22 by kueblc