ct-Open-Source/tuya-convert #880: Cannot flash Nedis (=Tuya) Smart Plug WIFIP120EWT

walterdevos, 2020-12-21:
I've been on it about 2 hours now. Tried different strategies but I will not flash.

Device is a Nedis, which is a Tuya based device, with the model name/number : WIFI Smart Plug WIFIP120EWT 
SN 010618/1903040004
It uses the NEDIS smart life app  (just Smart life app doesn't work eihter).
With that app , I can find the device and configure it.
says for firmware:
WiFI Module: 1.0.9
MCU Module 1.0.9

When rapid blinking  ( tried any other blinking slow, full on, off.. too)  the device is not connecting to vtrust-flash.

**Noting happens:**
`Starting smart config pairing procedure
Waiting for the device to install the intermediate firmware
Put device in EZ config mode (blinking fast)
Sending SSID                  vtrust-flash
Sending wifiPassword          
Sending token                 00000000
Sending secret                0101
................
SmartConfig complete.
Resending SmartConfig Packets
..................
SmartConfig complete.
Resending SmartConfig Packets
..................
SmartConfig complete.
Resending SmartConfig Packets
..................
SmartConfig complete.
Resending SmartConfig Packets
..................
SmartConfig complete.
Resending SmartConfig Packets
.................
SmartConfig complete.
Resending SmartConfig Packets
..................
SmartConfig complete.
Resending SmartConfig Packets
..................
SmartConfig complete.
Resending SmartConfig Packets
..................
SmartConfig complete.
Resending SmartConfig Packets
..................
SmartConfig complete.
Resending SmartConfig Packets
...
Device did not appear with the intermediate firmware
Check the *.log files in the scripts folder
Do you want to try flashing another device? [y/N] 
`

**LOGS:**
`==> smarthack-wifi.log <==
RTNETLINK answers: File exists
Starting DNSMASQ server...
Starting AP on wlx00e04c02ec14...
Configuration file: /dev/stdin
Using interface wlx00e04c02ec14 with hwaddr 00:e0:4c:02:ec:14 and ssid "vtrust-flash"
wlx00e04c02ec14: interface state UNINITIALIZED->ENABLED
wlx00e04c02ec14: AP-ENABLED 
wlx00e04c02ec14: AP-STA-CONNECTED ea:b5:b9:ec:f1:0f
wlx00e04c02ec14: AP-STA-DISCONNECTED ea:b5:b9:ec:f1:0f
wlx00e04c02ec14: AP-STA-CONNECTED ea:b5:b9:ec:f1:0f

==> smarthack-udp.log <==
Listening for Tuya broadcast on UDP 6666
Listening for encrypted Tuya broadcast on UDP 6667

==> smarthack-psk.log <==
don't panic this is probably just your phone!
new client on port 443 from 10.42.42.13:36646
could not establish sslpsk socket: [SSL: NO_SHARED_CIPHER] no shared cipher (_ssl.c:1123)
don't panic this is probably just your phone!
new client on port 443 from 10.42.42.13:36648
could not establish sslpsk socket: [SSL: NO_SHARED_CIPHER] no shared cipher (_ssl.c:1123)
don't panic this is probably just your phone!
new client on port 443 from 10.42.42.13:36650
could not establish sslpsk socket: [SSL: NO_SHARED_CIPHER] no shared cipher (_ssl.c:1123)
don't panic this is probably just your phone!

==> smarthack-web.log <==
Listening on 10.42.42.1:80

==> smarthack-mqtt.log <==
[ 8443.704122]~DLT~13794~INFO     ~FIFO /tmp/dlt cannot be opened. Retrying later...
1608572212: mosquitto version 1.6.9 starting
1608572212: Using default config.
1608572212: Opening ipv4 listen socket on port 1883.
1608572212: Opening ipv6 listen socket on port 1883.
`

kueblc, 2020-12-21:
It looks like this device is not supported, see [tuya-convert/wiki](https://github.com/ct-Open-Source/tuya-convert/wiki/Collaboration-document-for-PSK-Identity-02#known-affected-devices)

closed 2020-12-22 by kueblc

tanderson1992, 2021-01-04:
I don't want to open another ticket if it's the same as the PSK issue, but are you saying the PSK 02 issue can also present itself as not attempting to connect at all? I tried flashing 3 Novostella bulbs and received the PSK ID starting with 02, a clean kill. I moved on to Aoycocr X5 plug, and like the OP got no reaction at all. The light continued to flash, and there's no evidence of it connecting in the logs. If I understand correctly, that plug will also not work because of the new PSK ID? Thanks for your time. I seem to often to find out about projects like this just after the manufacturers have started closing off access. I'll keep an eye on the project and keep looking for older devices that are hopefully compatible.