ct-Open-Source/tuya-convert #881: Gosund WP5 - Issues Flashing

war1000, 2020-12-21:
Hi, just bought some gosund wp5 from amazon canada. But having trouble flashing them. Tried windows first and then raspberrypi 4. Attaching the log files. Can someone kindly point me in the right direction?

[smarthack-mqtt.log](https://github.com/ct-Open-Source/tuya-convert/files/5727008/smarthack-mqtt.log)
[smarthack-psk.log](https://github.com/ct-Open-Source/tuya-convert/files/5727009/smarthack-psk.log)
[smarthack-udp.log](https://github.com/ct-Open-Source/tuya-convert/files/5727010/smarthack-udp.log)
[smarthack-web.log](https://github.com/ct-Open-Source/tuya-convert/files/5727011/smarthack-web.log)
[smarthack-wifi.log](https://github.com/ct-Open-Source/tuya-convert/files/5727012/smarthack-wifi.log)

war1000, 2020-12-21:
I just tried this with a esp8266 for the wifi part and here are the results. Please note that in the udp log the 4 plugs you see are zoozee plugs that can't be flashed and I have them attached to the cloud. The ones I am trying to convert are the wp5s. 

https://www.amazon.ca/Gosund-Function-Remotely-Control-Required/dp/B085DBVSRK/ref=sr_1_6?dchild=1&keywords=gosund+wp5&qid=1608595062&sr=8-6

[smarthack-mqtt.log](https://github.com/ct-Open-Source/tuya-convert/files/5727042/smarthack-mqtt.log)
[smarthack-psk.log](https://github.com/ct-Open-Source/tuya-convert/files/5727043/smarthack-psk.log)
[smarthack-udp.log](https://github.com/ct-Open-Source/tuya-convert/files/5727044/smarthack-udp.log)
[smarthack-web.log](https://github.com/ct-Open-Source/tuya-convert/files/5727045/smarthack-web.log)
[smarthack-wifi.log](https://github.com/ct-Open-Source/tuya-convert/files/5727046/smarthack-wifi.log)

kueblc, 2020-12-22:
It looks like this device is not supported, see [tuya-convert/wiki](https://github.com/ct-Open-Source/tuya-convert/wiki/Collaboration-document-for-PSK-Identity-02#known-affected-devices) and #483 for more information.

closed 2020-12-22 by kueblc