ct-Open-Source/tuya-convert #882: Can't flash my second smart plug

IlFre, 2020-12-22:
Hello,

I have successfully flashed the smart plug but I cannot flash the second one, same brand and same model, what can I do?

[smarthack-mqtt.log](https://github.com/ct-Open-Source/tuya-convert/files/5729567/smarthack-mqtt.log)
[smarthack-psk.log](https://github.com/ct-Open-Source/tuya-convert/files/5729568/smarthack-psk.log)
[smarthack-udp.log](https://github.com/ct-Open-Source/tuya-convert/files/5729569/smarthack-udp.log)
[smarthack-web.log](https://github.com/ct-Open-Source/tuya-convert/files/5729570/smarthack-web.log)
[smarthack-wifi.log](https://github.com/ct-Open-Source/tuya-convert/files/5729571/smarthack-wifi.log)

darkcloud784, 2020-12-24:
I have the same issue with my Feit BPA800/RGBW/AG/2(P) FCCID [SYW-A19RGBWAGT2](https://fccid.io/SYW-A19RGBWAGT2)

Josvds, 2020-12-30:
I have the same issue with my Neo Coolcam NAS-WR01W. 
```
======================================================
Starting smart config pairing procedure
Waiting for the device to install the intermediate firmware
Put device in EZ config mode (blinking fast)
Sending SSID                  vtrust-flash
Sending wifiPassword
Sending token                 00000000
Sending secret                0101
................
SmartConfig complete.
Resending SmartConfig Packets
.................
SmartConfig complete.
Resending SmartConfig Packets
.................
SmartConfig complete.
Resending SmartConfig Packets
..................
SmartConfig complete.
Resending SmartConfig Packets
.................
SmartConfig complete.
Resending SmartConfig Packets
.................
SmartConfig complete.
Resending SmartConfig Packets
.................
SmartConfig complete.
Resending SmartConfig Packets
................
SmartConfig complete.
Resending SmartConfig Packets
.................
SmartConfig complete.
Resending SmartConfig Packets
.................
SmartConfig complete.
Resending SmartConfig Packets
............
Device did not appear with the intermediate firmware
Check the *.log files in the scripts folder
Do you want to try flashing another device? [y/N] n
======================================================
```

raman325, 2020-12-30:
refer to https://github.com/ct-Open-Source/tuya-convert/issues/483