ct-Open-Source/tuya-convert #884: Newer Bseed Smart Wifi Socket not convertible anymore ?!? (sslpsk error)

amoksworld, 2020-12-23:
Hi,

some time ago I bought some Bseed Smart Wifi Sockets ... and was happy that I could convert them with tuya-convert.
Because I was the first one, I also created a template for tasmota.

https://templates.blakadder.com/bseed_smart_socket.html

Now I bought some more of this devices, but sadly flasing is not possible anymore.

Saw the hint from the the page too late:

"Tuya-Convert might not be possible for this device if the manufacturer started using newest Tuya firmware since this template was published (2020-01-05)."

Someone edited the page and pointing me to #483

I read this thread and also found the Wiki:

https://github.com/ct-Open-Source/tuya-convert/wiki/Collaboration-document-for-PSK-Identity-02

But I dont understand, how I can help now.

Surprisingly I found an old device at home, that was never powered up before.
I tried to flash it with tuya-convert and got success.
During this try I got a Backup of the old Firmware.

Now I have 2 devices, one successfully flased, one that wont flash.
I have the Backup of the old Firmware ...
And now ... ?

Should I try to flash the old firmware to the old device and do an online update while I am sniffing the traffic?
Or is there any possibility to flash the old firmware (backup) to the newer devices?
Perhaps when I disassemble one and try with serial link?

I would be grateful for any advice.

amoksworld, 2020-12-23:
Seems that this devices can be opened very easily without distruction:

[image]

And hopefully I can flash the old firmware over serial connection of the board with esp-tool

amoksworld, 2020-12-23:
After soldering some wires to the VBAT,GND,2RX,2TX pads of the wifi board:

[image]

I connected it with a UART and minicom was showing some output:

[image]

And a reset was showning:

`****SystemReset****

V:BK7231S_1.0.5
CPSR:000000D3
R0:00000000
R1:00000004
R2:00000000
R3:0040483C
R4:00000000
R13:00402720
R14(LR):0005D2D0
ST:00000001
J 0x10000
prvHeapInit-start addr:0x41e650, size:137648
[01-01 18:12:15 TUYA Debug][uni_thread.c:215] Thread:sys_timer Exec Start. Set to Running Status
[01-01 18:12:15 TUYA Debug][online_log_serv.c:303] read from uf. max:2 first:0 last:1
[01-01 18:12:15 TUYA Debug][online_log_serv.c:540] log serv init success
[01-01 18:12:15 TUYA Debug][tuya_device.c:192] < TUYA IOT SDK V:2.0.0 BS:30.06_PT:2.2_LAN:3.3_CAD:1.0.2_CD:1.0.0 >
< tuya_iot_lib BUILD AT:2018_12_05_17_03_30 BY tuya_iot_team AT 8710_2M >
IOT DEFS < WIFI_GW:1 DEBUG:1 KV_FILE:0 SHUTDOWN_MODE:0 LITTLE_END:1 TLS_MODE:2 ENABLE_LOCAL_LINKAGE:0 ENABLE_CLOUD_OPERATION:0 ENABLE_SUBDEVICE:0 ENABLE_ENGINEER_TO_NORMAL:0 OPERATING_SYSTEM:2 ENABLE_SYS_RPC:0 TY_SECURITY_CHIP:0 RELIABLE>

[01-01 18:12:15 TUYA Debug][tuya_device.c:193] oem_bk7231s_rnd_switch:1.0.7
[01-01 18:12:15 TUYA Notice][tuya_device.c:194] firmware compiled at Jun 10 2020 15:59:57
0xcb 0x4e 0x3e 0xa4 0x0 0x30 0x9d 0xab 0x65 0x6d 0x8d 0xbf 0xe4 0xb9 0x3f 0x35 
[01-01 18:12:15 TUYA Notice][tuya_main.c:203] **********[oem_bk7231s_rnd_switch] [1.0.7] compiled at Jun 10 2020 15:59:54**********
[rx_iq]rx_amp_err_rd: 0xfffffffe
[rx_iq]rx_phase_err_rd: 0x065
[rx_iq]rx_ty2_rd: 0x004
*********** finally result **********
gbias_after_cal: 0x17
gav_tssi: 0x19
gtx_q_dc_comp:0x202
gtx_i_dc_comp:0x20e
gtx_i_gain_comp:1023
gtx_q_gain_comp:1016
gtx_phase_comp:551
gtx_phase_ty2:513
gtx_ifilter_corner over: 0x11
gtx_qfilter_corner over: 0xf
gtx_dcorMod:0x8, gtx_dcorPA:0xa
gtx_pre_gain:0x0
g_rx_dc_gain_tab 0 over: 0x80808080
g_rx_dc_gain_tab 1 over: 0x8278807c
g_rx_dc_gain_tab 2 over: 0x8a728c70
g_rx_dc_gain_tab 3 over: 0xac5ca068
g_rx_dc_gain_tab 4 over: 0xaa5ca85a
g_rx_dc_gain_tab 5 over: 0xaa5aaa5b
g_rx_dc_gain_tab 6 over: 0xaa5aaa5a
g_rx_dc_gain_tab 7 over: 0xab59aa59
grx_amp_err_wr:0x203
grx_phase_err_wr:0x032
**************************************
temp in flash is:251
lpf_i & q in flash is:17, 15
xtal in flash is:23
-----pwr_gain:12, g_idx:12, shift_b:0, shift_g:0
                                                -----[pwr_gain]12
                                                                 Initializing TCP/IP stack
[01-01 18:12:16 TUYA Notice][tuya_main.c:229] mf_init succ
[01-01 18:12:16 TUYA Notice][tuya_ble_api.c:292] ble sdk inited
[... 133 lines cut ...]
[01-01 18:14:56 TUYA Notice][tuya_device.c:1031] wifi_stat = 1
[01-01 18:15:01 TUYA Notice][tuya_device.c:1030] device_init ok  free_mem_size:74672
[01-01 18:15:01 TUYA Notice][tuya_device.c:1031] wifi_stat = 1
[01-01 18:15:06 TUYA Notice][tuya_device.c:1030] device_init ok  free_mem_size:74672
[01-01 18:15:06 TUYA Notice][tuya_device.c:1031] wifi_stat = 1
[01-01 18:15:11 TUYA Notice][tuya_device.c:1030] device_init ok  free_mem_size:74672
[01-01 18:15:11 TUYA Notice][tuya_device.c:1031] wifi_stat = 1
[01-01 18:15:16 TUYA Notice][gw_intf.c:658] stop smt cfg mthd:1
[01-01 18:15:16 TUYA Notice][gw_intf.c:661] lowpower or special mode stop tmm_short and tmm_long timer
!!!!!!!!!!tuya_bt_port_deinit
gapm_cmp_evt_handler operation = 0xd, status = 0x44 
[01-01 18:15:16 TUYA Notice][tuya_device.c:698] wifi stat is 0 
[01-01 18:15:16 TUYA Notice][tuya_device.c:1030] device_init ok  free_mem_size:79608
[01-01 18:15:16 TUYA Notice][tuya_device.c:1031] wifi_stat = 0
[01-01 18:15:21 TUYA Notice][tuya_device.c:1030] device_init ok  free_mem_size:79608
[01-01 18:15:21 TUYA Notice][tuya_device.c:1031] wifi_stat = 0
[01-01 18:15:26 TUYA Notice][tuya_device.c:1030] device_init ok  free_mem_size:79608
[01-01 18:15:26 TUYA Notice][tuya_device.c:1031] wifi_stat = 0
[01-01 18:15:31 TUYA Notice][tuya_device.c:1030] device_init ok  free_mem_size:79608
[01-01 18:15:31 TUYA Notice][tuya_device.c:1031] wifi_stat = 0
[01-01 18:15:36 TUYA Notice][tuya_device.c:1030] device_init ok  free_mem_size:79608
[01-01 18:15:36 TUYA Notice][tuya_device.c:1031] wifi_stat = 0
`

Now the question is, ... how to communicate with the device?

amoksworld, 2020-12-23:
This is the output of the local console during the try to flash with tuya-convert:

`[01-01 18:12:46 TUYA Notice][tuya_device.c:1030] device_init ok  free_mem_size:74648
[01-01 18:12:46 TUYA Notice][tuya_device.c:1031] wifi_stat = 1
[01-01 18:12:51 TUYA Notice][tuya_device.c:1030] device_init ok  free_mem_size:74648
[01-01 18:12:51 TUYA Notice][tuya_device.c:1031] wifi_stat = 1
[01-01 18:12:56 TUYA Notice][tuya_device.c:1030] device_init ok  free_mem_size:74648
[01-01 18:12:56 TUYA Notice][tuya_device.c:1031] wifi_stat = 1
sort_rev_data: 0x00c5 0x0080 0x0100 0x010e 0x0155 0x0153
sort_rev_data: 0x0086 0x0082 0x0130 0x0130 0x0130 0x0130
sort_rev_data: 0x00be 0x0083 0x0130 0x0131 0x0130 0x0131
sort_rev_data: 0x00af 0x0084 0x0176 0x0174 0x0172 0x0175
sort_rev_data: 0x00d4 0x0086 0x016c 0x0161 0x0173 0x0168
[01-01 18:12:57 TUYA Notice][tuya_bt_sdk.c:353] ty bt network cfg finish..
[01-01 18:12:57 TUYA Notice][tuya_ble_api.c:161] update bound state 1
device id key : 16
b3 d7 5c 59 35 98 bb a2 dc 87 45 da 83 97 5d c5 
!!!!!!!!!!tuya_bt_reset_adv
[01-01 18:12:57 TUYA Notice][tuya_ble_api.c:120] ble adv && resp changed
!!!!!!!!!!tuya_after_netcfg_cb:3
[01-01 18:12:57 TUYA Notice][gw_intf.c:658] stop smt cfg mthd:1
[01-01 18:12:57 TUYA Notice][gw_intf.c:661] lowpower or special mode stop tmm_short and tmm_long timer
gapm_cmp_evt_handler operation = 0xd, status = 0x44 
fast_connect
bssid 26-d0-49-c3-53-ef
enter low level!
mac d4:a6:51:b9: 2:b2
leave low level!
ssid:vtrust-flash, 1
ht in scan
scan_start_req_handler
found scan rst rssi -47 > -50
found scan rst rssi -47 > -50
found scan rst rssi -49 > -50
me_set_ps_disable:840 0 0 0 428597 905029
sm_auth_send:1
sm_auth_handler
sm_assoc_rsp_handler
rc_init: station_id=0 format_mod=0 pre_type=0 short_gi=437017 max_bw=905510
                                                                           rc_init: nss_max=0 mcs_max=0 r_idx_min=255 r_idx_max=437059 no_samples=905577
                                                                                                                                                        ctrl_port_hdl:1
me_set_ps_disable:840 0 0 0 428597 905029

configuring interface mlan (with DHCP client)
new ie: 0 : 76 74 72 75 73 74 2d 66 6c 61 73 68 
new ie: 1 : 82 84 b 16 
new ie: 3 : 1 

me_send_ps_req 2 0 0
[01-01 18:12:58 TUYA Notice][tuya_device.c:698] wifi stat is 6 
[01-01 18:12:58 TUYA Notice][tuya_bt_sdk.c:385] __dp_data_report_data length:11
__dp_data_report_data 11
09 02 04 00 00 00 00 01 01 01 00 
[01-01 18:12:58 TUYA Notice][tuya_ble_data_handler.c:25] ble get connect status:4
[01-01 18:12:58 TUYA Notice][tuya_ble_api.c:381] tuya_ble_sdk_send skip... bt not connect:4
[01-01 18:12:58 TUYA Err][smart_frame.c:255] devid: dparr[0]:1 not find
[01-01 18:12:58 TUYA Err][tuya_iot_com_api.c:718] dp composition fail.ret:-944
[01-01 18:12:58 TUYA Err][upload.c:124] upload_all_switch_dp_stat op_ret:-944
[01-01 18:12:58 TUYA Err][upload.c:142] upload_all_switch_dp_stat op_ret:-944
[01-01 18:12:58 TUYA Notice][tuya_bt_sdk.c:385] __dp_data_report_data length:3
__dp_data_report_data 3
00 03 00 
[01-01 18:12:58 TUYA Notice][tuya_ble_data_handler.c:25] ble get connect status:4
[01-01 18:12:58 TUYA Notice][tuya_ble_api.c:381] tuya_ble_sdk_send skip... bt not connect:4
[01-01 18:12:58 TUYA Err][smart_frame.c:255] devid: dparr[0]:0 not find
[01-01 18:12:58 TUYA Err][tuya_iot_com_api.c:718] dp composition fail.ret:-944
[01-01 18:12:58 TUYA Err][tuya_timer.c:972] dev_report_dp_json_async err:-944
[01-01 18:12:58 TUYA Notice][tuya_bt_sdk.c:385] __dp_data_report_data length:3
__dp_data_report_data 3
2b 03 00 
[... 127 lines cut ...]
                                   [01-01 18:14:11 TUYA Notice][tuya_device.c:1030] device_init ok  free_mem_size:71464
[01-01 18:14:11 TUYA Notice][tuya_device.c:1031] wifi_stat = 6
do td cur_t:283--last:idx:15,t:275 -- new:idx:16,t:287 
--0xc:08, shift_b:0, shift_g:0, X:0
                                   [01-01 18:14:16 TUYA Notice][tuya_device.c:1030] device_init ok  free_mem_size:71464
[01-01 18:14:16 TUYA Notice][tuya_device.c:1031] wifi_stat = 6
[01-01 18:14:21 TUYA Notice][tuya_device.c:1030] device_init ok  free_mem_size:71464
[01-01 18:14:21 TUYA Notice][tuya_device.c:1031] wifi_stat = 6
[01-01 18:14:26 TUYA Notice][tuya_device.c:1030] device_init ok  free_mem_size:71464
[01-01 18:14:26 TUYA Notice][tuya_device.c:1031] wifi_stat = 6
do td cur_t:268--last:idx:16,t:287 -- new:idx:15,t:275 
--0xc:08, shift_b:0, shift_g:0, X:1
                                   [01-01 18:14:31 TUYA Notice][tuya_device.c:1030] device_init ok  free_mem_size:71464
[01-01 18:14:31 TUYA Notice][tuya_device.c:1031] wifi_stat = 6
[01-01 18:14:36 TUYA Notice][tuya_device.c:1030] device_init ok  free_mem_size:71464
[01-01 18:14:36 TUYA Notice][tuya_device.c:1031] wifi_stat = 6
[01-01 18:14:41 TUYA Notice][tuya_device.c:1030] device_init ok  free_mem_size:71464
[01-01 18:14:41 TUYA Notice][tuya_device.c:1031] wifi_stat = 6
do td cur_t:264--last:idx:15,t:275 -- new:idx:14,t:263 
--0xc:08, shift_b:0, shift_g:0, X:2
                                   [01-01 18:14:46 TUYA Notice][tuya_device.c:1030] device_init ok  free_mem_size:71464
[01-01 18:14:46 TUYA Notice][tuya_device.c:1031] wifi_stat = 6
[01-01 18:14:51 TUYA Notice][tuya_device.c:1030] device_init ok  free_mem_size:71464
[01-01 18:14:51 TUYA Notice][tuya_device.c:1031] wifi_stat = 6
`

amoksworld, 2020-12-24:
Okay ... my fault ... I think I can stop here:

[image]

It has an WB2S Board ...

I'll try to replace it with some small ESP module...

amoksworld, 2020-12-24:
Because there is not much I can do now ... I started to find out differences of the two versions I have ...

It seems they canged the PCB a little bit and now using smaller relais: 
[image]

what is helpfull to identify the version without opening the device - the old ones (with the big relais) are 5g heavier:
[image]
[image]

The WB2S seems to be 100% pin compatible with the TYW2S because both WIFI boards are connected in the same way.
[image]

So I start searching for some replacement modules ...

Sadly the pinout of an ESP-M4 (ESP8285 Wifi Modul) is slightly different ... but perhaps its small enough to wire it inside.

aguilaair, 2020-12-26:
See #483

mrThe, 2021-01-09:
I'm very new to esp8266 and tuya specifics, but why can't we just replace this esp8266 to the new one and flash it like we want? It cost around $2 on ali, compared to price of wall socket(~$15) it's not much an issue. 

Since we can't flash it, replace is a best option at the moment, isnt?.

Am i wrong?

kueblc, 2021-01-09:
@mrThe you are spot on. Many users have done this successfully. Take a look around the issues here for walk throughs posted by community members.

Volcano-28, 2022-05-28:
I bought a set of Bseed sockets (Shuko type), some of them has TYWE2S chip on it, some of them has WB2S.
I have replaced WB2S to TYWE2S and tasmotize with success, but I bought 2 sockets more and I was surprised that the socket is completely different inside and gently outside.
WB2S/TYWE2S sockets has dept size 41,80 mm, the other types bigger 45,00 mm.

[image]

[image]

Construction and PCB inside is completely different.
[image]
[image]
[image]

The chip is made by Tuya
[image]

I think it cannot be replaced and/or tasmotized.

The back of the socket is easy to differentiate.
The wire connectors (especially ground) is different (left old type, right new type).

[image]

From the face it could be differentiate too.
In new socket the LED eye is much smaller than old version.
 
[image]

djdanip, 2022-09-02:
any solution to change yours chip, i have another one just like yours t34, i would like to change it for esp chip with tasmota.

openshwprojects, 2023-01-18:
@djdanip @Volcano-28 T34, among many other new Tuya platforms, is supported by multiplatform IoT firmware - OpenBeken:
https://github.com/openshwprojects/OpenBK7231T_App
https://openbekeniot.github.io/webapp/devicesList.html
https://www.youtube.com/watch?v=KU0tDwtjfjw&ab_channel=Elektrodacom