ct-Open-Source/tuya-convert #885: Can't Flash Teckin SP-22 smart plug

poseidogithub, 2020-12-27:
Hello

First of all, thanks to the developers and community for this amazing work, 
Second, sorry for my english!

I've used tuya-convert in the past with fantastic results  in some devices, and saddening result with the psk2 issue... so I know the basics.

my setup is a raspberry pi 3b with latest raspbian with a clean install, and with main release of tuya-convert.
I've also a ESP8266 nodemcu with a firmware that tries to connect to vtrust-flash (the led flashes until its connected), it just does that so I dont need a phone to connect to that wifi (lately I find that my phones disconnect from vtrust-flash ssid because they see there is no internet there...)
the devices I try to flash are teckin SP22 smart plugs with power meter. I've had luck in the past with another batch and tuya convert.

so, to the point. 
- I run ./start_flash.sh
- I get the usual messages until the line that ask me to press enter. At this point my nodemcu is already connected to vtrust-flash ssid (I know because the led stop flashing)
- next I long press the main (and only) button on the teckin plug, it powers off for a second and power automatically on and starts to blink red/blue rapidly
- I press ENTER
- after 1 or 2 seconds that I press enter, the LED on the plug goes to RED. this is not time-bound it reacts to me pressing enter on tuya-convert script. if I dont press enter it keeps blikning red-blue forever. this makes me think that tuya-convert is able to communicate with the plug.
- the script ends with the "device did not apper with the intermediate firmware, check the log... do you want..."
I press no, and My nodemcu goes to blink as the vtrust-flash disconnects.

`pi@raspberrypi-3b:~/tuya-convert $ ./start_flash.sh
tuya-convert v2.4.4
Checking for network interface wlan0... Found.
Checking UDP port 53... Available.
Checking UDP port 67... Available.
Checking TCP port 80... Available.
Checking TCP port 443... Available.
Checking UDP port 6666... Available.
Checking UDP port 6667... Available.
Checking TCP port 1883... Available.
Checking TCP port 8886... Available.
======================================================
  Starting AP in a screen
  Starting web server in a screen
  Starting Mosquitto in a screen
  Starting PSK frontend in a screen
  Starting Tuya Discovery in a screen

======================================================

IMPORTANT
1. Connect any other device (a smartphone or something) to the WIFI vtrust-flash
   This step is IMPORTANT otherwise the smartconfig may not work!
2. Put your IoT device in autoconfig/smartconfig/pairing mode (LED will blink fast). This is usually done by pressing and holding the primary button of the device
   Make sure nothing else is plugged into your IoT device while attempting to flash.
3. Press ENTER to continue

======================================================
Starting smart config pairing procedure
Waiting for the device to install the intermediate firmware
Put device in EZ config mode (blinking fast)
Sending SSID                  vtrust-flash
Sending wifiPassword
Sending token                 00000000
Sending secret                0101
...............
SmartConfig complete.
Resending SmartConfig Packets
.................
SmartConfig complete.
Resending SmartConfig Packets
.................
SmartConfig complete.
Resending SmartConfig Packets
..................
SmartConfig complete.
Resending SmartConfig Packets
..................
SmartConfig complete.
Resending SmartConfig Packets
..................
SmartConfig complete.
Resending SmartConfig Packets
................
SmartConfig complete.
Resending SmartConfig Packets
.................
SmartConfig complete.
Resending SmartConfig Packets
.................
SmartConfig complete.
Resending SmartConfig Packets
................
SmartConfig complete.
Resending SmartConfig Packets
.................
Device did not appear with the intermediate firmware
Check the *.log files in the scripts folder
Do you want to try flashing another device? [y/N] n
======================================================
Cleaning up...
Closing AP
Exiting...
`

the content of the logs is:

smarthack-mqtt.log:
```
1609096902: mosquitto version 1.5.7 starting
1609096902: Using default config.
1609096902: Opening ipv4 listen socket on port 1883.
1609096902: Opening ipv6 listen socket on port 1883.
^C1609097208: mosquitto version 1.5.7 terminating
```

smarthack-psk.log
```
^CTraceback (most recent call last):
  File "./psk-frontend.py", line 113, in <module>
    main()
  File "./psk-frontend.py", line 106, in main
    r,_,_ =  select.select(readables, [], [])
KeyboardInterrupt

```
smarthack-udp.log
```
Listening for Tuya broadcast on UDP 6666
Listening for encrypted Tuya broadcast on UDP 6667
192.168.1.211 {"ip":"192.168.1.211","gwId":"403656075002915c56c3","active":2,"ability":0,"mode":0,"encrypt":true,"productKey":"keyj979nf3q3theh","version":"3.3"}

smarthack-web.log:

Listening on 10.42.42.1:80
^CReceived SIGINT, exiting...

```
smarthack-wifi.log:

```

System info
===========
94acc3b
Linux raspberrypi-3b 5.4.79-v7+ #1373 SMP Mon Nov 23 13:22:33 GMT 2020 armv7l GNU/Linux
OpenSSL 1.1.1d  10 Sep 2019
Dnsmasq version 2.80  Copyright (c) 2000-2018 Simon Kelley
Compile time options: IPv6 GNU-getopt DBus i18n IDN DHCP DHCPv6 no-Lua TFTP conntrack ipset auth DNSSEC loop-detect inotify dumpfile

This software comes with ABSOLUTELY NO WARRANTY.
Dnsmasq is free software, and you are welcome to redistribute it
under the terms of the GNU General Public License, version 2 or 3.
hostapd v2.8-devel
User space daemon for IEEE 802.11 AP management,
IEEE 802.1X/WPA/WPA2/EAP/RADIUS Authenticator
Copyright (c) 2002-2019, Jouni Malinen <j@w1.fi> and contributors
Python 3.7.3
===========
Configuring AP interface...
RTNETLINK answers: File exists
RTNETLINK answers: File exists
Starting DNSMASQ server...
Starting AP on wlan0...
Configuration file: /dev/stdin
wlan0: Could not connect to kernel driver
Using interface wlan0 with hwaddr b8:27:eb:d2:XX:XX and ssid "vtrust-flash"
wlan0: interface state UNINITIALIZED->ENABLED
wlan0: AP-ENABLED
wlan0: AP-STA-CONNECTED ec:fa:bc:28:XX:XX
wlan0: AP-STA-CONNECTED 84:e3:42:4d:XX:XX
wlan0: interface state ENABLED->DISABLED
wlan0: AP-STA-DISCONNECTED 84:e3:42:4d:XX:XX
wlan0: AP-STA-DISCONNECTED ec:fa:bc:28:XX:XX
wlan0: AP-DISABLED
wlan0: CTRL-EVENT-TERMINATING
nl80211: deinit ifname=wlan0 disabled_11b_rates=0
AP closed
Stopping DNSMASQ server...

```
84:e3:42:4d:XX:XX is identified by https://macvendors.com/ as TUYA mac  -> I understand It's my plug
ec:fa:bc:28:XX:XX its my nodemcu MAC that connects automatically to vtrust-flash

worth to note that I dont press ctrl-c ctrl-X in any moment...

I've read somewhere that this means that intermediate software has been installed, but I dont know how to go from here...

I would appreciate any help. I would flash them using FTDI but they are almost impossible to open without destrying them.

kueblc, 2020-12-27:
> 84:e3:42:4d:XX:XX is identified by https://macvendors.com/ as TUYA mac  -> I understand It's my plug

Unfortunately this means it is not using an ESP82xx. All supported devices will have an OUI of "Espressif Inc". Sorry you will not be able to run ESP alternative firmware like Tasmota on this device, even if you disassemble the device.

closed 2020-12-27 by kueblc

poseidogithub, 2020-12-27:
> Unfortunately this means it is not using an ESP82xx. All supported devices will have an OUI of "Espressif Inc". Sorry you will not be able to run ESP alternative firmware like Tasmota on this [...]

seems you are right, based on blackadder info, I've been quite unlucky;
https://templates.blakadder.com/teckin_SP22.html
```
WARNING

There is a new version with a T102/WR2 WiFi module in it. This uses a RTL8710BN chip and can not be converted to Tasmota. Make sure that you get a plug with TYWE2S WiFi module based on ESP8266.

```

as a suggestion, couldnt tuya-convert display this info when the mac is not a esp8266 one, so idiots like me dont come here asking for help?

kueblc, 2020-12-27:
> couldnt tuya-convert display this info when the mac is not a esp8266 one

Would love that, and have planned for this and other debugging hints, just haven't gotten around to it. PRs welcome :)