ct-Open-Source/tuya-convert #886: gosund UP111 - 3 out of 4 successfully flashed

TimNe0, 2020-12-28:
**Did you have issues getting a device to flash?**
_Please be sure to include your device **name, model, and firmware version** as well as your **logs**!_

Bought a 4pk Gosund UP111 from Amazon.  First three flashed fine.  Last one refused to flash

new client on port 443 from 10.42.42.38:30868
ID: 0242416f68626d64366147393149465231d573656cd946df0ed77a2c37017f92f063cf2b90d615212bf4b6e3439851a749
PSK: 50a5c7431467e9b9eabbcbbb72305018ad3f91c578113460eac84e34e19ebcb5
could not establish sslpsk socket: [SSL: DECRYPTION_FAILED_OR_BAD_RECORD_MAC] decryption failed or bad record mac (_ssl.c:1056)

Am I truly unlucky at getting one with a new psk or a change in chip in a box of 4? 

[smarthack-web.log](https://github.com/ct-Open-Source/tuya-convert/files/5747836/smarthack-web.log)
[smarthack-psk.log](https://github.com/ct-Open-Source/tuya-convert/files/5747837/smarthack-psk.log)
[smarthack-wifi.log](https://github.com/ct-Open-Source/tuya-convert/files/5747838/smarthack-wifi.log)
[smarthack-mqtt.log](https://github.com/ct-Open-Source/tuya-convert/files/5747839/smarthack-mqtt.log)
[smarthack-udp.log](https://github.com/ct-Open-Source/tuya-convert/files/5747840/smarthack-udp.log)

raman325, 2020-12-30:
Sounds like you got lucky, not unlucky, as I just purchased a set of four Gosund WP3's and most if not all are on the new PSK (haven't checked them all yet). See here for more info: https://github.com/ct-Open-Source/tuya-convert/issues/483

kueblc, 2021-01-09:
Indeed, whether you consider it unlucky that one was blocked, or lucky that three worked, it looks like this is the case.

I'd prefer to see the glass half full. Or in this case three quarters.

The fourth is probably still flashable the old fashion way, over serial wire.

closed 2021-01-09 by kueblc