ct-Open-Source/tuya-convert #887: Woox R4152 failed to program

jan4fun, 2020-12-29:
Dear,

I tried to flash the Woox R4152 with the Tuya-convert but is fails.
I noticed that it is not yet listed in the supported devices.
Are there plans to also support this one or is there a possible workaround for this type?

The error I get is: Device did not appear with the intermediate firmware

Setup: Raspberry Pi 3B
Added that wemos board to monitor the AP is connected

Thank you

kueblc, 2020-12-29:
Without your logs I cannot say what the problem is. It is likely unsupported due to #483.

jan4fun, 2020-12-29:
Sorry here are the logs
[smarthack-mqtt.log](https://github.com/ct-Open-Source/tuya-convert/files/5752297/smarthack-mqtt.log)
[smarthack-psk.log](https://github.com/ct-Open-Source/tuya-convert/files/5752298/smarthack-psk.log)
[smarthack-udp.log](https://github.com/ct-Open-Source/tuya-convert/files/5752299/smarthack-udp.log)
[smarthack-web.log](https://github.com/ct-Open-Source/tuya-convert/files/5752300/smarthack-web.log)
[smarthack-wifi.log](https://github.com/ct-Open-Source/tuya-convert/files/5752301/smarthack-wifi.log)

kueblc, 2020-12-29:
Sorry to say this device is not supported due to #483. Please see the [wiki](https://github.com/ct-Open-Source/tuya-convert/wiki/Collaboration-document-for-PSK-Identity-02) if you'd like to help out.

closed 2020-12-29 by kueblc