ct-Open-Source/tuya-convert #888: Can't flash WiFi Smart LED Strip: Device did not appear with the intermediate firmware.

phyrz91, 2020-12-30:
Hi, I can't upload the software, everything is ok, the smartphone connects to vtrust-flash SSID, the led strip is in pairing mode. Unfortunately, I get this message:

```
Device did not appear with the intermediate firmware
Check the *.log files in the scripts folder
Do you want to try flashing another device? [y/N] 
```

I have tried several times but with no success.

Chipset: TYWE3S,  Led strip work with Tuya app.

LOG:
[https://pastebin.com/B532uMiq](url)
[https://pastebin.com/sxCTqjC5](url)
[https://pastebin.com/DDrZNrba](url)

Hope to avoid soldering.
thank you in advance.

raman325, 2020-12-30:
You are likely running into https://github.com/ct-Open-Source/tuya-convert/issues/483

kueblc, 2020-12-31:
This is indeed #483, a new firmware from Tuya which is not currently supported. See the wiki for more information.

closed 2020-12-31 by kueblc