ct-Open-Source/tuya-convert #889: Merkury Bulbs MI-BW210-999W Refuse to Flash

jmiahman, 2021-01-01:
This most likely has to do with these being sent out with patched firmware, but I didn't see this reported anywhere else so I bought 6 of them from Walmart (Dec-31-2021) and none would flash. I am seeing this is the scripts/smarthack-psk.log log.

`^CTraceback (most recent call last):
  File "./psk-frontend.py", line 113, in <module>
    main()
  File "./psk-frontend.py", line 106, in main
    r,_,_ =  select.select(readables, [], [])`

As described in with the error in the development branch the devices left easy pairing mode (LED stopped blinking quickly) when the config packets were sent. I have flashed an older Merkury Bulb 204 Bulb and a outlet just fine. I was so excited I bought these bulbs to light up the house lol only to be hit with nothing :( if there's anything else I can provide just ask. Thanks for your hard work! Hope this prevents someone else from going on a spending spree before a fix is found.

oziee, 2021-01-03:
i am seeing the same error in my log

jakejblock, 2021-01-04:
Seeing the same in my log files with a Globe 37783.

kueblc, 2021-01-09:
Many brands including Merkury have switched away from using the ESP82xx. Unfortunately these use a completely different firmware and is in no way compatible with tuya-convert or ESP alternative firmwares like Tasmota.

To confirm, you can check the OUI of the MAC address. ESP will always be listed as Espressif, Inc.

Big bummer, the Walmart in my area also had these on sale for dirt cheap. No point if I can't own the device.

koepnick, 2021-01-14:
I bought a few of these on a whim and sacrificed a dimable white Merkury bulb. MI-BW942-999W.

The module on this one is a WB2L (P/N 219980) [datasheet](https://developer.tuya.com/en/docs/iot/device-development/module/wifibt-dual-mode-module/wb-series-module/wb2l-datasheet?id=K9duegc9bualu)

The high voltage side is labeled EBP894-REV1.4 at the top and E199273 / PCEM-3 / 94V-0 opposite the MCU.

I'm looking at dumping the firmware right now. If there's a need, I can attach photos.

lmamakos, 2021-02-21:
i bought a box of two of these lamps at the local Wally-World, and I noticed in the instruction sheet that they talk about how to add the bulb to their app:  "Easiest: Add using Bluetooth discovery"   This seems to be a bad sign since the ESP8266 has no bluetooth radio.  This text is on a sticker placed on the relevant section of the instruction sheet.

Hmm.. I think this might be going back for a refund.
[image]

toastom, 2021-07-24:
Seems like tuya-convert only works with ESP-based smart devices. This device doesn't have an ESP, so this tool won't work.

TCPMeta, 2021-11-21:
I two bought a twin pack of these from walmart. The stamped on the bulb itself says 2009 but my guess that has nothing to do with it having a ESP chip or not. Anyone have any idea how to open one of these bulbs without destroying it?

kueblc, 2021-11-22:
@TCPMeta I'm sorry to say, in all likelihood, it does not use an ESP. Merkury isn't buying ESPs anymore, and all Walmart stores around me ran out of the ESP stock ages ago.

If you really want to check for yourself, the least destructive path is trying tuya-convert or pairing some other way and checking the MAC OUI. All ESP have an OUI of "Espressif, Inc". If you want to disassemble the bulb, the dome can be pried off with a pry tool, and then you should be able to see the top of the microcontroller board.

itkfilelor, 2021-11-23:
They can possibly be swapped with esp-01 depending on the current installed chip. I think Digi-Blur did a video on this.

pib, 2023-05-27:
There's a project that looks like it will be a good option for these newer ones, except they don't have something as convenient for flashing them without opening them up. https://github.com/openshwprojects/OpenBK7231T_App