ct-Open-Source/tuya-convert #892: Can’t connect to vtrust-flash

sominaik, 2021-01-05:
Hi,

I'm unable to connect to the AP from any other device. Pls help

Mrnt, 2021-01-06:
Did look at the log files generated in the `tuya-convert/scripts` folder?

sominaik, 2021-01-07:
No logs available in the folder. Do I have to initiate them?

Mrnt, 2021-01-08:
What are you running this on: OS? python version? python virtual environment?

Did you see the following show up in the console?:
```
check_eula
check_config
check_port udp 53 "resolve DNS queries"
check_port udp 67 "offer DHCP leases"
check_port tcp 80 "answer HTTP requests"
check_port tcp 443 "answer HTTPS requests"
check_port udp 6666 "detect unencrypted Tuya firmware"
check_port udp 6667 "detect encrypted Tuya firmware"
check_port tcp 1883 "run MQTT"
check_port tcp 8886 "run MQTTS"
check_firewall
check_blacklist
```

I had trouble getting it to work on Debian Buster - even after installing all the dependencies it would just not do anything. It turned out that on my install the command "ping" required sudo so I had to modify start_flash.sh 

changing
`
	while ! ping -c 1 -W 1 -n "$GATEWAY" &> /dev/null; do
`
to
`
	while ! sudo ping -c 1 -W 1 -n "$GATEWAY" &> /dev/null; do 
`
and
`
	while ! ping -c 1 -W 1 -n 10.42.42.42 &> /dev/null; do
`
to
`
	while ! sudo ping -c 1 -W 1 -n 10.42.42.42 &> /dev/null; do
`

sominaik, 2021-01-08:
Thanks for your response. Following is my setup

1-Virtual machine using Virtual Box -- 
2- Ubuntu 16.04 installed as an OS
3- TP-LINK_Archer_T4UH is used as wifi dongle .. driver succesfully installed
4- Python installed on the ubuntu system is 2.7.12

when i start the flash, I can see normal output on the console. The problem is that i cannot connect to the wifi dongle from my phone or laptop. how to capture the logs?

Mrnt, 2021-01-08:
Does the WiFi access point show up on your phone/laptop?

sominaik, 2021-01-08:
Yes, I can actually see that and can click on it to connect
it just does not get connected

Mrnt, 2021-01-08:
It is supposed to be using python 3 I think. Check you have that and then you may want to set a python virtual environment using python 3 if it is using python 2 by default.

sominaik, 2021-01-08:
any guide would be helpful

Mrnt, 2021-01-08:
https://packaging.python.org/guides/installing-using-pip-and-virtual-environments/

So something like this in the tuya-convert folder:

```
python3 -m venv env
source env/bin/activate
```
Which will create a python 3 virtual environment and then activate it.

Then rerun `install_prereq.sh` to make sure the python files are in the virtual environment you just created.

Then run `start_flash.sh`

Mrnt, 2021-01-08:
I’m not sure that setting the virtual environment will help you here, I got into the habit of doing it to have more control over what python modules are installed, but the tuya scripts will use python 3 anyway.

Have you checked your Linux is up to date?

YannikW, 2021-01-14:
Hi guys,

I'm having trouble to connect to the vtrust-flash wifi, too.
When I tap it on my phone or tables it seems to try to connect, but seconds later the network disappears. After refreshing the wifi list it shows again but I won't connect.

My setup:
Raspberry Pi 4 with fresh installed Raspberry Pi OS Lite from 2021-01-11

This is my smarthack-wifi.log :

```

System info
===========
94acc3b
Linux raspberrypi 5.4.83-v7l+ #1379 SMP Mon Dec 14 13:11:54 GMT 2020 armv7l GNU/Linux
OpenSSL 1.1.1d  10 Sep 2019
Dnsmasq version 2.80  Copyright (c) 2000-2018 Simon Kelley
Compile time options: IPv6 GNU-getopt DBus i18n IDN DHCP DHCPv6 no-Lua TFTP conntrack ipset auth DNSSEC l$

This software comes with ABSOLUTELY NO WARRANTY.
Dnsmasq is free software, and you are welcome to redistribute it
under the terms of the GNU General Public License, version 2 or 3.
hostapd v2.8-devel
User space daemon for IEEE 802.11 AP management,
IEEE 802.1X/WPA/WPA2/EAP/RADIUS Authenticator
Copyright (c) 2002-2019, Jouni Malinen <j@w1.fi> and contributors
Python 3.7.3
===========
Attempting to stop wpa_supplicant
Configuring AP interface...
RTNETLINK answers: File exists
Starting DNSMASQ server...
Starting AP on wlan0...
Configuration file: /dev/stdin
wlan0: Could not connect to kernel driver
Using interface wlan0 with hwaddr dc:a6:32:d7:fc:99 and ssid "vtrust-flash"
wlan0: interface state UNINITIALIZED->ENABLED
wlan0: AP-ENABLED
wlan0: AP-STA-CONNECTED 2e:8f:45:f8:14:5a
wlan0: AP-STA-DISCONNECTED 2e:8f:45:f8:14:5a
wlan0: AP-STA-CONNECTED 2e:8f:45:f8:14:5a
wlan0: AP-STA-DISCONNECTED 2e:8f:45:f8:14:5a
wlan0: AP-STA-CONNECTED 2e:8f:45:f8:14:5a
wlan0: AP-STA-DISCONNECTED 2e:8f:45:f8:14:5a
wlan0: AP-STA-CONNECTED 2e:8f:45:f8:14:5a
wlan0: AP-STA-DISCONNECTED 2e:8f:45:f8:14:5a
wlan0: AP-STA-CONNECTED 2e:8f:45:f8:14:5a
wlan0: AP-STA-DISCONNECTED 2e:8f:45:f8:14:5a
wlan0: interface state ENABLED->DISABLED
wlan0: AP-DISABLED
wlan0: CTRL-EVENT-TERMINATING
nl80211: deinit ifname=wlan0 disabled_11b_rates=0
AP closed
Stopping DNSMASQ server...
```

Everything seems fine but I get an instant disconenct after the connect. 
Can anybody help?

kotzer3, 2021-01-16:
Had this issue also on a RPi (2b), it was solved by setting WiFi country via
$sudo raspi-config
After that my mobile was staying in the vtrust-flash wifi network.

fatmcgav, 2021-01-18:
> Had this issue also on a RPi (2b), it was solved by setting WiFi country via
> $sudo raspi-config
> After that my mobile was staying in the vtrust-flash wifi network.

Just to add this worked for me on a fresh RPI4 using latest Raspbian :)

zeddski, 2021-01-19:
Ran into this (apparently) same issue.
Seemed that Wifi was being blocked by the Raspberry software, which is the default setting on some OS-es.

Fixed it with: "sudo rfkill unblock wifi" (if that doesn't work, replace "wifi" with "all").
Second: Although not the issue of @yannikW, make sure DNSMASQ is running, otherwise you won't get an IP address - which would explain why you cannot connect.

icebrian, 2021-01-29:
I am also experiencing this issue and none of the sugestion in the comments seem to have worked. Here are my log outputs:

`System info
===========
61ba060
Linux kmaster 5.10.5-v8+ #1392 SMP PREEMPT Sat Jan 9 18:56:30 GMT 2021 aarch64 GNU/Linux
OpenSSL 1.1.1d  10 Sep 2019
Dnsmasq version 2.80  Copyright (c) 2000-2018 Simon Kelley
Compile time options: IPv6 GNU-getopt DBus i18n IDN DHCP DHCPv6 no-Lua TFTP conntrack ipset auth DNSSEC loop-detect inotify dumpfile

This software comes with ABSOLUTELY NO WARRANTY.
Dnsmasq is free software, and you are welcome to redistribute it
under the terms of the GNU General Public License, version 2 or 3.
hostapd v2.8-devel
User space daemon for IEEE 802.11 AP management,
IEEE 802.1X/WPA/WPA2/EAP/RADIUS Authenticator
Copyright (c) 2002-2019, Jouni Malinen <j@w1.fi> and contributors
Python 3.7.3
===========
Attempting to stop wpa_supplicant
Configuring AP interface...
RTNETLINK answers: File exists
Error: Device for nexthop is not up.
Starting DNSMASQ server...
Starting AP on wlan0...
Configuration file: /dev/stdin
wlan0: Could not connect to kernel driver
Using interface wlan0 with hwaddr dc:a6:32:b8:4c:ea and ssid "vtrust-flash"
wlan0: interface state UNINITIALIZED->ENABLED
wlan0: AP-ENABLED

==> ./scripts/smarthack-web.log <==
Listening on 10.42.42.1:80

==> ./scripts/smarthack-mqtt.log <==
1611879440: mosquitto version 1.5.7 starting
1611879440: Using default config.
1611879440: Opening ipv4 listen socket on port 1883.
1611879440: Opening ipv6 listen socket on port 1883.

==> ./scripts/smarthack-udp.log <==
Listening for Tuya broadcast on UDP 6666
Listening for encrypted Tuya broadcast on UDP 6667`

eugeneniemand, 2021-03-02:
I'm also experiencing issues where it will just continuously try to connect to the vtrust-flash network. I've tried an iPhone 12 and a very old Android tablet and neither will connect. I'm on RPi 4 using the docker instructions. It created the backups folder but there are not logs. I had similar issues with network disappearing but the `sudo killall wpa_supplicant` fixed that. Any ideas why there are no logs and not connecting to the AP?

eugeneniemand, 2021-03-02:
I looked at docker logs and found this
```
*** Running /etc/my_init.d/00_regen_ssh_host_keys.sh...
*** Running /etc/my_init.d/10_syslog-ng.init...
Mar  2 15:52:07 raspberrypi syslog-ng[13]: syslog-ng starting up; version='3.13.2'
*** Booting runit daemon...
*** Runit started as PID 22
Mar  2 15:52:08 raspberrypi cron[26]: (CRON) INFO (pidfile fd = 3)
Mar  2 15:52:08 raspberrypi cron[26]: (CRON) INFO (Running @reboot jobs)
Mar  2 15:52:25 raspberrypi rfkill: unblock set for type wifi
Mar  2 15:52:27 raspberrypi dnsmasq[133]: failed to bind DHCP server socket: Address already in use
Mar  2 15:52:27 raspberrypi dnsmasq[133]: FAILED to start up
Mar  2 15:52:30 raspberrypi hostapd: wlan0: STA 2c:f4:32:6b:70:81 IEEE 802.11: associated
Mar  2 15:52:30 raspberrypi hostapd: wlan0: STA 2c:f4:32:6b:70:81 RADIUS: starting accounting session CFD54F897F827CED
Mar  2 15:53:14 raspberrypi hostapd: wlan0: STA fa:90:eb:9f:95:13 IEEE 802.11: associated
Mar  2 15:53:14 raspberrypi hostapd: wlan0: STA fa:90:eb:9f:95:13 RADIUS: starting accounting session EDF770553358A889
Mar  2 15:53:47 raspberrypi hostapd: wlan0: STA 2c:f4:32:6b:70:81 IEEE 802.11: disassociated
...
```

eugeneniemand, 2021-03-02:
But the Console output seems to think everything is available
```
Checking for network interface wlan0... Found.
Checking UDP port 53... Available.
Checking UDP port 67... Available.
Checking TCP port 80... Available.
Checking TCP port 443... Available.
Checking UDP port 6666... Available.
Checking UDP port 6667... Available.
Checking TCP port 1883... Available.
Checking TCP port 8886... Available.
======================================================
  Starting AP in a screen
  Starting web server in a screen
  Starting Mosquitto in a screen
  Starting PSK frontend in a screen
  Starting Tuya Discovery in a screen

======================================================

IMPORTANT
1. Connect any other device (a smartphone or something) to the WIFI vtrust-flash
   This step is IMPORTANT otherwise the smartconfig may not work!
2. Put your IoT device in autoconfig/smartconfig/pairing mode (LED will blink fast). This is usually done by pressing and holding the primary button of the device
   Make sure nothing else is plugged into your IoT device while attempting to flash.
3. Press ENTER to continue
```

NickVeld, 2021-04-28:
I ran into the problems with the AP too. I saw SSID in the available AP list but cannot connect. In my case, the log files were not generated somewhy.

But this advice helped me: https://github.com/ct-Open-Source/tuya-convert/issues/551#issuecomment-653799381
Only one note: WifiAnalyzer does not show me empty channels so I try checking random channels. With the 14th channel the AP dissappears from the list. In my analyzer only  1-11 channels were shown (so I recommend this range) thus I took the 6th channel and it works!

See also the PR #944

NickVeld, 2021-04-28:
As for the missing log files see the PR #943

MikiDi, 2022-02-23:
Running in to this too (docker-compose method).
Phone says "obtaining IP-address" for quite a while, and then disconnects. Tried setting country code or other channels in `hostapd.conf` to no avail.
```
tuya_1  | *** Running /etc/my_init.d/00_regen_ssh_host_keys.sh...
tuya_1  | *** Running /etc/my_init.d/10_syslog-ng.init...
tuya_1  | Feb 23 18:06:33 fedora syslog-ng[14]: syslog-ng starting up; version='3.25.1'
tuya_1  | *** Booting runit daemon...
tuya_1  | *** Runit started as PID 22
tuya_1  | Feb 23 18:06:34 fedora cron[26]: (CRON) INFO (pidfile fd = 3)
tuya_1  | Feb 23 18:06:34 fedora cron[26]: (CRON) INFO (Running @reboot jobs)
tuya_1  | Feb 23 18:06:40 fedora rfkill: unblock set for type wifi
tuya_1  | Feb 23 18:06:40 fedora dnsmasq[123]: started, version 2.80 cachesize 150
tuya_1  | Feb 23 18:06:40 fedora dnsmasq[123]: compile time options: IPv6 GNU-getopt DBus i18n IDN DHCP DHCPv6 no-Lua TFTP conntrack ipset auth nettlehash DNSSEC loop-detect inotify dumpfile
tuya_1  | Feb 23 18:06:40 fedora dnsmasq-dhcp[123]: DHCP, IP range 10.42.42.10 -- 10.42.42.40, lease time 12h
tuya_1  | Feb 23 18:06:40 fedora dnsmasq-dhcp[123]: DHCP, sockets bound exclusively to interface wlp2s0
tuya_1  | Feb 23 18:06:40 fedora dnsmasq[123]: read /etc/hosts - 2 addresses
tuya_1  | Feb 23 18:06:54 fedora hostapd: wlp2s0: STA mac:omitted IEEE 802.11: authenticated
tuya_1  | Feb 23 18:06:54 fedora hostapd: wlp2s0: STA mac:omitted IEEE 802.11: associated (aid 1)
tuya_1  | Feb 23 18:06:54 fedora hostapd: wlp2s0: STA mac:omitted RADIUS: starting accounting session 062C48007F9FC32C
tuya_1  | Feb 23 18:07:36 fedora dnsmasq[123]: exiting on receipt of SIGTERM
```

```
User space daemon for IEEE 802.11 AP management,
IEEE 802.1X/WPA/WPA2/EAP/RADIUS Authenticator
Copyright (c) 2002-2019, Jouni Malinen <j@w1.fi> and contributors
Python 3.8.10
===========
Configuring AP interface...
RTNETLINK answers: File exists
RTNETLINK answers: File exists
Starting DNSMASQ server...
Starting AP on wlp2s0...
Configuration file: /dev/stdin
wlp2s0: interface state UNINITIALIZED->COUNTRY_UPDATE
ACS: Automatic channel selection started, this may take a bit
wlp2s0: interface state COUNTRY_UPDATE->ACS
wlp2s0: ACS-STARTED 
wlp2s0: ACS-COMPLETED freq=2412 channel=1
Using interface wlp2s0 with hwaddr mac:1:omitted and ssid "vtrust-flash"
wlp2s0: interface state ACS->ENABLED
wlp2s0: AP-ENABLED 
wlp2s0: AP-STA-CONNECTED mac:2:omitted
wlp2s0: AP-STA-DISCONNECTED mac:2:omitted
wlp2s0: AP-STA-CONNECTED mac:2:omitted
wlp2s0: AP-STA-DISCONNECTED mac:2:omitted
wlp2s0: AP-STA-CONNECTED mac:2:omitted
wlp2s0: AP-STA-DISCONNECTED mac:2:omitted
wlp2s0: INTERFACE-DISABLED 
wlp2s0: INTERFACE-ENABLED 

```
Any ideas on where to look next? thx

RaumZeit, 2022-04-15:
I had the exact same problem. Killing `wpa_supplicant` always resulted in newly spawn `wpa_supplicant`s that always hindered the AP to work correctly. Either it was not available or disappeared after connecting with my phone. What helped for me was to disable `NetworkManager.service`, i.e.
```
sudo systemctl stop NetworkManager.service
```
and kill all remaining `wpa_supplicant` instances.
It seems the script base always tried to stop `network-manager.service` to no avail, so `NetworkManager` always restarted it

minax007, 2022-04-17:
@RaumZeit 

Thank you very much!

That resolved the problem.