ct-Open-Source/tuya-convert #893: Stuck at smartconfig, device not connecting to vtrust-flash.

basti85, 2021-01-05:
Hi,

I'm trying to flash 2 Smart Bulbs, but i'm stuck in a loop with smartconfig.
https://templates.blakadder.com/bakibo_TB95.html

One of them is paired with smartlife app giving me main fw V1.1.7 and the same for MCU fw.
The other one would't pair with smartlife, so maybe sth else is broken. I can use it for reference, but as of now, I consider it broken and postponed.

I'm using RTL8188CUS 802.11n WLAN Adapter.

When in EZ mode, the blinking will stop after I press Enter in the script, but the device doesnt connect to the wifi. I don't see anything in the logs of TC.

The good bulb was seen once in smarthack-wifi.log (mac-adress confirmed in smartlife app), but i can't reproduce that and i don't have the logs anymore (took me 2 days to get this far bc of a broken wifi driver). -psk.log ist flooded with 
`could not establish sslpsk socket: [SSL: NO_SHARED_CIPHER] no shared cipher (_ssl.c:1123)
don't panic this is probably just your phone!
`

Any help would be appreciated :)
regards, bb

basti85, 2021-01-05:
I read about the issue with psk ID: 02 ( https://github.com/ct-Open-Source/tuya-convert/wiki/Collaboration-document-for-PSK-Identity-02 )

This might or might not be related, i'm not getting far enough yet.

JonMerlevede, 2021-01-06:
I have the same problem but with a smart plug. It might be related to the PSK issue, but I'm not so sure as I do not see any connection attempt from the plug on the side of my PC; no `SSL: DECRYPTION_FAILED_OR_BAD_RECORD_MAC`-messages, only my phone appears in the logs. There must be some communication though, as the plug shuts itself off when I press enter.

basti85, 2021-01-07:
Same behaviour here, except the mac address of the plug shows up in smarthack-wifi.log.

Can You confirm that, @JonMerlevede ? That's probably when the led stops blinking..

kueblc, 2021-01-09:
Check the OUI of the MAC address to see if it is using a supported chip. Only the ESP82xx is supported, which will always have an OUI of Espressif, Inc.

pdfowler, 2021-01-09:
I'm seeing the same thing, and it looks like my device may not be using ESP8266 anymore. Here's some details:

### Pairing:
1. run the start-flash.sh script ... etc
2. Join iPhone to vtrust-flash network
2. put device in EZ mode
3. press **ENTER** to continue
4. Device stops flashing after about 5 dots (`.....`) are rendered below the `Sending secret                0101` line.
5. Script cycles SmartConfig complete. Resending SmartConfig Packets, ..................... until it gives up.

##### Notes
1. Aside from the "this is probably just your phone" log messages, there is nothing logged to any of the log files when this happens.

### Device Info
Lumary "Smart Recessed Slim LED Panel Light" bought as replacement for accidentally fried one (purchased ~6 mos ago, and working no problem ever since ... until I shorted out the line)

**NEW Device MAC lookup** `D4:A6:51  us - アメリカ合衆国 Tuya Smart Inc.`
**OLD Device MAC lookup** `24:62:AB  cn - 中華人民共和国 Espressif Inc.`

JonMerlevede, 2021-01-09:
@basti85 I unfortunately can't check that right now; I remember it not appearing in any logs, but I might not have looked at the wifi logs carefully enough.

In any case, I also have a device with OUI of Tuya Smart Inc; MAC starting with 84:E3:42. So damn :(. Anyone has an idea what kind of chips are now used instead?

basti85, 2021-01-21:
@JonMerlevede yes, same here.
@kueblc it's indeed the new chip, no esp82xx, so I'm closing here.

closed 2021-01-21 by basti85