ct-Open-Source/tuya-convert #894: unable to connect to the GoSound WP3

comet424, 2021-01-09:
i not sure  i read somewhere it may not be possible with template?

i installed the tuya-convert and setup the phone.  and i get the GoSound plug to flash when i hit Enter.. 

it flashes a few more times then goes Red..  it says for 10 times
Smartconfig complete ...resending smartconfig packets
then tellls 
Device did not appear with the intermediate firmware
Check the *.log files in the scripts folder
Do you want to try flashing another device? [y/N] n

i just bought these off amazon came today.. but i dunno how to get these to work so i can use Home Assistant..  is there something i need to do else?

comet424, 2021-01-09:
i guess i lost the webserver of gosound..    it doesnt show up in the wifi.. i guess running the tuya convert erased it?  but when  i re run app it does make it stop flashing.. and goes red...  it will when i shut the convert tuya app off the wifi plug will flash red and then purple.. 

but i guess these are broken now?

kueblc, 2021-01-09:
Please include your logs when opening a new issue as directed.

In this case however, these devices are not supported due to patched firmware. Please see the wiki and #483 for more information.

Your devices are not broken. They are simply paired with the tuya-convert AP called `vtrust-flash`. You can put them back into pairing mode and use the vendor app to add them again.

closed 2021-01-09 by kueblc

comet424, 2021-01-09:
there is no vendor app for Windows  i dont have a Cell phone so cant use it
i  trying to install this for home assistant

i dont know anything about patched firmware versions  
all i know is its a GoSound  Wp3 ...  figured u guys know what version etc.. i  never done this before.. i googled OnSound  for home assisttant.. got your tuya-convert to erase  GoSound   and now i here cuz it didnt work.. i dont know anything else

so the wiki  probably  wont help me.. but ill look at it .. and ill try to get logs files.. dont know how to get them off the raspberry pi anyways.. but ill try

comet424, 2021-01-09:
smarthack-mqtt.log
`1610148259: mosquitto version 1.5.7 starting
1610148259: Using default config.
1610148259: Opening ipv4 listen socket on port 1883.
1610148259: Opening ipv6 listen socket on port 1883.
^C1610148878: mosquitto version 1.5.7 terminating
1610148928: mosquitto version 1.5.7 starting
1610148928: Using default config.
1610148928: Opening ipv4 listen socket on port 1883.
1610148928: Opening ipv6 listen socket on port 1883.
^C1610149844: mosquitto version 1.5.7 terminating
1610149868: mosquitto version 1.5.7 starting
1610149868: Using default config.
1610149868: Opening ipv4 listen socket on port 1883.
1610149868: Opening ipv6 listen socket on port 1883.
^C1610151403: mosquitto version 1.5.7 terminating
1610158302: mosquitto version 1.5.7 starting
1610158302: Using default config.
1610158302: Opening ipv4 listen socket on port 1883.
1610158302: Opening ipv6 listen socket on port 1883.
^C1610158455: mosquitto version 1.5.7 terminating
1610158638: mosquitto version 1.5.7 starting
1610158638: Using default config.
1610158638: Opening ipv4 listen socket on port 1883.
1610158638: Opening ipv6 listen socket on port 1883.
^C1610159679: mosquitto version 1.5.7 terminating
1610161435: mosquitto version 1.5.7 starting
1610161435: Using default config.
1610161435: Opening ipv4 listen socket on port 1883.
1610161435: Opening ipv6 listen socket on port 1883.
^C1610161663: mosquitto version 1.5.7 terminating
`
smarthack-psk
`new client on port 443 from 10.42.42.38:10044
ID: 0242416f68626d64366147393149465231a3b6b73f69826681a937b14c9465a387612127fd7c15db03efd2719b696817a1
PSK: 0f73f3d26484bb24c76ce2bc8f21137a042558f5defd6748cc9156e194a73919
could not establish sslpsk socket: [SSL: DECRYPTION_FAILED_OR_BAD_RECORD_MAC] decryption failed or bad record mac (_ssl.c:1056)
new client on port 443 from 10.42.42.38:22820
ID: 0242416f68626d643661473f3f3f3f3f3fa3b6b73f69826681a937b14c9465a387612127fd7c15db03efd2719b696817a1
Prefix: b'BAohbmd6aG??????'
PSK: 6bf55b5863b9f75c30f86be224c757478316616a8b2dfc90a882372001e51479
could not establish sslpsk socket: [SSL: DECRYPTION_FAILED_OR_BAD_RECORD_MAC] decryption failed or bad record mac (_ssl.c:1056)
new client on port 443 from 10.42.42.38:15695
ID: 0242416f68626d64366147393149465231a3b6b73f69826681a937b14c9465a387612127fd7c15db03efd2719b696817a1
PSK: 0f73f3d26484bb24c76ce2bc8f21137a042558f5defd6748cc9156e194a73919
could not establish sslpsk socket: [SSL: DECRYPTION_FAILED_OR_BAD_RECORD_MAC] decryption failed or bad record mac (_ssl.c:1056)
new client on port 443 from 10.42.42.38:21119
ID: 0242416f68626d64366147393149465231a3b6b73f69826681a937b14c9465a387612127fd7c15db03efd2719b696817a1
PSK: 0f73f3d26484bb24c76ce2bc8f21137a042558f5defd6748cc9156e194a73919
could not establish sslpsk socket: [SSL: DECRYPTION_FAILED_OR_BAD_RECORD_MAC] decryption failed or bad record mac (_ssl.c:1056)
new client on port 443 from 10.42.42.38:28531
ID: 0242416f68626d64366147393149465231a3b6b73f69826681a937b14c9465a387612127fd7c15db03efd2719b696817a1
PSK: 0f73f3d26484bb24c76ce2bc8f21137a042558f5defd6748cc9156e194a73919
could not establish sslpsk socket: [SSL: DECRYPTION_FAILED_OR_BAD_RECORD_MAC] decryption failed or bad record mac (_ssl.c:1056)
new client on port 443 from 10.42.42.38:11412
ID: 0242416f68626d64366147393149465231a3b6b73f69826681a937b14c9465a387612127fd7c15db03efd2719b696817a1
PSK: 0f73f3d26484bb24c76ce2bc8f21137a042558f5defd6748cc9156e194a73919
could not establish sslpsk socket: [SSL: DECRYPTION_FAILED_OR_BAD_RECORD_MAC] decryption failed or bad record mac (_ssl.c:1056)
new client on port 443 from 10.42.42.38:38163
ID: 0242416f68626d64366147393149465231a3b6b73f69826681a937b14c9465a387612127fd7c15db03efd2719b696817a1
PSK: 0f73f3d26484bb24c76ce2bc8f21137a042558f5defd6748cc9156e194a73919
could not establish sslpsk socket: [SSL: DECRYPTION_FAILED_OR_BAD_RECORD_MAC] decryption failed or bad record mac (_ssl.c:1056)
new client on port 443 from 10.42.42.38:16336
ID: 0242416f68626d64366147393149465231a3b6b73f69826681a937b14c9465a387612127fd7c15db03efd2719b696817a1
PSK: 0f73f3d26484bb24c76ce2bc8f21137a042558f5defd6748cc9156e194a73919
could not establish sslpsk socket: [SSL: DECRYPTION_FAILED_OR_BAD_RECORD_MAC] decryption failed or bad record mac (_ssl.c:1056)
new client on port 443 from 10.42.42.38:11814
ID: 0242416f68626d64366147393149465231a3b6b73f69826681a937b14c9465a387612127fd7c15db03efd2719b696817a1
PSK: 0f73f3d26484bb24c76ce2bc8f21137a042558f5defd6748cc9156e194a73919
could not establish sslpsk socket: [SSL: DECRYPTION_FAILED_OR_BAD_RECORD_MAC] decryption failed or bad record mac (_ssl.c:1056)
[... 2308 lines cut ...]
Stopping NetworkManager...
Configuring AP interface...
RTNETLINK answers: File exists
Starting DNSMASQ server...
Starting AP on wlan0...
Configuration file: /dev/stdin
wlan0: Could not connect to kernel driver
Using interface wlan0 with hwaddr 52:40:c3:9d:5d:ca and ssid "vtrust-flash"
wlan0: interface state UNINITIALIZED->ENABLED
wlan0: AP-ENABLED 
wlan0: AP-STA-CONNECTED cc:2f:71:a3:64:f2
wlan0: AP-STA-CONNECTED 5c:8d:4e:e2:f2:bb
wlan0: AP-STA-CONNECTED 70:03:9f:d1:6e:d0
wlan0: interface state ENABLED->DISABLED
wlan0: AP-STA-DISCONNECTED 70:03:9f:d1:6e:d0
wlan0: AP-STA-DISCONNECTED 5c:8d:4e:e2:f2:bb
wlan0: AP-STA-DISCONNECTED cc:2f:71:a3:64:f2
wlan0: AP-DISABLED 
wlan0: CTRL-EVENT-TERMINATING 
nl80211: deinit ifname=wlan0 disabled_11b_rates=0
AP closed
Stopping DNSMASQ server...
Restarting NetworkManager...

`

comet424, 2021-01-09:
so i skimmed that #480 article   picture isnt of a OnSound device
i searched the page..  there is no word of OnSound 
dont know what that is they talking about in there or anything

comet424, 2021-01-09:
@kueblc  so since your saying these GoSound devices are garbage cuz of a firmware patch

well what brand is good to work for home assistant then since 483  didnt tell me how to repair these Gosound device  using the raspberry pi  as i dont have a cell phone to use   just a laptop... are there any good brands?