ct-Open-Source/tuya-convert #895: flashing SF-550.avs / NX-4458-675 possible?

W11T, 2021-01-09:
There is a Tasmota Template listed for Luminea NX-4458 Power Monitoring Outdoor Plug in [https://templates.blakadder.com/luminea_NX-4458.html](https://templates.blakadder.com/luminea_NX-4458.html).
So i tried to flash it using tuya-convert. The result:

`
tuya-convert v2.4.4
Checking for network interface wlan0... Found.
Checking UDP port 53... Available.
Checking UDP port 67... Available.
Checking TCP port 80... Available.
Checking TCP port 443... Available.
Checking UDP port 6666... Available.
Checking UDP port 6667... Available.
Checking TCP port 1883... Available.
Checking TCP port 8886... Available.
======================================================
  Starting AP in a screen
  Starting web server in a screen
  Starting Mosquitto in a screen
  Starting PSK frontend in a screen
  Starting Tuya Discovery in a screen

======================================================

IMPORTANT
1. Connect any other device (a smartphone or something) to the WIFI vtrust-flash
   This step is IMPORTANT otherwise the smartconfig may not work!
2. Put your IoT device in autoconfig/smartconfig/pairing mode (LED will blink fast). This is usually done by pressing and holding the primary button of the device
   Make sure nothing else is plugged into your IoT device while attempting to flash.
3. Press ENTER to continue

======================================================
Starting smart config pairing procedure
Waiting for the device to install the intermediate firmware
Put device in EZ config mode (blinking fast)
Sending SSID                  vtrust-flash
Sending wifiPassword
Sending token                 00000000
Sending secret                0101
................
SmartConfig complete.
Resending SmartConfig Packets
.................
...
..................
SmartConfig complete.
Resending SmartConfig Packets
..................
SmartConfig complete.
Resending SmartConfig Packets
.............
Device did not appear with the intermediate firmware
Check the *.log files in the scripts folder
`

Is this device supported?

kueblc, 2021-01-09:
Since you've removed the MAC addresses from your logs, I can't check for you. You can look up the OUI and see if you have a device from "Espressif Inc".

W11T, 2021-01-09:
Sorry! The OUIs are:
mac1	b8:27:eb	RasPi (server)
mac2	b2:a8:58	
mac3	18:69:d8	Tuya Smart Inc.
mac4	a6:73:28

alfakenzo, 2021-01-09:
Hi, I got the same error on Luminea ZX-2993-675

W11T, 2021-01-09:
I tried the development branch.
The macs are different from the ones i know.
b8:27:eb:* is the raspi which is the server
b2:a8:58:* appears to be my iphone
a6:73:28:* appears to be another samsung mobile device i connected.
18:69:d8:* appears to be the luminea plug

This time i get:

```
...
................
SmartConfig complete.
Resending SmartConfig Packets
..............
Device did not appear with the intermediate firmware
Check the *.log files in the scripts folder
If the device left easy pairing mode (LED stopped blinking quickly) when config
packets were sent, your device may have too new of firmware for this tool:
see https://github.com/ct-Open-Source/tuya-convert/wiki/Collaboration-document-for-PSK-Identity-02#known-affected-devices

```

alfakenzo, 2021-01-09:
Yeah bitchie hardware

kueblc, 2021-01-10:
> mac3 18:69:d8 Tuya Smart Inc.

Since this is not an Espressif device, you can't use tuya-convert or ESP firmware such as Tasmota. Unfortunately this is becoming very common, better luck next time!

closed 2021-01-10 by kueblc

alfakenzo, 2021-01-10:
But I don't open the hardware!

W11T, 2021-01-10:
all the macs listed look strange to me. None of them matches the ones i know. No OUI listed matches any of the real manufacturers.
The OUI 18:69:d8 shown when i start the plug is registered to Tuya Smart Inc.!
So i don't give much on this.
I'm wondering that this device has a Tasmota Template: https://templates.blakadder.com/luminea_NX-4458.html.

kueblc, 2021-01-10:
We can only support the ESP82xx which always has an OUI of Espressif Inc. Manufacturers have silently switched over to other chips which cannot run ESP firmware, so even device models which used the ESP in the past are susceptible.

W11T, 2021-01-24:
It contains a WB2S module :-(

danielrheinbay, 2022-08-31:
Meanwhile, there is https://github.com/kuba2k2/libretuya which seems to support WB2S boards:

https://docs.libretuya.ml/boards/wb2s/

W11T, 2023-01-14:
Thank you, Daniel!
Do you know of any "beginners guide" for libretuya?
I guess i need some sort of flashing device.
The [doc](https://docs.libretuya.ml/)  just tells me to install PlatformIO available for Windows, Linux and Macs, create a project and start ...
I'm pretty confused! I think there is something missing?