ct-Open-Source/tuya-convert #896: start_flash dumps Gosund WP3 out of pairing mode and fails to connect

gittlp, 2021-01-11:
Hi,

Not sure if this is related to the PSK issue, but when I try to run start_flash against a Gosund WP3 'smart plug', it audibly 'clicks' the relay and goes from being in pairing mode (flashing blue light) to some other mode (solid dull red light), and start_flash goes into retry loop, never connecting ("Resending SmartConfig Packets .............. Device did not appear with the intermediate firmware").  If I don't run start_flash, the WP3 stays in flashing-blue pairing mode indefinitely... but as soon as I hit "Enter" on the command line after connecting to vtrust-flash, etc., the plug dumps.

Not sure which log is most relevant, but I am getting "could not establish sslpsk socket: [SSL: DECRYPTION_FAILED_OR_BAD_RECORD_MAC] decryption failed or bad record mac (_ssl.c:1056)" in smarthack-psk.log.

Logs and output attached.

Many thanks for your work!

[tuya-convert-logs.txt](https://github.com/ct-Open-Source/tuya-convert/files/5794374/tuya-convert-logs.txt)
[tuya-convert-output.txt](https://github.com/ct-Open-Source/tuya-convert/files/5794382/tuya-convert-output.txt)

allenbina, 2021-01-12:
I'm also having this issue with 2 out of 4 of my new Gosund smart plugs ([wp6](https://www.amazon.com/Function-Overload-Protection-Certification-Supports/dp/B07Q2P77JL)).  Two worked fine, but two have this exact same issue.  How did you combine your logs so nicely without doing it by hand?

gittlp, 2021-01-13:
Thanks for reporting @allenbina.  It makes me wonder if 2 working and 2 not working maybe could be because, even if you bought them together, perhaps they came from two (at least) different factory firmware version lots?   Otherwise would be interesting what else could account for the problem (if PSK-related, then why no problem on the working-two?, etc.).  Not sure if there's a way to get the firmware version without running the app, which I believe we're advised not to should it push down an updated firmware.

Re the combined log trick, I learned that from some other issue thread for tuya-convert when I was looking for similar reports, can't remember which one, but to do it just tail \*.log, so e.g. if you're in the main directory that you run start_flash in (say ~/tuya-convert/) it would be "tail -f scripts/*.log" .