ct-Open-Source/tuya-convert #900: vtrust-flash doesn't turn up despite of successful report

loglux, 2021-01-19:
Used docker
```
yes
Checking for network interface wlp3s0... Found.
Checking UDP port 53... Available.
Checking UDP port 67... Available.
Checking TCP port 80... Available.
Checking TCP port 443... Available.
Checking UDP port 6666... Available.
Checking UDP port 6667... Available.
Checking TCP port 1883... Available.
Checking TCP port 8886... Available.
======================================================
  Starting AP in a screen
  Starting web server in a screen
  Starting Mosquitto in a screen
  Starting PSK frontend in a screen
  Starting Tuya Discovery in a screen

======================================================

```
As you see the report is positive but AP is not turned up.

nochkin, 2021-01-29:
You should be able to see the log file located at `scripts/smarthack-wifi.log`

Cyber1000, 2021-01-30:
If using the docker image smarthack-wifi.log may be located in ./data/backups/{timestamp} (if you took the same volume mapping as in docker-compose.sample.yml)

NickVeld, 2021-04-28:
Just so we are clear, the main output does not say anything about a success or failure of AP starting. See `scripts/smarthack-wifi.log` as mentioned before.

Cyber1000, 2021-04-30:
Ah ok, just saw the mentioning of docker (which I added to this project initially) and of scripts/smarthack-wifi.log, thought the thread owner didn't find the file ...
Ok then my PR won't change anything here, since output of success or failure isn't docker specific.

Thanks for pointing that out!

NickVeld, 2021-05-03:
So, may the issue be closed then due to no question and no activity from the author of the issue?