ct-Open-Source/tuya-convert #901: Moes ufo-r2-rf with WR3E

iMatux, 2021-01-19:
Hello! Yesterday I received this IR/RF blaster from Aliexpress, I though it had an ESP chip like ufo-r1 version, but after teardown I saw that it has a WR3E chip, which seems not compatible with Tasmota, as it has a RTL chip inside, like [Tuya Developer site](https://developer.tuya.com/en/docs/iot/service-market/certification-service/general-certification/module-test-tool?id=K9qswhqd5ymnp) says.
This is a really great product has it has IR and 315/433 mhz for RF and it's really cheap.
If anyone needs pics here they are, including wiring prototype from a doc on Tuya site linked before:
[image]
[image]
[image]
[image]

Python1320, 2021-06-28:
[This hint](https://esp8266.ru/forum/threads/modul-wbr3.5289/#post-75621) worked for me just today to make my [WR3E](https://developer.tuya.com/en/docs/iot/wr3e-module-datasheet?id=K9elwlqbfosbc) output ` <RTL8195A> ROM:` on a similar IR device. As I wrote [here](https://github.com/arendst/Tasmota/issues/4855#issuecomment-756707796) it seems the old esp-alike chip has been depreciated so all new devices will come with this unsupported chip from now on... so make sure to request the old version. It's probably going to take a while before anyone manages to port Tasmota or much anything to these chips.

iMatux, 2022-01-16:
A year after I tried again, bought a Esp8266 12S, soldered without pulling any pin, flashed Tasmota and it work, now I have to figure it out how to set GPIO and template in Tasmota and it should work, hopefully.
[image]

MacSass, 2022-01-16:
Hi, 
very interesting - LED wise the interior looks very much like my YTF IR bridge (https://tasmota.github.io/docs/devices/YTF-IR-Bridge/) which still came with ESP8266 and that I´m using as IR blaster.
Due to the several LED it has a great reach - even gets "around corners", obviously reflecting at walls ...

Looking at the HW this one seems to have 3 antennas for sending - so I would guess it should have a good reach for RF too ...

I found yours on AliExpress for about 20€, which is not too bad in pricing.

I´d be interested if you find the GPIO layout (guess it should not be too hard finding it by guess and try ...

Please share your experience - thanks - MacSass

yvesdm3000, 2022-02-09:
Hello,

Could you elaborate a bit how you open the case of this device?
Been pulling it from the side but it feels like it would damage the plastic...
Thanks !

Yves

gonzzovela, 2022-02-23:
@yvesdm3000 I've just opened one and confirm there are no external screws.

You just need to use one of the tipical smartphone repair tools to pen it thorugh the middle. You have two tabs on each side close to the corner.

gonzzovela, 2022-05-09:
has anyone figured out the pins used by the board? I'd like to use it by replacing the WR3E by an ESP.

Thanks,

Agnes56fr, 2022-09-30:
How did you open it ?

yvesdm3000, 2022-09-30:
I pried it open with a plastic tool used to open a smartphone / tablet.

 
Yves

 
-----Original message-----
From: Agnes56fr
Sent: Friday, 30th September 2022, 16:00
To: ct-Open-Source/tuya-convert
Cc: Yves De Muyter; Mention
Subject: Re: [ct-Open-Source/tuya-convert] Moes ufo-r2-rf with WR3E (#901)
 
 
How did you open it ?

RoganDawes, 2023-01-04:
LibreTuya-ESPHome is working fine on this RTL8710-based module, fwiw. But I would love to find a way to do the initial installation OTA if possible, rather than attaching wires to the module pins.

ChrisGroenewald, 2023-09-14:
> LibreTuya-ESPHome is working fine on this RTL8710-based module, fwiw. But I would love to find a way to do the initial installation OTA if possible, rather than attaching wires to the module pins.

@RoganDawes I'd love a bit more info on how you did it. I ended up with some RTL circuit breakers on advise from my electrician and decided I'd figure out how to integrate them later. Dumb idea.

RoganDawes, 2023-09-14:
I'm guessing you have the CBI Astute smart controllers. The only way that I know of to flash them is to open them up, and connect wires to the appropriate pins as per the libretiny project. If you have access to a 3D printer, there are some models that can put pogo pins in the right places to save soldering, although I have not had the best results that way!

Details can be found here: http://docs.libretiny.eu/docs/platform/realtek-ambz/

Use ltchiptool once you have created a suitable config file (basically, the bare minimum, including the `ota:` and `wifi:` details.) Even with nothing wired up, ltchiptool will try to connect to it, and when it times out, will print out the wiring required. You'll also want to find the WR3E pinout to get the corresponding pins on the board/module.

ChrisGroenewald, 2023-09-14:
You guessed correctly. I'm no stranger to the soldering iron, so it's not a problem. I'm considering picking up a spare one and playing with that first to get the process down pat. Awesome info, thank you! I'm looking forward to getting them out of the walled garden.

RoganDawes, 2023-09-14:
Good luck. Shout if you need help with the yaml configuration.

MayorBug, 2023-10-23:
> Good luck. Shout if you need help with the yaml configuration.

Hi Rogan, could you share your yaml configuration for this device?

RoganDawes, 2023-10-23:
> Good luck. Shout if you need help with the yaml configuration.

This is what I had prior to the change where `libretiny` needs to be updated to the cpu platform (`rtl87xx:`). I no longer have any of these devices, so have not bothered to update it. It is strongly recommended to include the `web_server:` and `captive_portal:` components in all firmware uploaded to these devices, in case you enter the wifi credentials incorrectly!). If you are having trouble getting current and power readings from the device, try adding the `inverted:` option to the CF and CF1 pins.

```
substitutions:
  devicename: asc
  upper_devicename: Smart Controller

esphome:
  name: ${devicename}

rtl87xx:
  board: wr3e
  framework:
    version: latest

# Enable logging
logger:
  baud_rate: 0

# Enable Home Assistant API
api:

ota:

wifi:
#  ssid: !secret wifi_ssid
#  password: !secret wifi_password
  ap:
    ssid: $upper_devicename

web_server:
  port: 80

captive_portal:

sensor:
  - platform: hlw8012
    model: BL0937
    sel_pin: 
      number: PA19
      inverted: true
    cf_pin: PA15
    cf1_pin: PA14
    voltage_divider: 1742
    current:
      name: "${upper_devicename} Current"
    voltage:
      name: "${upper_devicename} Voltage"
    power:
      name: "${upper_devicename} Power"
    energy:
      name: "${upper_devicename} Energy"
    update_interval: 5s

binary_sensor:
  - platform: gpio
    id: button
    pin:
      number: PA22
      inverted: true
      mode:
        input: true
    name: "${upper_devicename} Button"
    filters:
      - delayed_on: 10ms
      - delayed_off: 10ms
    on_press:
      - switch.toggle: relay_template

switch:
  - platform: output
    id: relay
    output: relay_output

  - platform: template
    name: "${upper_devicename} Relay"
    id: relay_template
    lambda: |-
      if (id(relay).state) {
        return true;
      } else {
        return false;
      }
    turn_on_action:
      - light.turn_on: red_led
      - switch.turn_on: relay
    turn_off_action:
      - light.turn_off: red_led
      - switch.turn_off: relay
    # optimistic: true # this will change the toggle switch to individual on/off buttons

light:
  - platform: binary
    id: red_led
    output: red_led_output

output:
  - id: red_led_output
    platform: gpio
    pin: PA23
    inverted: true

  - id: relay_output
    platform: gpio
    pin: PA12
    inverted: false

# Blue LED
status_led:
    pin: 
      number: PA05
      inverted: true

```

MayorBug, 2023-10-23:
Thanks for your contribution Rogan, but i think you uploaded code for wrong device. The Moes ufo-r2-rf isn't power metering switch, it is an IR/RF remote.

RoganDawes, 2023-10-23:
> Thanks for your contribution Rogan, but i think you uploaded code for wrong device. The Moes ufo-r2-rf isn't power metering switch, it is an IR/RF remote.

Sorry, I thought you were responding to my post in e.g. https://github.com/ct-Open-Source/tuya-convert/issues/901#issuecomment-1719122345, where I was talking about the CBI Astute Smart devices, which are also based on the WR3E module.

Unfortunately, I do not have the Moes ufo-r2-rf, so have no idea what sort of configuration would be required.

olivluca, 2024-07-15:
I just got one and it's completely different, now the wifi module is a [cbu](https://developer.tuya.com/en/docs/iot/cbu-module-datasheet?id=Ka07pykl5dk4u) (i.e. a BK7231N, which matches the definition at [libretiny](https://upk.libretiny.eu/?profile=moes-ufo-r2-rf-ir-rf-repeater)) and for the rf it uses a module that I cannot identify.
I guess I could flash esphome using libretiny but then I'd lose the rf functionality (the profile is almost empty).
There's also the [OpenBK7231T_App](https://github.com/openshwprojects/OpenBK7231T_App) but, again, not much information about this device.
Any hint on how to identify the rf module and the GPIOs to control it?

[image]

[image]