ct-Open-Source/tuya-convert #902: BW-SHP14 "Device did not appear with the intermediate firmware"

acmeGR, 2021-01-19:
### Before opening a new issue, please read the following

**Are you reporting a successful flash?**
_Please add it to the wiki instead!_

**Do you have a question about configuring your newly flashed device?**
_Please ask in your alternative firmware's support channel instead!_

**Did you have issues getting a device to flash?**
_Please be sure to include your device **name, model, and firmware version** as well as your **logs**!_

**Please remember to close your issue when the problem has been addressed!**

okh-mzny, 2021-01-19:
Heya, 

the log files can be found in the "scripts" directory.
It won't hurt to see all of them, but you're probably most interested in the "smarthack-psk.log", as I suspect the plug is too new, and is running patched firmware which can't be flashed by tuya-convert anymore. (https://github.com/ct-Open-Source/tuya-convert/issues/483)

The log file should show the bulb trying to connect to your Raspberry Pi. Look for its ID, if it starts with 02, you're out of luck and you'll probably have to take apart the socket, check for an ESP8266 and flash it trough Serial (https://tasmota.github.io/docs/Getting-Started/)

Remember to NEVER connect mains power while you're flashing or working on the plug. It could fry the plug, your computer, and even you.

If you need further help with getting Tasmota to run, hit me up! :)

acmeGR, 2021-01-30:
tbh I cant find whre the log files are.

closed 2021-01-30 by acmeGR