ct-Open-Source/tuya-convert #905: Feit Electric Smart Wifi Bulb Failing to Flash

Adrian-at-CrimsonAuzre, 2021-01-23:
I'm trying to flash my new Feit bulb with the included version of Tasmota and it is failing. I'm not getting PSK errors so I'm not sure it's the patched firmware causing trouble. My phone connected on `10.42.42.17` while the bulb connected on `10.42.42.18` and was blinking the whole time during the flashing attempt.

console
```
tuya-convert v2.4.4
Checking for network interface wlp2s0... Found.
Checking UDP port 53... Occupied by systemd-resolve with PID 401348.
Port 53 is needed to resolve DNS queries
Do you wish to terminate systemd-resolve? [y/N] y
Attempting to stop systemd-resolved.service
Checking UDP port 67... Available.
Checking TCP port 80... Available.
Checking TCP port 443... Available.
Checking UDP port 6666... Available.
Checking UDP port 6667... Available.
Checking TCP port 1883... Available.
Checking TCP port 8886... Available.
======================================================
  Starting AP in a screen...................................................................................................................................................................................................................................................................
  Starting web server in a screen
  Starting Mosquitto in a screen
  Starting PSK frontend in a screen
  Starting Tuya Discovery in a screen

======================================================

IMPORTANT
1. Connect any other device (a smartphone or something) to the WIFI vtrust-flash
   This step is IMPORTANT otherwise the smartconfig may not work!
2. Put your IoT device in autoconfig/smartconfig/pairing mode (LED will blink fast). This is usually done by pressing and holding the primary button of the device
   Make sure nothing else is plugged into your IoT device while attempting to flash.
3. Press ENTER to continue

======================================================
Starting smart config pairing procedure
Waiting for the device to install the intermediate firmware
Put device in EZ config mode (blinking fast)
Sending SSID                  vtrust-flash
Sending wifiPassword          
Sending token                 00000000
Sending secret                0101
....................
SmartConfig complete.
Resending SmartConfig Packets
....................
SmartConfig complete.
Resending SmartConfig Packets
....................
SmartConfig complete.
Resending SmartConfig Packets
.....................
SmartConfig complete.
Resending SmartConfig Packets
..................
SmartConfig complete.
Resending SmartConfig Packets
...................
SmartConfig complete.
Resending SmartConfig Packets
......................
SmartConfig complete.
Resending SmartConfig Packets
....................
SmartConfig complete.
Resending SmartConfig Packets
................
Device did not appear with the intermediate firmware
Check the *.log files in the scripts folder
Stopping smart config
Do you want to try flashing another device? [y/N] n
======================================================
Cleaning up...
Closing AP
Exiting...
```

smarthack-mqtt.log
```
[161169.483700]~DLT~406078~INFO     ~FIFO /tmp/dlt cannot be opened. Retrying later...
1611377286: mosquitto version 1.6.9 starting
1611377286: Using default config.
1611377286: Opening ipv4 listen socket on port 1883.
1611377286: Opening ipv6 listen socket on port 1883.
^C1611377446: mosquitto version 1.6.9 terminating
[161338.987235]~DLT~406078~WARNING  ~Lost log messages in user buffer when exiting: 9
```
smarthack-psk.log
```
new client on port 443 from 10.42.42.17:41772
could not establish sslpsk socket: [SSL: NO_SHARED_CIPHER] no shared cipher (_ssl.c:1123)
don't panic this is probably just your phone!
new client on port 443 from 10.42.42.17:41794
could not establish sslpsk socket: [SSL: NO_SHARED_CIPHER] no shared cipher (_ssl.c:1123)
don't panic this is probably just your phone!
new client on port 443 from 10.42.42.17:41802
could not establish sslpsk socket: [SSL: NO_SHARED_CIPHER] no shared cipher (_ssl.c:1123)
don't panic this is probably just your phone!
new client on port 443 from 10.42.42.17:41804
could not establish sslpsk socket: [SSL: NO_SHARED_CIPHER] no shared cipher (_ssl.c:1123)
don't panic this is probably just your phone!
^CTraceback (most recent call last):
  File "./psk-frontend.py", line 113, in <module>
    main()
  File "./psk-frontend.py", line 106, in main
    r,_,_ =  select.select(readables, [], [])
KeyboardInterrupt
```
smarthack-udp.log
```
Listening for Tuya broadcast on UDP 6666
Listening for encrypted Tuya broadcast on UDP 6667
```
smarthack-web.log
```Listening on 10.42.42.1:80
[I 210122 23:48:26 web:2239] 302 GET /generate_204 (10.42.42.17) 0.70ms
[I 210122 23:48:28 web:2239] 200 GET / (10.42.42.17) 0.96ms
[I 210122 23:48:28 web:2239] 302 GET /favicon.ico (10.42.42.17) 0.62ms
[I 210122 23:48:28 web:2239] 304 GET / (10.42.42.17) 0.51ms
[I 210122 23:48:29 web:2239] 302 GET /generate_204 (10.42.42.17) 0.59ms
[I 210122 23:48:29 web:2239] 302 GET /generate_204 (10.42.42.17) 0.58ms
[I 210122 23:48:31 web:2239] 302 GET /generate_204 (10.42.42.17) 0.62ms
[I 210122 23:48:32 web:2239] 302 GET /generate_204 (10.42.42.17) 0.52ms
^CReceived SIGINT, exiting...
```
smarthack-wifi.log
```
System info
===========
94acc3b
Linux onyx 5.4.0-59-generic #65-Ubuntu SMP Thu Dec 10 12:01:51 UTC 2020 x86_64 x86_64 x86_64 GNU/Linux
OpenSSL 1.1.1f  31 Mar 2020
Dnsmasq version 2.80  Copyright (c) 2000-2018 Simon Kelley
Compile time options: IPv6 GNU-getopt DBus i18n IDN DHCP DHCPv6 no-Lua TFTP conntrack ipset auth DNSSEC loop-detect inotify dumpfile

This software comes with ABSOLUTELY NO WARRANTY.
Dnsmasq is free software, and you are welcome to redistribute it
under the terms of the GNU General Public License, version 2 or 3.
hostapd v2.9
User space daemon for IEEE 802.11 AP management,
IEEE 802.1X/WPA/WPA2/EAP/RADIUS Authenticator
Copyright (c) 2002-2019, Jouni Malinen <j@w1.fi> and contributors
Python 3.8.5
===========
Attempting to stop wpa_supplicant
Stopping NetworkManager...
Configuring AP interface...
RTNETLINK answers: File exists
Starting DNSMASQ server...
Starting AP on wlp2s0...
Configuration file: /dev/stdin
Using interface wlp2s0 with hwaddr 9c:b6:d0:8e:eb:37 and ssid "vtrust-flash"
wlp2s0: interface state UNINITIALIZED->ENABLED
wlp2s0: AP-ENABLED 
wlp2s0: AP-STA-CONNECTED c0:ee:fb:37:ed:d5
wlp2s0: AP-STA-CONNECTED 18:69:d8:7a:83:ad
wlp2s0: interface state ENABLED->DISABLED
wlp2s0: AP-STA-DISCONNECTED 18:69:d8:7a:83:ad
wlp2s0: AP-STA-DISCONNECTED c0:ee:fb:37:ed:d5
wlp2s0: AP-DISABLED 
wlp2s0: CTRL-EVENT-TERMINATING 
nl80211: deinit ifname=wlp2s0 disabled_11b_rates=0
AP closed
Stopping DNSMASQ server...
Restarting NetworkManager...
```

IsaacMSchultz, 2021-01-23:
Could you add what the model number / number underneath the wifi logo on the bulb is? You might have an unsupported bulb. I bought some Feit bulbs thinking they were ESP based but it seems they use a different chip.

Here is the bulb listed on blakadder's site
https://templates.blakadder.com/unsupported/feit_electric-BPA800RGBWAG2P.html

Adrian-at-CrimsonAuzre, 2021-01-23:
I can't check right now but that is the right wattage and feature set. Damn it, I bought it because I saw that people were able to easily flash them and it ticked all the boxes.

IsaacMSchultz, 2021-01-24:
Yeah I was thinking the same thing. I ended up returning mine to the store I bought them.

It is probably still worth it to double check the serial number though! You may have gotten lucky and got the right ones and might be having a different issue with tuya-convert.

Adrian-at-CrimsonAuzre, 2021-01-24:
Damn it, mine has `Intertek 5005458` printed on it. Looks like I'm returning and finding something else.

closed 2021-01-24 by Adrian-at-CrimsonAuzre

mkono87, 2021-09-25:
> 
> 
> Damn it, mine has `Intertek 5005458` printed on it. Looks like I'm returning and finding something else.

Where is that printed?