ct-Open-Source/tuya-convert #906: Can’t flash XS-A08 Plug psk Id 01 - fails to reach intermediate flash

jpwise9, 2021-01-23:
Another 01 psk plug that's been reported to work in the past, but now drops out of pairing mode when attempting to flash.

Looks to be the same issue as 
[Can’t flash Aoycocr U3s with psk Id 01
](https://github.com/ct-Open-Source/tuya-convert/issues/825)
[Aoycocr U3S/HYS-U1S-SOCKET-V1.3 not appearing with intermediate firmware](https://github.com/ct-Open-Source/tuya-convert/issues/744)

Board markings are similar to the above - marked as HYS-01-101 V1.3 dated 2019-12-26.

Logs attached. Will be attempting to work out the pinouts and pull a copy of the flash before converting to Tasmota.
[image]
[smarthack-mqtt.log](https://github.com/ct-Open-Source/tuya-convert/files/5859911/smarthack-mqtt.log)
[smarthack-psk.log](https://github.com/ct-Open-Source/tuya-convert/files/5859912/smarthack-psk.log)
[smarthack-udp.log](https://github.com/ct-Open-Source/tuya-convert/files/5859913/smarthack-udp.log)
[smarthack-web.log](https://github.com/ct-Open-Source/tuya-convert/files/5859914/smarthack-web.log)
[smarthack-wifi.log](https://github.com/ct-Open-Source/tuya-convert/files/5859915/smarthack-wifi.log)

kueblc, 2021-01-27:
Thanks for reporting @jpwise9! I'll be interested to take a look at the firmware once you have a backup. Hopefully that will be enough to work out what's missing.

jpwise9, 2021-02-09:
OK, Well trying to do it as non destrestructively as possible was a bust. Pinouts are completely different to anything else I could find, including the video linked off one of the above cases. So one was sacrificed for the cause.

Pinouts of PCB, Plug Interface, and backup of current flash attached.
[image]
[image]
[XS-A18-image1M.zip](https://github.com/ct-Open-Source/tuya-convert/files/5949775/XS-A18-image1M.zip)

jpwise9, 2021-02-11:
Edit : finally found a unit with a similar layout for power and rx/tx, but gpio0 varies. Double checked mine and looks like I've got a short from the pin I determined to be gpio0 to the adjacent pin which matches the other diagram. 
https://tasmota.github.io/docs/devices/Hyleton-313-Smart-Plug/
Might be I've scratched/damaged/shorted some tracks when I was trying to trace it without removal. So FYI if you're trying a serial flash and can't get into programming mode - try the next pin along.

jpwise9, 2021-02-13:
OK, so short version - yes, I damaged the board in the first plug.  I can still talk to the ESP, but the wireless isn't working anymore, so can't try updating it. Following results are from a second plug.
Original firmware from the second plug was captured, then tried to do a packet capture while pairing with the TUYA app.  Had some hiccups with the wireless on my pi, and the crappy old tablet I was using for the app.  Ended up with 3 packet captures progressing through the process, and hopefully it didn't miss any key packets while doing so. But my understanding is that they should only change to psk02 once the firmware is updated - can anyone confirm if the update is a manual step, or automatic?
Since the Tuya app kept hanging on the old tablet I was using I ended up crashing out, but when i went back in it reported that an RGB plug was paired and showed in the device list. Tried tuya-convert again and now i've suddenly got a psk02 ID.
Plug 1 - Original psk01 firmware - [XS-A18-image1M.zip](https://github.com/ct-Open-Source/tuya-convert/files/5949775/XS-A18-image1M.zip)
Plug 2 - Original psk01 firmware - 
[XS-A18-Plug2-image1M-Virgin.zip](https://github.com/ct-Open-Source/tuya-convert/files/5975395/XS-A18-Plug2-image1M-Virgin.zip)
Plug 2 - Post Pairing - now psk02 - 
[XS-A18-Plug2-image1M-psk02.zip](https://github.com/ct-Open-Source/tuya-convert/files/5975397/XS-A18-Plug2-image1M-psk02.zip)
Packet Captures - 
[packet-capture.zip](https://github.com/ct-Open-Source/tuya-convert/files/5975399/packet-capture.zip)

@kueblc - Over to you if you want to include them in the psk02 research, although I'd be interested in hearing your thoughts about whether they normally fw update without prompting, or the sudden change from psk01 to 02 after pairing.

Unrelated - but looks like these plugs, and probably the Aoyocr ones with HYS markings + others are probably manufactured/odm by Shenzhen Hysiry Technology Co., Ltd - http://www.hysiry.com/
The exact same esp board in these ones also features in several of their FCC submissions for other smart plugs with similar small form factor layouts - see internal photos for [FCCID - 2AKBP-X6](https://apps.fcc.gov/oetcf/eas/reports/ViewExhibitReport.cfm?mode=Exhibits&RequestTimeout=500&calledFromFrame=N&application_id=Y8QibS6vNvTd%2FsNf59P6OA%3D%3D&fcc_id=2AKBP-X6)

Thx.
Jp.

vjversatile, 2021-03-17:
my 01 device says this in the smarthack-web.log

POST /d.json?a=tuya.device.config.get&et=1&other={"token":"00000000","region":"US"}&t=30&uuid=a99fb2d31ca184ab&v=4.1&sign=32c238100b1e6deaffad450f6c
7e7a28
Host: a2.tuyaus.com
User-Agent: TUYA_IOT_SDK
Connection: keep-alive
Content-Type: application/x-www-form-urlencoded; charset=UTF-8
Content-Length: 37

payload FDC1D2488B4990F7E9809C7C242D4FB4
WARNING: it appears this device does not use an ESP82xx and therefore cannot install ESP based firmware
Answer tuya.device.dynamic.config.get
reply {"result":"fn8oBaEAaJe3I+enO7OkYdd0LkD9WHnY6k8vmvctrJZO2mxsulPaudQJ2GnW79rK9F2VBmqNqTqJcWuiX4YubCbzlMGOSEbifnknbojomLe7mevrGF2WUpAllBDW/Q5l","
t":1615967044,"sign":"76beb081f83360b4"}
[I 210317 07:44:04 web:2243] 200 POST /d.json?a=tuya.device.config.get&et=1&other={"token":"00000000","region":"US"}&t=30&uuid=a99fb2d31ca184a
b&v=4.1&sign=32c238100b1e6deaffad450f6c7e7a28 (10.42.42.1) 6.31ms

jobinau, 2024-08-08:
The discussions in this thread were super useful for me, especially the image by @jpwise9.  
Here is my ESP8266 from a smart plug
[image]
Which has the same pin configuration.  The Reset point was not marked on my board. So, your image solved that puzzle.
Thanks a lot.
By the way, even after opening the shield, the board is still working fine and I am able to flash :)