ct-Open-Source/tuya-convert #912: KS-632-1 Smart Wall Socket Device did not appear with the intermediate firmware

zsmaczko, 2021-01-27:
This device came in a box like previous KS-621 Smart Wall Socket, which is possible to convert with tuya-convert. There is a template for tasmota, etc. The appearance of the devices are the same.

Using the factory recommended android program after paired it says the type is KS-632-1
The problem is tuya convert is not possible (for me).

Here are the logs:
[smarthack-mqtt.log](https://github.com/ct-Open-Source/tuya-convert/files/5879202/smarthack-mqtt.log)
[smarthack-psk.log](https://github.com/ct-Open-Source/tuya-convert/files/5879203/smarthack-psk.log)
[smarthack-udp.log](https://github.com/ct-Open-Source/tuya-convert/files/5879204/smarthack-udp.log)
[smarthack-web.log](https://github.com/ct-Open-Source/tuya-convert/files/5879205/smarthack-web.log)
[smarthack-wifi.log](https://github.com/ct-Open-Source/tuya-convert/files/5879206/smarthack-wifi.log)

Please help, if you can.

kueblc, 2021-01-27:
Did you running `./start_flash.sh` more than once?

zsmaczko, 2021-01-27:
I run ./start_flash.sh many times. I tried the flashing another device section with the same device too.

dimmuboy, 2021-09-28:
If it's about this device https://templates.blakadder.com/KS-621.html I flashed it today via tuya-convert and it works well

kueblc, 2021-11-13:
Possibly related #919 #999