ct-Open-Source/tuya-convert #913: Problem to flash 3 of my 4 Tuya SP22

ardu1, 2021-01-27:
Hello, 

I have 4 Tuya SP22, 2200W socks. All I bought in one package before 3 years. I tried to flash the first with Tasmota. It works fine. Than I took the next, but it does not work with the other sockets. The packets could not send. This is shown in the terminal:

_Sending secret                0101
................
SmartConfig complete.
Resending SmartConfig Packets
.................
It tired serveral times, than stopped it._

In the log I found this: 

_could not establish sslpsk socket: [SSL: NO_SHARED_CIPHER] no shared cipher (_ssl.c:1056)_ 

Has anybody any idea?

Thanks and best regards!

problem is, that the ### Before opening a new issue, please read the following

**Are you reporting a successful flash?**
_Please add it to the wiki instead!_

**Do you have a question about configuring your newly flashed device?**
_Please ask in your alternative firmware's support channel instead!_

**Did you have issues getting a device to flash?**
_Please be sure to include your device **name, model, and firmware version** as well as your **logs**!_

**Please remember to close your issue when the problem has been addressed!**

kueblc, 2021-01-27:
> Please be sure to include your device **name, model, and firmware version** as well as your **logs**!

Please post the logs so I can help

ardu1, 2021-01-27:
pi@raspberrypi:~/tuya-convert/scripts $ cat smarthack-udp.log
Listening for Tuya broadcast on UDP 6666
Listening for encrypted Tuya broadcast on UDP 6667
192.168.178.27 {"ip":"192.168.178.27","gwId":"84534801600194bac405","active":2,"ability":0,"mode":0,"encrypt":true,"productKey":"0fHWRe8ULjtmnBNd","version":"3.3"}
192.168.178.57 {"ip":"192.168.178.57","gwId":"84534801600194badda6","active":2,"ability":0,"mode":0,"encrypt":true,"productKey":"0fHWRe8ULjtmnBNd","version":"3.3"}
^CListening for Tuya broadcast on UDP 6666
Listening for encrypted Tuya broadcast on UDP 6667
192.168.178.57 {"ip":"192.168.178.57","gwId":"84534801600194badda6","active":2,"ability":0,"mode":0,"encrypt":true,"productKey":"0fHWRe8ULjtmnBNd","version":"3.3"}

ardu1, 2021-01-27:
pi@raspberrypi:~/tuya-convert/scripts $ cat smarthack-mqtt.log
1611693385: mosquitto version 1.5.7 starting
1611693385: Using default config.
1611693385: Opening ipv4 listen socket on port 1883.
1611693385: Opening ipv6 listen socket on port 1883.
^C1611693507: mosquitto version 1.5.7 terminating
1611740630: mosquitto version 1.5.7 starting
1611740630: Using default config.
1611740630: Opening ipv4 listen socket on port 1883.
1611740630: Opening ipv6 listen socket on port 1883.
^C1611742526: mosquitto version 1.5.7 terminating

ardu1, 2021-01-27:
pi@raspberrypi:~/tuya-convert/scripts $ cat smarthack-web.log
Listening on 10.42.42.1:80
[I 210126 20:36:44 web:2243] 302 GET /hotspot-detect.html (10.42.42.29) 3.71ms
[I 210126 20:36:44 web:2243] 200 GET / (10.42.42.29) 4.55ms
[I 210126 20:36:45 web:2243] 302 GET /hotspot-detect.html (10.42.42.29) 2.47ms
[I 210126 20:36:45 web:2243] 200 GET / (10.42.42.29) 2.49ms
[I 210126 20:36:45 web:2243] 302 GET /hotspot-detect.html (10.42.42.29) 3.17ms
[I 210126 20:36:45 web:2243] 302 GET /hotspot-detect.html (10.42.42.29) 3.30ms
[I 210126 20:36:45 web:2243] 200 GET / (10.42.42.29) 3.22ms
[I 210126 20:36:45 web:2243] 200 GET / (10.42.42.29) 2.96ms
^CReceived SIGINT, exiting...
Listening on 10.42.42.1:80
[I 210127 09:44:35 web:2243] 302 GET /hotspot-detect.html (10.42.42.29) 3.80ms
[I 210127 09:44:35 web:2243] 200 GET / (10.42.42.29) 4.33ms
[I 210127 09:44:36 web:2243] 302 GET /hotspot-detect.html (10.42.42.29) 2.57ms
[I 210127 09:44:36 web:2243] 200 GET / (10.42.42.29) 3.17ms
[I 210127 09:44:36 web:2243] 302 GET /hotspot-detect.html (10.42.42.29) 3.60ms
[I 210127 09:44:36 web:2243] 200 GET / (10.42.42.29) 3.05ms
[I 210127 09:44:36 web:2243] 302 GET /hotspot-detect.html (10.42.42.29) 3.28ms
[I 210127 09:44:36 web:2243] 200 GET / (10.42.42.29) 3.32ms
[I 210127 09:51:57 web:2243] 302 GET /hotspot-detect.html (10.42.42.29) 4.60ms
[I 210127 09:51:57 web:2243] 200 GET / (10.42.42.29) 3.76ms
[I 210127 09:51:58 web:2243] 302 GET /hotspot-detect.html (10.42.42.29) 2.93ms
[I 210127 09:51:58 web:2243] 200 GET / (10.42.42.29) 3.00ms
[I 210127 09:51:58 web:2243] 302 GET /hotspot-detect.html (10.42.42.29) 3.01ms
[I 210127 09:51:58 web:2243] 302 GET /hotspot-detect.html (10.42.42.29) 3.15ms
[I 210127 09:51:58 web:2243] 200 GET / (10.42.42.29) 3.06ms
[I 210127 09:51:58 web:2243] 200 GET / (10.42.42.29) 3.23ms
^CReceived SIGINT, exiting...

ardu1, 2021-01-27:
^Cpi@raspberrypi:~/tuya-convert/scripts $ cat smarthack-psk.log
new client on port 443 from 10.42.42.29:63944
could not establish sslpsk socket: [SSL: NO_SHARED_CIPHER] no shared cipher (_ssl.c:1056)
don't panic this is probably just your phone!
new client on port 443 from 10.42.42.29:63945
could not establish sslpsk socket: [SSL: NO_SHARED_CIPHER] no shared cipher (_ssl.c:1056)
don't panic this is probably just your phone!
new client on port 443 from 10.42.42.29:63946
could not establish sslpsk socket: [SSL: WRONG_SSL_VERSION] wrong ssl version (_ssl.c:1056)
don't panic this is probably just your phone!
new client on port 443 from 10.42.42.29:63950
could not establish sslpsk socket: [SSL: NO_SHARED_CIPHER] no shared cipher (_ssl.c:1056)
don't panic this is probably just your phone!
new client on port 443 from 10.42.42.29:55406
could not establish sslpsk socket: [SSL: NO_SHARED_CIPHER] no shared cipher (_ssl.c:1056)
don't panic this is probably just your phone!
new client on port 443 from 10.42.42.10:6133
ID: 0242416f68626d64366147393149465231796589db869841e058be5cea74356bcf4ef685c594b77ff290efa1a2c517e9e2
PSK: da55a451927bbf321829f33b89d2252d828cc4de136997f377a4beb234e4d711
could not establish sslpsk socket: [SSL: DECRYPTION_FAILED_OR_BAD_RECORD_MAC] decryption failed or bad record mac (_ssl.c:1056)
new client on port 443 from 10.42.42.10:9591
ID: 0242416f68626d64366147393149465231796589db869841e058be5cea74356bcf4ef685c594b77ff290efa1a2c517e9e2
PSK: da55a451927bbf321829f33b89d2252d828cc4de136997f377a4beb234e4d711
could not establish sslpsk socket: [SSL: DECRYPTION_FAILED_OR_BAD_RECORD_MAC] decryption failed or bad record mac (_ssl.c:1056)

kueblc, 2021-01-27:
> ID: 0242416f68626d64366147393149465231796589db869841e058be5cea74356bcf4ef685c594b77ff290efa1a2c517e9e2

Unfortunately OTA flashing is not currently possible with this firmware, see #483 and the wiki for more information

closed 2021-01-27 by kueblc

ardu1, 2021-01-27:
Thanks, I tried to understand #483, but it is too strong for me :-(. Could I solve the problem? I yes, could you give me some more hints?

Thanks!