ct-Open-Source/tuya-convert #916: CE Smart Home - Wi-Fi Smart Dimmer - Won't connect to AP

hoatz, 2021-02-01:
I have 2 CE Smarthome Dimmers I bought from Costco some time ago and can't get them to connect to the Tuya Convert AP, which just times out, only seeing my phone connected.

The AP is running on Linux Mint with a TPlink T2UA wifi adapter, all seems to be working fine and I can connect to the AP with my smartphone. I've tried the 6 press method to get the dimmer into AP mode many times, also turned power on and off.

Dimmer is working manually, lights are flashing quickly as expected. I have not used the CE Smarthome AP, so these are fresh out of the box.
 
Dimmer: CE Smart Home - Wi-Fi Smart Dimmer (Single-Pole)
WF500D (1181 Ver 3.2 on side) Wenzhou MTLC Electric Appliances
Not sure where to find firmware version...

The IP of 10.42.42.1:80 in smarthack-web.log seems odd, for what I think is a type of TFTP address?

Many thanks to all who have helped with this project!!!

**Console Output**
_user@Linux-Mint:~/tuya-convert$ sudo ./start_flash.sh 
[sudo] password for user:             
tuya-convert v2.4.5
Checking for network interface wlxec086b0c2f17... Found.
Checking UDP port 53... Occupied by systemd-resolve with PID 538.
Port 53 is needed to resolve DNS queries
Do you wish to terminate systemd-resolve? [y/N] y
Attempting to stop systemd-resolved.service
Checking UDP port 67... Available.
Checking TCP port 80... Available.
Checking TCP port 443... Available.
Checking UDP port 6666... Available.
Checking UDP port 6667... Available.
Checking TCP port 1883... Occupied by mosquitto with PID 925.
Port 1883 is needed to run MQTT
Do you wish to terminate mosquitto? [y/N] y
Attempting to stop mosquitto.service

Checking TCP port 8886... Available.
======================================================
  Starting AP in a screen.........................................................................................................................................................................................................................................................................................................................................................................................................................................................................................................................................................................................................................................................................................................................................................................................................
  Starting web server in a screen
  Starting Mosquitto in a screen
  Starting PSK frontend in a screen
  Starting Tuya Discovery in a screen

======================================================

IMPORTANT
1. Connect any other device (a smartphone or something) to the WIFI vtrust-flash
   This step is IMPORTANT otherwise the smartconfig may not work!
2. Put your IoT device in autoconfig/smartconfig/pairing mode (LED will blink fast). This is usually done by pressing and holding the primary button of the device
   Make sure nothing else is plugged into your IoT device while attempting to flash.
3. Press ENTER to continue

======================================================
Starting smart config pairing procedure
Waiting for the device to install the intermediate firmware
Put device in EZ config mode (blinking fast)
Sending SSID                  vtrust-flash
Sending wifiPassword          
Sending token                 00000000
Sending secret                0101
..................
SmartConfig complete.
Resending SmartConfig Packets
..................
SmartConfig complete.
Resending SmartConfig Packets
..................
SmartConfig complete.
Resending SmartConfig Packets
...................
SmartConfig complete.
Resending SmartConfig Packets
..................
SmartConfig complete.
Resending SmartConfig Packets
...................
SmartConfig complete.
Resending SmartConfig Packets
..................
SmartConfig complete.
Resending SmartConfig Packets
....................
SmartConfig complete.
Resending SmartConfig Packets
...................
SmartConfig complete.
Resending SmartConfig Packets
..............
Timed out while waiting for the device to (re)connect
Stopping smart config
======================================================
Attempting to diagnose the issue...
No ESP82xx based devices connected according to your wifi log.
Here is a list of all the MAC addresses that connected:
ea:1b:68:03:49:cd

If you see your IoT device in this list, it is not an ESP82xx based device.
Otherwise, another issue may be preventing it from connecting.
For additional information, check the *.log files inside the scripts folder.
Please include these logs when opening a new issue on our GitHub issue tracker.
======================================================
Do you want to try flashing another device? [y/N] n
======================================================
Cleaning up...
Closing AP
Exiting..._

[smarthack-udp.log](https://github.com/ct-Open-Source/tuya-convert/files/5906445/smarthack-udp.log)
[smarthack-web.log](https://github.com/ct-Open-Source/tuya-convert/files/5906446/smarthack-web.log)
[smarthack-wifi.log](https://github.com/ct-Open-Source/tuya-convert/files/5906447/smarthack-wifi.log)
[smarthack-mqtt.log](https://github.com/ct-Open-Source/tuya-convert/files/5906448/smarthack-mqtt.log)
[smarthack-psk.log](https://github.com/ct-Open-Source/tuya-convert/files/5906449/smarthack-psk.log)

hoatz, 2021-02-09:
It seems some are having luck when they connect a computer to the vtrust AP, instead of a smartphone as I did.
I'll test again and report back.

greenhawk89, 2021-02-11:
> 
> 
> It seems some are having luck when they connect a computer to the vtrust AP, instead of a smartphone as I did.
> I'll test again and report back.

Any luck with a PC vs using phones for the vtrust AP dummy?

hoatz, 2021-02-16:
I'll let you know, haven't had a chance to tinker with this in the last week.