ct-Open-Source/tuya-convert #917: Flashing fails on GoSund model WP5

aug70, 2021-02-08:
### Before opening a new issue, please read the following

**Are you reporting a successful flash?**
No

**Do you have a question about configuring your newly flashed device?**
No

**Did you have issues getting a device to flash?**
Yes

**Please remember to close your issue when the problem has been addressed!**
I tried to flash with tuya-convert OTA, followed the steps below;
* Connected mobile phone to `vtrust-flash` wifi network.
* Made IOT device (GoSund small plug Model: WP5) blink fast in pairing mode.
* Output is below, procedure takes too short and starts retrying sending data too soon!

Eventually fails!

```
IMPORTANT
1. Connect any other device (a smartphone or something) to the WIFI vtrust-flash
   This step is IMPORTANT otherwise the smartconfig may not work!
2. Put your IoT device in autoconfig/smartconfig/pairing mode (LED will blink fast). This is usually done by pressing and holding the primary button of the device
   Make sure nothing else is plugged into your IoT device while attempting to flash.
3. Press ENTER to continue

======================================================
Starting smart config pairing procedure
Waiting for the device to install the intermediate firmware
Put device in EZ config mode (blinking fast)
Sending SSID                  vtrust-flash
Sending wifiPassword
Sending token                 00000000
Sending secret                0101
................
SmartConfig complete.
Resending SmartConfig Packets
.................
SmartConfig complete.
Resending SmartConfig Packets
.................
SmartConfig complete.
Resending SmartConfig Packets
................
SmartConfig complete.
Resending SmartConfig Packets
.................
SmartConfig complete.
Resending SmartConfig Packets
.................
SmartConfig complete.
Resending SmartConfig Packets
................
SmartConfig complete.
Resending SmartConfig Packets
.................
SmartConfig complete.
Resending SmartConfig Packets
.................
SmartConfig complete.
Resending SmartConfig Packets
................
SmartConfig complete.
Resending SmartConfig Packets
..............
Timed out while waiting for the device to (re)connect
======================================================
Attempting to diagnose the issue...
Your device's firmware is too new.
Tuya patched the PSK vulnerability that we use to establish a connection.
You might still be able to flash this device over serial.
For more information and to follow progress on solving this issue see:
https://github.com/ct-Open-Source/tuya-convert/wiki/Collaboration-document-for-PSK-Identity-02
======================================================
Do you want to try flashing another device? [y/N]
```

kueblc, 2021-02-08:
Explanation is printed above

> Attempting to diagnose the issue...
> Your device's firmware is too new.
> Tuya patched the PSK vulnerability that we use to establish a connection.
> You might still be able to flash this device over [...]

closed 2021-02-08 by kueblc

MarcelRobitaille, 2021-04-10:
@aug70 were you able to get this to work? Did you flash by serial? I cannot find a way to open my device without damaging it. Sorry if this is not the correct place for this.

kohrar, 2021-05-13:
> @aug70 were you able to get this to work? Did you flash by serial? I cannot find a way to open my device without damaging it. Sorry if this is not the correct place for this.

I had to pry apart the casing in order to flash it via serial. As long as you don't rip the casing, you can use some heat to mold it back into shape and then superglue it shut.

I added the following firmware dumps for this model to the PSK02 wiki page.
* [wp5_02.bin.gz](https://github.com/ct-Open-Source/tuya-convert/files/6470124/wp5_02.bin.gz)
* [wp5_03.bin.gz](https://github.com/ct-Open-Source/tuya-convert/files/6470144/wp5_03.bin.gz)

aug70, 2021-06-21:
Yes I had to crack open the shell from the bottom round plate, it looks a bit mutilated on the bottom now but it works just fine... I put back together and used a glue gun...