ct-Open-Source/tuya-convert #919: Neo Coolcam Smoke detector

SM-26, 2021-02-19:
Hello folks!

I'm trying to flash my Neo Coolcam smoke detector.
I was using the Smart life app, but for some reason I can't connect to it anymore, so I thought I'd switch FW.

Setup is RPI4 - running 32 bit 2021-01-11-raspios-buster-armhf.
I did Raspi-config to set up wifi.

Using my Android 11 Xiaomi Mi6 I've got into the vtrust-flash AP.

Set my IoT into fast blinking mode.

but got stuck at:
```
SmartConfig complete.
Auto retry in 1s. ...............................................................................................................
Timed out while waiting for the device to (re)connect
```

I've looked at all of the logs, but honestly I don't understant what I'm even looking for
any help would be much appreciated 
Thanks in advanced.

/scripts/Logs:
smarthack-mqtt.log:
``` log
1613741170: mosquitto version 1.5.7 starting
1613741170: Using default config.
1613741170: Opening ipv4 listen socket on port 1883.
1613741170: Opening ipv6 listen socket on port 1883.
1613741180: New connection from 10.42.42.1 on port 1883.
1613741180: New client connected from 10.42.42.1 as 38037867d8bfc0dc8c3c (c1, k60, u'38037867d8bfc0dc8c3c').
1613741180: No will message specified.
1613741180: Sending CONNACK to 38037867d8bfc0dc8c3c (0, 0)
1613741180: Received SUBSCRIBE from 38037867d8bfc0dc8c3c
1613741180:     smart/device/in/38037867d8bfc0dc8c3c (QoS 0)
1613741180: 38037867d8bfc0dc8c3c 0 smart/device/in/38037867d8bfc0dc8c3c
1613741180: Sending SUBACK to 38037867d8bfc0dc8c3c
1613741198: New connection from 10.42.42.1 on port 1883.
1613741198: Client 38037867d8bfc0dc8c3c already connected, closing old connection.
1613741198: Socket error on client 38037867d8bfc0dc8c3c, disconnecting.
1613741198: New client connected from 10.42.42.1 as 38037867d8bfc0dc8c3c (c1, k60, u'38037867d8bfc0dc8c3c').
1613741198: No will message specified.
1613741198: Sending CONNACK to 38037867d8bfc0dc8c3c (0, 0)
1613741198: Received SUBSCRIBE from 38037867d8bfc0dc8c3c
1613741198:     smart/device/in/38037867d8bfc0dc8c3c (QoS 0)
1613741198: 38037867d8bfc0dc8c3c 0 smart/device/in/38037867d8bfc0dc8c3c
1613741198: Sending SUBACK to 38037867d8bfc0dc8c3c
1613741207: New connection from 127.0.0.1 on port 1883.
1613741207: New client connected from 127.0.0.1 as b5ca2932-d955-471b-bc73-29b9ea2d2785 (c1, k60).
1613741207: No will message specified.
1613741207: Sending CONNACK to b5ca2932-d955-471b-bc73-29b9ea2d2785 (0, 0)
1613741207: Received PUBLISH from b5ca2932-d955-471b-bc73-29b9ea2d2785 (d0, q0, r0, m0, 'smart/device/in/38037867d8bfc0dc8c3c', ... (111 bytes))
1613741207: Sending PUBLISH to 38037867d8bfc0dc8c3c (d0, q0, r0, m0, 'smart/device/in/38037867d8bfc0dc8c3c', ... (111 bytes))
1613741207: Received DISCONNECT from b5ca2932-d955-471b-bc73-29b9ea2d2785
1613741207: Client b5ca2932-d955-471b-bc73-29b9ea2d2785 disconnected.
1613741288: Client 38037867d8bfc0dc8c3c has exceeded timeout, disconnecting.
1613741288: Socket error on client 38037867d8bfc0dc8c3c, disconnecting.
^C1613741304: mosquitto version 1.5.7 terminating
1613741431: mosquitto version 1.5.7 starting
1613741431: Using default config.
1613741431: Opening ipv4 listen socket on port 1883.
1613741431: Opening ipv6 listen socket on port 1883.
1613741438: New connection from 10.42.42.1 on port 1883.
1613741438: New client connected from 10.42.42.1 as 38037867d8bfc0dc8c3c (c1, k60, u'38037867d8bfc0dc8c3c').
1613741438: No will message specified.
1613741438: Sending CONNACK to 38037867d8bfc0dc8c3c (0, 0)
1613741438: Received SUBSCRIBE from 38037867d8bfc0dc8c3c
1613741438:     smart/device/in/38037867d8bfc0dc8c3c (QoS 0)
1613741438: 38037867d8bfc0dc8c3c 0 smart/device/in/38037867d8bfc0dc8c3c
[... 608 lines cut ...]
RTNETLINK answers: File exists
Starting DNSMASQ server...
Starting AP on wlan0...
Configuration file: /dev/stdin
wlan0: Could not connect to kernel driver
Using interface wlan0 with hwaddr dc:a6:32:9e:08:73 and ssid "vtrust-flash"
wlan0: interface state UNINITIALIZED->ENABLED
wlan0: AP-ENABLED 
wlan0: AP-STA-CONNECTED 7e:d3:c7:ca:f5:ca
wlan0: AP-STA-CONNECTED d8:bf:c0:dc:8c:3c
wlan0: AP-STA-DISCONNECTED d8:bf:c0:dc:8c:3c
wlan0: AP-STA-CONNECTED d8:bf:c0:dc:8c:3c
wlan0: AP-STA-DISCONNECTED d8:bf:c0:dc:8c:3c
wlan0: AP-STA-CONNECTED d8:bf:c0:dc:8c:3c
wlan0: AP-STA-DISCONNECTED d8:bf:c0:dc:8c:3c
wlan0: AP-STA-CONNECTED d8:bf:c0:dc:8c:3c
wlan0: AP-STA-DISCONNECTED d8:bf:c0:dc:8c:3c
wlan0: interface state ENABLED->DISABLED
wlan0: AP-STA-DISCONNECTED 7e:d3:c7:ca:f5:ca
wlan0: AP-DISABLED 
wlan0: CTRL-EVENT-TERMINATING 
nl80211: deinit ifname=wlan0 disabled_11b_rates=0
AP closed
Stopping DNSMASQ server...
```

kueblc, 2021-02-19:
Hi @SM-26 

Thanks for taking the time to post your logs for review.

It looks like it's getting stuck at the Tuya firmware configuration step. This is where the device expects to receive schema information, but due to the fact we have to support a large variety of devices, we send a generic schema instead.

Have you tried the process multiple times? You may want to try with a fresh instance of `start_flash.sh`. The first connection attempt by any given device will receive a short schema, and subsequent attempts will try a different schema.

SM-26, 2021-02-19:
Hello @kueblc 

I've tried to repeat it several times, here is what I did.

0. clean the logs.
1. run script (in sudo).
2. after the first ``` "SmartConfig complete." ``` (about 10 seconds in)
the fast blinking stops.
then I get time out error.
3. When asked if I want to flash another device, I type Y
4. then on the second try, I get a bit ferther, 
```
SmartConfig complete.
Resending SmartConfig Packets
.................................................
```
5. tried again, now got only to step 2.
6. when asked again, I said no.
7. re ran the script again - this time without sudo.

I've tried again to look at the logs, can't understand what's going on. sorry

Logs:
(hope it's ok that I just dragged them here)
[smarthack-udp (1).log](https://github.com/ct-Open-Source/tuya-convert/files/6011804/smarthack-udp.1.log)
[smarthack-wifi (1).log](https://github.com/ct-Open-Source/tuya-convert/files/6011800/smarthack-wifi.1.log)
[smarthack-psk (1).log](https://github.com/ct-Open-Source/tuya-convert/files/6011802/smarthack-psk.1.log)
[smarthack-mqtt (1).log](https://github.com/ct-Open-Source/tuya-convert/files/6011801/smarthack-mqtt.1.log)
[smarthack-web (1).log](https://github.com/ct-Open-Source/tuya-convert/files/6011799/smarthack-web.1.log)

SM-26, 2021-02-22:
I've thought maybe I'd give Docker way a try

[image]

I must be doing something wrong here.

SM-26, 2021-02-26:
if there is anything more I can contribute?

I feel like the device is turning off just after the initial pairing, and would not stay on for the duration of the flashing.
am I close?

SM-26, 2021-03-08:
Would tuya cconvert over work on sonoff s26?
That way I could maybe try another device?