ct-Open-Source/tuya-convert #920: proposed PR: automate dockerbuild and push to dockerhub

scyto, 2021-02-20:
Hi,

When I find a project like this i tend to fork, make my own multi-architecture containers using github workflows and push to dockerhub as that then becomes easy for me to use down the line.

I am in middle of doing this, i saw you like discussing things before a PR is raised (and that i need to raise it against the development branch).

1) when I get this working would you like me to submit the PR?  (it will include doc changes and file changes to docker-compose example, the docker file and new workflow files, alternately I can keep it mine, but may not keep it that upto date.)
2) if so it would require you to create your own secrets in the master project for your docker repo
3) wow this a large image - are all those reqs really required - eps the dev version of python etc - i find i can reduce image size by switching over to run time version only.  
4) what does the gateway env var do (i am building the container before i have even run the app, lol, so maybe that will become clear to me, so apologies if a dumb question.

(nice project, looking forward to converting 4 devices i have)

Scyto

scyto, 2021-02-21:
for anyone who wants to try my completed solution and comment here see https://github.com/scyto/tuya-convert/blob/master/README.md#use-prebuilt-image

kueblc, 2021-03-12:
Hi @scyto 

Thanks for sharing your thoughts. I'm not familiar with Docker but happy to accept community contributions to improve the experience for everyone.

I'm including @Cyber1000 @ohadlevy @aravindkoneru on this discussion as they too have thoughts on the Docker approach. I defer to the experts, please let me know when you reach a consensus on the best approach and I'll be happy to merge.

Cyber1000, 2021-03-29:
@scyto sounds nice, had a look on your sources
Some points:
**ad 3:**
> are all those reqs really required 

I don't know exactly for now, the install_prereq.sh was here before and has nothing to do with docker itself - I just used the existing install_prereq.sh, maybe @kueblc can elaborate a little bit more on this (why python3-dev is in like you asked). I can't remember if I tried to remove some of the dependencies in it, it's too long since.

**ad 4:**
> what does the gateway env var do

exactly the same as the GATEWAY-variable in config.txt, so no docker specific part here too (except that it actually replaces the GATEWAY-variable in config.txt within the docker container)
Normally it works with 10.42.42.1, sorry I've forgotten which values are allowed here or if this is more or less fixed @kueblc 

**One thing to consider:**
In my PR #915 I've changed to alpine as base, as phusion/baseimage wasn't the best choice to run this on arm like raspberry and seemed to have issues with fedora too. I don't know debian:buster-slim well, but since you've added a lot of platforms/architectures in .github/workflows/build-and-push.yml I assume it may work. Could you try it on different architectures? Alpine is more lightweight (at least without installs 2-3 MB compared to 20+ in debian-buster-slim), so this might be a better choice here. 

But all in all well done

scyto, 2021-03-29:
@Cyber1000 sure i can try alpine, many of my other containers use it and it is often my starting point.  Can't recall why I did it here (did i just default to debian slim or did i hit an alpine roadblock); ah just checked your PR.

Oh I think it was because i couldn't be bothered to chase all the req's on alpine - its always a PITA :-) - that was likely my root of the question about reqs too! (alpine is esp annoying if one has something that needs glibc).

I actually ran these tests on my pi - everything I do is multiarchitecture :-)
[image]

What size is your image coming out at?

scyto, 2021-03-29:
probably better for me to wait for your move to alpine to be merged, then i can rebase and then re-do the github build actions if folks want

scyto, 2021-03-30:
yes, its good to remove git and other items once not needed and yes -dev is generally not needed for just runtime.
(and yes i turned trying to get images small into a personal war :-) - some of my alpine ones for other projects are tiny) 

If i have time at weekend i will take a look.

Cyber1000, 2021-04-02:
as far as I've seen:
- git is needed for git+https://github.com/drbild/sslpsk.git (in the next line with "RUN python3"), perhaps there is an alternative, lateron git isn't needed by the container itself
- -dev seem too much for me too
- bash: the shebang in the scripts is #!/bin/bash, we'll have to try if it is really needed

Pr0mises, 2021-05-13:
> 
> 
> for anyone who wants to try my completed solution and comment here see my forked instructions https://github.com/scyto/tuya-convert/blob/master/README.md#use-prebuilt-image

@scyto  Thank you! This version finally worked for my rasi4 docker version
I also tried the new docker version from @Cyber1000 merged into development branch
had some errors with it tho. 

> tulp not found

> ssl errors: ssl.SSLEOFError: EOF occurred in violation of protocol (_ssl.c:1131) (when connecting my phone)

Cyber1000, 2021-05-15:
Funny that it worked on my machine, no pi4 but that shouldn't matter with docker, maybe another alpine-version available there (which misses something)

Anyway, many thanks for the information! tput seems to be a part of ncurses, I'll have a look at it within the next few days. 
@Pr0mises Which OS version: rasberry OS 32 or 64bit? (Or something different?)

Notice to myself:
- search usages of tput
- install ncurses: `RUN apk add ncurses`

Pr0mises, 2021-05-15:
@Cyber1000  I use the 64bit version of raspberry OS