ct-Open-Source/tuya-convert #921: Stuck in Starting AP in a screen....

TheAirGotCold, 2021-03-01:
Running Linux mint 20.1
tried on both 88x2bu and 8814au chipsets
not sure how to get any other logs or what other logs to get. 
I did look at the previous 2 similar issues and those didn't help.

./start_flash.sh 
tuya-convert v2.4.5
Checking for network interface ... Found.
Checking UDP port 53... Occupied by systemd-resolve with PID 22652.
Port 53 is needed to resolve DNS queries
Do you wish to terminate systemd-resolve? [y/N] y
Attempting to stop systemd-resolved.service

Checking UDP port 67... Available.
Checking TCP port 80... Available.
Checking TCP port 443... Available.
Checking UDP port 6666... Available.
Checking UDP port 6667... Available.
Checking TCP port 1883... Available.
Checking TCP port 8886... Available.
======================================================
  Starting AP in a screen...........

soRailicious, 2021-03-19:
I have exactly the same problems on my raspberry pi 4...

evandavey, 2021-03-23:
Are you flashing via ssh and have lost network when the access point is started?  I solved this by connecting to the AP, re-connecting via ssh (using the IP opened in the web browser on AP connection)  and then re-attaching the screen (using screen -x).

TheAirGotCold, 2021-03-23:
I think this might actually be an issue with my WiFi adapter and AP mode. Sorry for wasting all y'all time.