ct-Open-Source/tuya-convert #922: Add armv7 image for docker

Daveyr, 2021-03-05:
The base image for the Dockerfile is `phusion/baseimage:master`. When I build the docker image it says that an armv7 version cannot be found. Please can you consider compiling an armv7 version of this base image? Pretty sure the rest of the Dockerfile will build just fine otherwise.

Cyber1000, 2021-03-10:
phusion/baseimage:master splits into 386, amd64, arm64, so no armv7 here.
alpine has a lot more options here including armv7, so as you said my PR should fix this.

You might try my branch, there were no other changes to original develop-branch for now.

Daveyr, 2021-03-15:
Thanks. I actually copied most of your Dockerfile, including the base image, and I was able to run on Raspberry Pi 4. The only things that didn't work was the volume (could be my docker-compose.yml) and the entry point (I manually ran `/usr/bin/tuya_convert/start.sh`). But this might not be an issue with your branch; just my cherry picking of changes.

nmeachen, 2021-12-16:
Whatever the issue was here at the time, there is an armv7 image for this now.

Cyber1000, 2022-02-21:
There was a PR merged in May 2020 (after this issue), which solved this, so this issue can be closed.