ct-Open-Source/tuya-convert #924: Issue flash Milfra TB41 switch

becksen, 2021-03-08:
Hi all,

I can't successfully complete flashing Milfra TB41 switch with Tasmota. Getting the following error. I'm using RPI 3B with Raspian to do the flashing.

Regards:

smarthack-wifi.log:

`dnsmasq: failed to create listening socket for 10.42.42.1: Address already in use Starting AP on wlan0... Configuration file: /dev/stdin wlan0: Could not connect to kernel driver Using interface wlan0 with hwaddr b8:27:eb:15:02:2d and ssid "vtrust-flash" wlan0: interface state UNINITIALIZED->ENABLED wlan0: AP-ENABLED wlan0: AP-STA-CONNECTED 8c:ce:4e:f1:d5:43 wlan0: AP-STA-CONNECTED a0:a4:c5:d8:fc:1b wlan0: AP-STA-DISCONNECTED 8c:ce:4e:f1:d5:43 wlan0: AP-STA-CONNECTED 8c:ce:4e:f1:d5:43 wlan0: AP-STA-DISCONNECTED 8c:ce:4e:f1:d5:43 wlan0: AP-STA-CONNECTED 8c:ce:4e:f1:d5:43 wlan0: AP-STA-DISCONNECTED 8c:ce:4e:f1:d5:43 wlan0: AP-STA-CONNECTED 8c:ce:4e:f1:d5:43 wlan0: AP-STA-DISCONNECTED 8c:ce:4e:f1:d5:43 wlan0: AP-STA-CONNECTED 8c:ce:4e:f1:d5:43 wlan0: AP-STA-DISCONNECTED 8c:ce:4e:f1:d5:43 wlan0: AP-STA-CONNECTED 8c:ce:4e:f1:d5:43 wlan0: AP-STA-DISCONNECTED 8c:ce:4e:f1:d5:43 wlan0: AP-STA-CONNECTED 8c:ce:4e:f1:d5:43 wlan0: AP-STA-DISCONNECTED 8c:ce:4e:f1:d5:43 wlan0: AP-STA-CONNECTED 8c:ce:4e:f1:d5:43 wlan0: AP-STA-DISCONNECTED 8c:ce:4e:f1:d5:43 wlan0: interface state ENABLED->DISABLED wlan0: AP-STA-DISCONNECTED a0:a4:c5:d8:fc:1b wlan0: AP-DISABLED nl80211: deinit ifname=wlan0 disabled_11b_rates=0 AP closed Stopping DNSMASQ server... Restarting NetworkManager... Failed to restart network-manager.service: Unit network-manager.service not found.`

In the processing steps it is repeating those steps:
`SmartConfig complete. 
Resending SmartConfig Packets`

Regards

becksen, 2021-03-12:
It seems firmare is to new?
Which firware are support to be flashed over Tuya-Convert?

kueblc, 2021-03-12:
Hi @becksen 

Please post the rest of your logs if you want help determining the issue.

> Which firware are support to be flashed over Tuya-Convert?

There is no way to determine this without trying.

becksen, 2021-03-13:
Hi @kueblc ,

I'm trying to flash Milfra Smart module TB41 with main module and MCU-Module on V1.0.2.
Having carefully read the diagnosis attemps at the end it says firmare is too new. Still looking for a way of moving tasmota on it.

Thanks.

`pi@raspopenhab:~/tuya-convert $ sudo ./start_flash.sh 
tuya-convert v2.4.5
Checking for network interface wlan0... Found.
Checking UDP port 53... Available.
Checking UDP port 67... Available.
Checking TCP port 80... Available.
Checking TCP port 443... Available.
Checking UDP port 6666... Available.
Checking UDP port 6667... Available.
Checking TCP port 1883... Available.
Checking TCP port 8886... Available.
======================================================
  Starting AP in a screen
  Starting web server in a screen
  Starting Mosquitto in a screen
  Starting PSK frontend in a screen
  Starting Tuya Discovery in a screen

======================================================

IMPORTANT
1. Connect any other device (a smartphone or something) to the WIFI vtrust-flash
   This step is IMPORTANT otherwise the smartconfig may not work!
2. Put your IoT device in autoconfig/smartconfig/pairing mode (LED will blink fast). This is usually done by pressing and holding the primary button of the device
   Make sure nothing else is plugged into your IoT device while attempting to flash.
3. Press ENTER to continue

======================================================
Starting smart config pairing procedure
Waiting for the device to install the intermediate firmware
Put device in EZ config mode (blinking fast)
Sending SSID                  vtrust-flash
Sending wifiPassword          
Sending token                 00000000
Sending secret                0101
................
SmartConfig complete.
Resending SmartConfig Packets
.................
SmartConfig complete.
Resending SmartConfig Packets
.................
SmartConfig complete.
Resending SmartConfig Packets
................
SmartConfig complete.
Resending SmartConfig Packets
.................
SmartConfig complete.
Resending SmartConfig Packets
.................
SmartConfig complete.
Resending SmartConfig Packets
.................
SmartConfig complete.
Resending SmartConfig Packets
..................
SmartConfig complete.
Resending SmartConfig Packets
..................
SmartConfig complete.
Resending SmartConfig Packets
.................
SmartConfig complete.
Resending SmartConfig Packets
............
Timed out while waiting for the device to (re)connect
======================================================
Attempting to diagnose the issue...
**Your device's firmware is too new.**
Tuya patched the PSK vulnerability that we use to establish a connection.
You might still be able to flash this device over serial.
For more information and to follow progress on solving this issue see:
https://github.com/ct-Open-Source/tuya-convert/wiki/Collaboration-document-for-PSK-Identity-02
======================================================
Do you want to try flashing another device? [y/N] ^[c
======================================================
Cleaning up...
^[cClosing AP
Exiting...
`

kueblc, 2021-03-13:
You could try flashing it the old fashion way, over serial. Unfortunately there's no way to flash this firmware OTA until someone discovers a new vulnerability that we can exploit.

closed 2021-03-13 by kueblc

becksen, 2021-03-13:
OTA was fine. Never did it via serial. Where can I find more details to flash it via serial?

Regards.

kueblc, 2021-03-13:
You can find many guides online. See the [Tasmota Documentation](https://tasmota.github.io/docs/Getting-Started/#hardware-preparation) for an example.

This is the only way to flash firmware on your device at this time.