ct-Open-Source/tuya-convert #926: Problems flashing Aoycocr U3S

martingjohn, 2021-03-20:
I seem to be having issues flashing Aoycocr U3S, which I've successfully flashed before. Well this is a different box but can't see any obvious differences, I'm presuming it's come with different base firmware

I pulled the latest github repo and am running this on docker on Raspberry Pi 4 (same set up as before - I also tried the previous docker image in case there was an issue with the new version)

```
# docker-compose exec tuya start
tuya-convert v2.4.5
======================================================
TUYA-CONVERT

https://github.com/ct-Open-Source/tuya-convert
TUYA-CONVERT was developed by Michael Steigerwald from the IT security company VTRUST (https://www.vtrust.de/) in collaboration with the techjournalists Merlin Schumacher, Pina Merkert, Andrijan Moecker and Jan Mahn at c't Magazine. (https://www.ct.de/)

======================================================
PLEASE READ THIS CAREFULLY!
======================================================
TUYA-CONVERT creates a fake update server environment for ESP8266/85 based tuya devices. It enables you to backup your devices firmware and upload an alternative one (e.g. ESPEasy, Tasmota, Espurna) without the need to open the device and solder a serial connection (OTA, Over-the-air).
Please make sure that you understand the consequences of flashing an alternative firmware, since you might lose functionality!

Flashing an alternative firmware can cause unexpected device behavior and/or render the device unusable. Be aware that you do use this software at YOUR OWN RISK! Please acknowledge that VTRUST and c't Magazine (or Heise Medien GmbH & Co. KG) CAN NOT be held accountable for ANY DAMAGE or LOSS OF FUNCTIONALITY by typing yes + Enter

yes
Checking for network interface wlan0... Found.
Checking UDP port 53... Available.
Checking UDP port 67... Available.
Checking TCP port 80... Available.
Checking TCP port 443... Available.
Checking UDP port 6666... Available.
Checking UDP port 6667... Available.
Checking TCP port 1883... Available.
Checking TCP port 8886... Available.
======================================================
  Starting AP in a screen
  Starting web server in a screen
  Starting Mosquitto in a screen
  Starting PSK frontend in a screen
  Starting Tuya Discovery in a screen

======================================================

IMPORTANT
1. Connect any other device (a smartphone or something) to the WIFI vtrust-flash
   This step is IMPORTANT otherwise the smartconfig may not work!
2. Put your IoT device in autoconfig/smartconfig/pairing mode (LED will blink fast). This is usually done by pressing and holding the primary button of the device
   Make sure nothing else is plugged into your IoT device while attempting to flash.
3. Press ENTER to continue

======================================================
Starting smart config pairing procedure
Waiting for the device to install the intermediate firmware
Put device in EZ config mode (blinking fast)
Sending SSID                  vtrust-flash
Sending wifiPassword
Sending token                 00000000
Sending secret                0101
................
SmartConfig complete.
Resending SmartConfig Packets
..../start_flash.sh: line 138:   187 Terminated              ./smartconfig/main.py
............................................................................................................
Timed out while waiting for the device to (re)connect
======================================================
Attempting to diagnose the issue...
An ESP82xx based device connected according to your wifi log.
If this is the device you are trying to flash, another issue may be preventing it from flashing.
Otherwise, it could be that the device does not use an ESP82xx or it did not connect.
For additional information, check the *.log files inside the scripts folder.
Please include these logs when opening a new issue on our GitHub issue tracker.
======================================================
Do you want to try flashing another device? [y/N] y
======================================================
```
Logs:
[smarthack-wifi.log](https://github.com/ct-Open-Source/tuya-convert/files/6175918/smarthack-wifi.log)
[smarthack-web.log](https://github.com/ct-Open-Source/tuya-convert/files/6175919/smarthack-web.log)
[smarthack-psk.log](https://github.com/ct-Open-Source/tuya-convert/files/6175920/smarthack-psk.log)

Sadly don't mean a great deal to me

Shad0wWulf, 2021-03-24:
I have the same issue with a 4 pack i just bought. The other 20 i have flashed without issue.

stefan-sherwood, 2021-05-20:
I have gotten a handful of individual ones and they have all worked. I just got a 4-pack and none of them would flash. I am ordering the individuals now to see if those will work.

MorganBlack, 2021-06-17:
I'm experiencing similar issue. ID is 01 so it should work and logs do have any error inside. Any ideas?

[smarthack-wifi.log](https://github.com/ct-Open-Source/tuya-convert/files/6669481/smarthack-wifi.log)
[smarthack-mqtt.log](https://github.com/ct-Open-Source/tuya-convert/files/6669482/smarthack-mqtt.log)
[smarthack-psk.log](https://github.com/ct-Open-Source/tuya-convert/files/6669483/smarthack-psk.log)
[smarthack-udp.log](https://github.com/ct-Open-Source/tuya-convert/files/6669484/smarthack-udp.log)
[smarthack-web.log](https://github.com/ct-Open-Source/tuya-convert/files/6669486/smarthack-web.log)

stefan-sherwood, 2021-06-17:
As a followup to my previous comment, the individual units also failed. I took one apart in the hopes of brute force (wired) reprogramming and the problem turns out to be that it no longer uses the ESP chip. Boo!

MorganBlack, 2021-06-17:
> 
> 
> As a followup to my previous comment, the individual units also failed. I took one apart in the hopes of brute force (wired) reprogramming and the problem turns out to be that it no longer uses the ESP chip. Boo!

Thank you. I might have different issue.

84:f3:eb:68:14:e9 still identifies as Espressif Inc.

Can you post photos of the plug you disassembled?

stefan-sherwood, 2021-06-17:
[image]
[image]
[image]

stefan-sherwood, 2021-06-17:
Interesting discussion about the Beken chip and how to mod devices that won't flash OTA so that they will: https://community.home-assistant.io/t/unkown-tuya-chip/153591/28

MorganBlack, 2021-06-17:
Thank you for the photos. 

I think you're posting in wrong issue thread. 
Unless there is round version of U3S with UK socket that I don't know about.

From your photos your plug looks like one of: (both are round)
https://templates.blakadder.com/aoycocr_X6.html
https://templates.blakadder.com/aoycocr_X5P.html

I'm referencing to: (square)
https://templates.blakadder.com/aoycocr_U3S.html  
Same as in this issue title.

stefan-sherwood, 2021-06-18:
It's worth noting that Aoycocr has been updating all of their devices, first on the ESP8266 with new firmware that makes Tuya-Convert not work, then with new hardware that makes Tasmota not work. I suspect they're doing this across the board as earlier versions of the identical plug I purchased have the same board as in the U3S model.

MorganBlack, 2021-06-18:
You're absolutely right in your suspicion that the board could be updated.

But please look again at the information that original poster & myself provided.

ID of his & my socket in logs starts with 01.

My socket:
"ID: 0142416f68626d64366147393149465231a6fcef7280efeaafa37e4987fa8144665f462786d7c83f9af62693931a9a72186a"

As per this information from this thread https://github.com/ct-Open-Source/tuya-convert/wiki/Collaboration-document-for-PSK-Identity-02:

> PSK ID begins with:
>     01 tuya-convert handles these nicely
 

Also if you look at the ID of the socket in the logs of the original poster you'll notice it starts with 01 as well.

And I have to mention again that the my sockets chip identifies as Espressif Inc. (ESP chip).
His also b4:e6:2d:1e:17:c4 - Espressif Inc.

That's why I suspect that my & his plug is an old version with flashable chip.
And our issue is unrelated to updated board/chip as it is in your case.

Im also aware of:

> It seems that devices which ship with PSK ID 01 firmware and then are updated to firmware which uses PSK ID 02, do not store the pskKey at 0xfb000 as expected.

from the same thread mentioned above but my sockets didn't flash straight from the box. 
They were never connected to the internet so the firmware was never updated from the original.

Again thank you for your attempt to help.

stefan-sherwood, 2021-06-18:
My comments are meant not only for you but also for anyone else with "Problems flashing Aoycocr U3S" who comes across this thread.

MorganBlack, 2021-06-18:
Anyone has any ideas regarding original issue?