ct-Open-Source/tuya-convert #929: Is there a way to tell if the device can be flashed before buying it?

left4taco, 2021-03-31:
Sorry for this seemingly off-topic ticket. I'm aware of the fact that Tuya released a patch that prevent tuya-convert from working, and some devices are already shipped with the new firmware. Therefore, is there anyway to tell if the device could be flashed or not? Or, if it is using esp8266 or realtek chip?

Much obliged.

kueblc, 2021-03-31:
Unfortunately not. At least not with perfect accuracy, as retailers don't advertise this and most wouldn't even know. You can check the community maintained wiki, but this information may not be up to date or reflect differences between batches.

left4taco, 2021-04-01:
@kueblc OK I'll try to buy some and update the list if I found them now compatible. Thanks

closed 2021-04-01 by left4taco