ct-Open-Source/tuya-convert #933: Unable to flash SPH10

langestefan, 2021-04-10:
Hi, I am trying to flash the Blitzwolf SPH10. I do not know which FW version it has, how can I find this?

Anyway, it is succesful up to the point that it connects. Then it gets stuck at:

`SmartConfig complete.
Resending SmartConfig Packets
..................
SmartConfig complete.
Resending SmartConfig Packets
..................
SmartConfig complete.
Resending SmartConfig Packets
...........
`

I can hear the relay click and the blue LED stops flashing when I start tuya convert. The only thing I noticed in the PSK logs is the following:

`new client on port 443 from 10.42.42.18:37151
argument 2 must be str, not bytes
new client on port 443 from 10.42.42.23:21544
argument 2 must be str, not bytes
new client on port 443 from 10.42.42.23:15173
argument 2 must be str, not bytes`

Here are the logs I collected:

[smarthack-mqtt.log](https://github.com/ct-Open-Source/tuya-convert/files/6290840/smarthack-mqtt.log)
[smarthack-psk.log](https://github.com/ct-Open-Source/tuya-convert/files/6290841/smarthack-psk.log)
[smarthack-udp.log](https://github.com/ct-Open-Source/tuya-convert/files/6290842/smarthack-udp.log)
[smarthack-web.log](https://github.com/ct-Open-Source/tuya-convert/files/6290843/smarthack-web.log)
[smarthack-wifi.log](https://github.com/ct-Open-Source/tuya-convert/files/6290844/smarthack-wifi.log)
 
Any help would be appreciated, thanks!

kueblc, 2021-04-13:
> argument 2 must be str, not bytes

Hmm this looks like a Python 2/3 issue. Do you have a virtual environment for Python? It could be that the dependencies are in the wrong place.

langestefan, 2021-04-13:
> > argument 2 must be str, not bytes
> 
> Hmm this looks like a Python 2/3 issue. Do you have a virtual environment for Python? It could be that the dependencies are in the wrong place.

Hi, thank you for responding. No, I used the system python on my linux machine. I had to manually install a lot of python modules befor I was able to run the flash script, perhaps I did something wrong there. Anything I can share to help debug?

langestefan, 2021-04-13:
I just uninstalled sslpsk, tried again and... it worked??? I didn't even reinstall sslpsk. No clue why this worked but my plug is up and running now! :)

kueblc, 2021-04-13:
Good old unplug it and plug it back in :)

Glad you got it working, seems like you may have had multiple installations of sslpsk.

closed 2021-04-13 by kueblc