ct-Open-Source/tuya-convert #934: new Tuya Plug - not flashing

ilker-aktuna, 2021-04-14:
I have a new Tuya plug which is not being flashed.
It has ESP eprom on it I believe because the device name starts with ESP

When I use the tuya-convert tool, it connects to the AP and receives IP address 10.42.42.22 (not 10.42.42.42)

and the smartconfig fails.

I get this log on psk screen:

```
new client on port 443 from 10.42.42.21:52032
could not establish sslpsk socket: [SSL: NO_SHARED_CIPHER] no shared cipher (_ssl.c:1056)
don't panic this is probably just your phone!
new client on port 443 from 10.42.42.21:52034
could not establish sslpsk socket: [SSL: NO_SHARED_CIPHER] no shared cipher (_ssl.c:1056)
don't panic this is probably just your phone!
new client on port 443 from 10.42.42.21:52036
could not establish sslpsk socket: [SSL: NO_SHARED_CIPHER] no shared cipher (_ssl.c:1056)
don't panic this is probably just your phone!
new client on port 443 from 10.42.42.21:52038
could not establish sslpsk socket: [SSL: NO_SHARED_CIPHER] no shared cipher (_ssl.c:1056)
don't panic this is probably just your phone!
new client on port 443 from 10.42.42.21:52040
could not establish sslpsk socket: [SSL: NO_SHARED_CIPHER] no shared cipher (_ssl.c:1056)
don't panic this is probably just your phone!
new client on port 443 from 10.42.42.21:52042
could not establish sslpsk socket: [SSL: NO_SHARED_CIPHER] no shared cipher (_ssl.c:1056)
don't panic this is probably just your phone!
new client on port 443 from 10.42.42.21:52044
could not establish sslpsk socket: [SSL: NO_SHARED_CIPHER] no shared cipher (_ssl.c:1056)
don't panic this is probably just your phone!
new client on port 443 from 10.42.42.22:28194
ID: 0242416f68626d64366147393149465231723688c35df8f0a8c2d2c7beccede9406987f35f5c77ca7cd5469857461ed75e
PSK: d308b3383747fbcce3fd683546e096b3c12765510e643c8f08446c97706e7c28
could not establish sslpsk socket: [SSL: DECRYPTION_FAILED_OR_BAD_RECORD_MAC] decryption failed or bad record mac (_ssl.c:1056)

```

is this a new device ?
what can I do to flash it with Tasmota firmware ?

If I can contribute by supplying information from current firmware I would happily do that.
if that's not required, I will flash using USB flasher. But I don't know how to connect this to my ESP flasher.
where can I find info on it ?

FerFemNemBem, 2021-04-15:
Here you may contribute your information: https://github.com/ct-Open-Source/tuya-convert/wiki/Collaboration-document-for-PSK-Identity-02

ilker-aktuna, 2021-04-15:
Thanks. I've read the documentation. But I could not understand what to do.
The document has a lot of information but does not provide a best practice or steps to follow. 

I'd really appreciate if you can point me where to start from.

FerFemNemBem, 2021-04-15:
This document is only a collection of information about tuya devices with the new firmware with "ID: 02". At the moment it is not possible to flash them using tuya-convert. Hopefully some of the experts will find a solution...

ilker-aktuna, 2021-04-15:
so , there is no contribution that I can add as a user ?
new logs etc from my device , not necessary ?

if that's the case, I really would like to consider flashing the ESP chip.
where can I find a guide for this operation ?

ilker-aktuna, 2021-04-15:
I've opened the devie , but it seems it's not easy to access the chip for flashing.
is there anyone with experience to open these ?
what can I do ?

[image]
[image]

ilker-aktuna, 2021-04-18:
anybody with experience opening this ?
it really seems impossible to disassemble without breaking.

dolnet, 2021-04-21:
I think that the four pads near D2 are for programming. RX TX etc.
You need to measure whether the appropriate places are entering the ESP8266

ilker-aktuna, 2021-04-21:
You mean these :
[image]

How can I measure them ?

ilker-aktuna, 2021-04-22:
I've found information on disassembling the plug on  following page:

https://github.com/xoseperez/espurna/wiki/Hardware-NEO-COOLCAM-NAS-WR01W

but I don'T know how to take the module with 3 layers out of the case.

other info:
https://templates.blakadder.com/neo_coolcam_NAS-WR01W-10A.html

tarontop, 2021-04-28:
I chose it in the end
https://templates.blakadder.com/preflashed.html

hc2hunter, 2021-12-23:
> but I don'T know how to take the module with 3 layers out of the case.

https://www.youtube.com/watch?v=CV8Y61gEQpQ